MarlinFirmware/Marlin #6807: WARNING:  Current bugfix-v1.1.0 is crashing nozzle into the bed when probing

Roxy-3D, 2017-05-20:
WARNING:   Something has changed in the Z-Probing that is causing the nozzle to crash into the bed.

I'm using a BL-Touch and have not confirmed the problem is happening on servo's with a kick down probe leg.     My printer was damaged by what ever is causing this.   So...   I busy trying to figure out how to fix it instead of trying to diagnose this problem.

stevie72, 2017-05-20:
100% Confirmed! 

I´ve had a crash with a mounted Inductive Sensor for Z - Probing on my Heat Bed. 
In a earlier Version of Marlin that worked fine.
It looks like that the Z Axes Endstops having Problems with a Z MIN and a Z MAX Endstop mounted.
The M119 command is showing that the Sensors correctly mounted and working.
I´ll be going to testing out that Problem.

Roxy-3D, 2017-05-20:
I do NOT have a Z-Max endstop.   

But I do have a Y-Max endstop on the machine that got damaged.

stevie72, 2017-05-20:
It looks like that the M119 Command showing the correct function of all Switches X Y Z and the Z- Probing too.
but for real there is no function by tipping on the switch or in the other direction holding metall to the Inductive Sensor. The Motor keeps on running without any effect.

Roxy-3D, 2017-05-20:
It is intermittent too.   It doesn't always do it.   Can somebody else confirm what we are seeing?  

(Please raise your nozzle pretty high so you have time to reset the controller so you don't damage your printer!!!)

stevie72, 2017-05-20:
I think that I´ll going to downgrade my Marlin Software Tomorow. To see if the error has disappeared.

thinkyhead, 2017-05-20:
Please sleuth it out and isolate the cause.

Roxy-3D, 2017-05-21:
Right now we don't have enough clues.   It is possible that it is a BL-Touch problem, except stevie72 thinks he had it happen with an inductive probe.   I'm gluing carriage parts back together again and in the morning I can probably re-assemble things enough to print replacement parts.   At point...  It maybe OK to crash the nozzle a few times because the glued parts are going to be coming off the printer any way.

Bob-the-Kuhn, 2017-05-21:
I'll put my test head on the robot & see if I can get it to crash.  This is just a BLTouch mounted on a spring hinge so it doesn't get hurt if there's a crash.

Roxy-3D, 2017-05-21:
I've had it happen with both G28's and G29 P1's.    It seems the higher the nozzle starts, the more likely that is to happen.  (I'm not sure that is really true.)

bgort, 2017-05-21:
What exactly happened?  I understand that the nozzle crashed into the bed, but did the axis not stop after the bltouch was triggered, or did the bltouch pin not deploy at all?  Or not trigger?  Or did you not see?

I assume you're using endstop interrupts?

Is it possible the bltouch was damaged in the crash (not sure what your configuration looks like), or is it more likely it was damaged or impeded before?  (I've had a piece of stray filament along the shaft increase friction and make it harder for it to move.)

Roxy-3D, 2017-05-21:
> What exactly happened? I understand that the nozzle crashed into the bed, but did the axis not stop after the bltouch was triggered, or did the bltouch pin not deploy at all? Or not trigger? Or did you not see?

The BL-Touch did its usual deployment.   But as the nozzle got close to the glass, I'm not sure if the BL-Touch pulled the pin up or not.   I don't have much test data.   And it is really hard to see things clearly because everything happens over such a small amount of distance.

> I assume you're using endstop interrupts?

I was...   But I turned them off.   And after a few more G28's, it happened again.   This time cracking some of the plastic pieces of the carriage.  

> Is it possible the bltouch was damaged in the crash (not sure what your configuration looks like), or is it more likely it was damaged or impeded before? (I've had a piece of stray filament along the shaft increase friction and make it harder for it to move.)

Right now...  Just about anything is possible.   But after the first crash, I had the nozzle pretty high up and triggered the probe with my finger and everything seem ok.   But when I did the G28 from pretty high up...  It drilled into the bed and broke stuff.

Bob-the-Kuhn, 2017-05-21:
I just did two G29 P1 cycles with 15x15 grids with no problems.  One was done with the probe deploying about 15mm from the bed and the other deployed at 50mm above the bed.

This is on an a Cartesian machine.

martindb, 2017-05-21:
In the last week I have 2 random crashes doing G29 (ABL). I'm with bugfix1.1.x from 2017/05/08. Inductive sensor as Z min stop.

Roxy-3D, 2017-05-21:
I've backed up to 5-10 or 5-11...   There are still random nose dives into the bed when probing.   I don't know what we are fighting...

Grogyan, 2017-05-21:
I am using the same version, Bilinear, no nose dives.
Though currently wondering why Z offset isn't respected after G29.

Bob-the-Kuhn, 2017-05-21:
Is the problem limited to deltas?

martindb, 2017-05-21:
Mine is a prusa i3 clone.

Roxy-3D, 2017-05-21:
#6742 ????

stevie72, 2017-05-21:
Good Morning!
I will be downgrading to an pre version of Marlin Today I´m running on my Mendel90.
Is there the Chance that Eeprom Contend may effect this Problem?
In that case I´ll erasing the Eeprom first and look if the problem is vanished.

thinkyhead, 2017-05-21:
> Though currently wondering why Z offset isn't respected after `G29`.

Which Z offset, and how is it failing to be applied?

bgort, 2017-05-21:
Just tried the latest and Z probing is working fine here.  I only have min endstops, endstop interrupts are enabled.  BLTouch (obviously).

martindb, 2017-05-21:
It's random here. In the last week I've printed at least 20 times. Fails 2 times during G29 (abl).

Roxy-3D, 2017-05-21:
I have my machine glued back together.   I have it doing a 'Physical Bed Leveling' mesh right now with a new BL-Touch.   Everything seems ok with current firmware.     

@bgort   I'm getting ridiculously good values for M48.   I did see the range keep expanding with the very first probes M48 did...   And every M48 after that had .0015 or .002 for a Standard Deviation.   
I need to get my bed leveled, and get my Z_PROBE_OFFST_FROM_EXTRUDER dialed in.   And I need to print some replacement parts.

But it is strange...  (With very limited data) I'm seeing the opposite of you.   The first 5 or 6 numbers the BL-Touch reports keep expanding...  And then they are rock solid.

Roxy-3D, 2017-05-21:
@martindb    
- What kind of probe setup are you using?
- Have you noticed any correlation between a failure and how high the nozzle was when the probe started?

bgort, 2017-05-21:
> @bgort I'm getting ridiculously good values for M48. I did see the range keep expanding with the very first probes M48 did... And every M48 after that had .0015 or .002 for a Standard Deviation. (all heaters off!) ... But it is strange... (With very limited data) I'm seeing the opposite of you. The first 5 or 6 numbers the BL-Touch reports keep expanding... And then they are rock solid.

That's really excellent.  As soon as I'm finished printing I'll try M48 again....

bgort, 2017-05-21:
Done sooner than I anticipated...

```
Send: M48 V4
Recv: 1 of 10: z: -0.045 mean: -0.0450 sigma: 0.000000 min: -0.045 max: -0.045 range: 0.000
Recv: 2 of 10: z: -0.050 mean: -0.0475 sigma: 0.002500 min: -0.050 max: -0.045 range: 0.005
Recv: 3 of 10: z: -0.049 mean: -0.0479 sigma: 0.002125 min: -0.050 max: -0.045 range: 0.005
Recv: 4 of 10: z: -0.048 mean: -0.0478 sigma: 0.001849 min: -0.050 max: -0.045 range: 0.005
Recv: 5 of 10: z: -0.048 mean: -0.0478 sigma: 0.001658 min: -0.050 max: -0.045 range: 0.005
Recv: 6 of 10: z: -0.063 mean: -0.0502 sigma: 0.005702 min: -0.063 max: -0.045 range: 0.018
Recv: 7 of 10: z: -0.064 mean: -0.0521 sigma: 0.007094 min: -0.064 max: -0.045 range: 0.019
Recv: 8 of 10: z: -0.068 mean: -0.0541 sigma: 0.008356 min: -0.068 max: -0.045 range: 0.023
Recv: 9 of 10: z: -0.070 mean: -0.0558 sigma: 0.009336 min: -0.070 max: -0.045 range: 0.025
Recv: 10 of 10: z: -0.068 mean: -0.0570 sigma: 0.009523 min: -0.070 max: -0.045 range: 0.025
Recv: Mean: -0.057000 Min: -0.070 Max: -0.045 Range: 0.025
Recv: Standard Deviation: 0.009523
Recv: 
Send: M48 V4
Recv: 1 of 10: z: -0.079 mean: -0.0788 sigma: 0.000000 min: -0.079 max: -0.079 range: 0.000
Recv: 2 of 10: z: -0.088 mean: -0.0831 sigma: 0.004375 min: -0.088 max: -0.079 range: 0.009
Recv: 3 of 10: z: -0.085 mean: -0.0838 sigma: 0.003680 min: -0.088 max: -0.079 range: 0.009
Recv: 4 of 10: z: -0.088 mean: -0.0847 sigma: 0.003577 min: -0.088 max: -0.079 range: 0.009
Recv: 5 of 10: z: -0.081 mean: -0.0840 sigma: 0.003482 min: -0.088 max: -0.079 range: 0.009
Recv: 6 of 10: z: -0.069 mean: -0.0815 sigma: 0.006512 min: -0.088 max: -0.069 range: 0.019
Recv: 7 of 10: z: -0.074 mean: -0.0804 sigma: 0.006605 min: -0.088 max: -0.069 range: 0.019
Recv: 8 of 10: z: -0.056 mean: -0.0773 sigma: 0.010086 min: -0.088 max: -0.056 range: 0.031
Recv: 9 of 10: z: -0.058 mean: -0.0751 sigma: 0.011372 min: -0.088 max: -0.056 range: 0.031
Recv: 10 of 10: z: -0.045 mean: -0.0721 sigma: 0.014076 min: -0.088 max: -0.045 range: 0.043
Recv: Mean: -0.072125 Min: -0.088 Max: -0.045 Range: 0.043
Recv: Standard Deviation: 0.014076
Recv:
Send: M48 V4
Recv: 1 of 10: z: -0.028 mean: -0.0275 sigma: 0.000000 min: -0.028 max: -0.028 range: 0.000
Recv: 2 of 10: z: -0.018 mean: -0.0225 sigma: 0.005000 min: -0.028 max: -0.018 range: 0.010
Recv: 3 of 10: z: -0.009 mean: -0.0179 sigma: 0.007660 min: -0.028 max: -0.009 range: 0.019
Recv: 4 of 10: z: -0.016 mean: -0.0175 sigma: 0.006673 min: -0.028 max: -0.009 range: 0.019
Recv: 5 of 10: z: -0.020 mean: -0.0180 sigma: 0.006052 min: -0.028 max: -0.009 range: 0.019
Recv: 6 of 10: z: -0.029 mean: -0.0198 sigma: 0.006824 min: -0.029 max: -0.009 range: 0.020
Recv: 7 of 10: z: -0.038 mean: -0.0223 sigma: 0.008850 min: -0.038 max: -0.009 range: 0.029
Recv: 8 of 10: z: -0.048 mean: -0.0255 sigma: 0.011742 min: -0.048 max: -0.009 range: 0.039
Recv: 9 of 10: z: -0.061 mean: -0.0294 sigma: 0.015780 min: -0.061 max: -0.009 range: 0.053
Recv: 10 of 10: z: -0.073 mean: -0.0338 sigma: 0.019772 min: -0.073 max: -0.009 range: 0.064
Recv: Mean: -0.033750 Min: -0.073 Max: -0.009 Range: 0.064
Recv: Standard Deviation: 0.019772
```

Same as before:  early probes are better/more consistent.  This is with the bed at temperature and stable.

martindb, 2017-05-21:
Roxy, I've an inductive sensor. Currently using endstop interrupts.
As prior to G29 I do a G28, I think the noozle high at the begining is not relevant?

Roxy-3D, 2017-05-21:
I'll have 'Hot' numbers later...   But it is strange that we are seeing similar weirdness with the numbers, but opposite in when they happen.    Are you using the BL-Touch that has the little Blue LED inside it?  That is what I have mounted on the extruder right now.

bgort, 2017-05-21:
Yes, it is.  And yes - I'm using the one you sent.  It has a blue LED.

Roxy-3D, 2017-05-21:
>  I think the nozzle height at the beginning is not relevant?
I don't know.   I only had 3 or 4 crashes within a 15 minute time frame.   But it seemed like most of the crashes happened when I started the home operation with the nozzle up 150mm.    It may be coincidence.  But it kind seemed like that was part of the recipe for failure on my machine.

bgort, 2017-05-21:
FWIW, I tried probing from ~100mm and had no trouble.

Roxy-3D, 2017-05-21:
> It should be some not so usual combination of conditions....

Its also possible I had a bad probe or a bad electrical connection to the probe.   It would seem unlikely.  But something definitely was wrong.    But I've been doing lots and lots of probes because I'm working on the mesh building and mesh editing routines...   And no problems so far with the new probe installed.

fbarcenas, 2017-05-22:
Try in 2 machines, several times with 1.1.x version and no problem

lrpirlet, 2017-05-24:
@Roxy-3D 
Let me bet... 
You have ```#define ENDSTOP_INTERRUPTS_FEATURE```...

I was not sure of the problem but it seems that since I disabled interrupt I do **not have anymore** the Z axes motor refusing to stop after the Z switch is pressed...
 ```//#define ENDSTOP_INTERRUPTS_FEATURE``` in the configuration.h seems to solve the issue

lrpirlet, 2017-05-24:
I have the bad feeling that we see interraction between several interrupts...

benlye, 2017-05-24:
My 2c, and another data point - I had this problem with ENDSTOP_INTERRUPTS enabled on RC8, so in my experience it is not a new behaviour.  

I found that around 1 in 3 G29s (bilinear, 7x7 grid) would send the nozzle into the bed while probing a point (the point was never the same) because the Z motors didn't stop when the probe was triggered.  I disabled ENDSTOP_INTERRUPTS and haven't had the problem again.  My printer is an Atmega1284p-based Sanguinololu clone, the Anet A8.

martindb, 2017-05-24:
I have endstop interrupts enabled.... and a sanguinololu clone... I'll disable this feature. Thanks for the tip.

Roxy-3D, 2017-05-24:
> Let me bet...
> You have #define ENDSTOP_INTERRUPTS_FEATURE...

Yes...

fiveangle, 2017-05-24:
But above you said it happened with and without ?

> > I assume you're using endstop interrupts?
> 
> I was... But I turned them off. And after a few more G28's, it happened again.

Roxy-3D, 2017-05-24:
Yes.   I've seen it both ways.    But mostly....   I've been running with the endstop interrupts turned on.

DarkEnceladus, 2017-05-24:
I tried to get the probing working the last week and really was puzzled, G30 and G29 worked and didn't work up do different levels.
Also the printer often stops after I try to execute G30 or G29. 
```
G30
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
Bed X: 0.00 Y: -19.00 Z: 0.00
```

Is that the error this issue is about? -> https://youtu.be/PVSiQxidACg

I also would like to say that the printer is not halted after trying to probe, it only halts when I try to execute G29 or G30 after it failed before or it halts on the first try and then doesn't even start to move.

I'm hoping for a few suggestions and am willing to try a few things and report! (I've already rewired 2 times because I wasn't able to find the error.

Roxy-3D, 2017-05-24:
@DarkEnceladus   What bed leveling system are you using?

DarkEnceladus, 2017-05-24:
I'm using 3D touch - which, I have to admit, I just noticed. So I am using a clone of the BL-touch.
The G29 command worked for 1 probing point, then 2 times for two points and one time 3 points before hitting the bed and then leaving "updwards" as if the probing points would sit on a stair.
Only one time today my probing was "successful" when it comes to probing all points. (ABL_GRID_POINTS_X = 5)
So far the reporting from M119 seemed correct and all "M280 P0 ..." commands worked  (S10, S90, S120, S160).
Manually operating the probe seems to deliver the expected behaviour as well.
Of course I cannot rule out the possibility that my probe "misbehaves" and that this is causing the problems. But the similarity within the behaviour, as far as I understood, let me think that it might be the same issue.

Roxy-3D, 2017-05-25:
What I meant was:  Which one of these are you using?
```cpp
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
```

DarkEnceladus, 2017-05-25:
I'm using 
`#define AUTO_BED_LEVELING_BILINEAR`
I'll try
`#define AUTO_BED_LEVELING_UBL`

jeno007, 2017-05-25:
Just to have more confirmation.
I had this happened twice in a row last week.
I'm using 1.1.1 with BLTouch and AUTO_BED_LEVELING_BILINEAR with ENDSTOP_INTERRUPTS enabled on a cartesian machine.
I ran a BLTouch test for a minute and it has not happened since then, for about a week now. I don't know if there were any correlation.
I'm visually checking all G29 since then, quite boring with a 9x9 grid.

DarkEnceladus, 2017-05-27:
With the  bugfix-1.1.x branch and UBL bed leveling I am no longer experience any problems with probing. I can actually start to probe from the full Z-height. When I send G30 it probes, G29 does not, but starting the process from the LCD on my printer it is working without errors or misbehaviour so far!

Roxy-3D, 2017-05-27:
> When I send G30 it probes, G29 does not, 

G29 P1 should....

martindb, 2017-06-01:
> I have endstop interrupts enabled.... and a sanguinololu clone... I'll disable this feature. Thanks for the tip.
Since I changed to no endstops interrupts, printed a lot of times, no more issues...

fiveangle, 2017-06-06:
I had to dig into the AT90USB1286  data sheet to figure out that not all of my and stops are interrupt-capable. If you're running with endstop interrupts on pins that are not interrupt capable, would all sorts of wackiness be possible ?

Roxy-3D, 2017-06-06:
> If you're running with endstop interrupts on pins that are not interrupt capable, would all sorts of wackiness be possible ?

I don't know what the symptoms would be.   But I would hope there are sanity checks to keep that from happening.    Supposedly, the RAMPS boards have all the endstops on interrupt-able pins.   That is what I have on all my printers so I haven't dug any deeper.

Bob-the-Kuhn, 2017-06-06:
You'll be running without endstops on the non-interrupt pins.  If they're involved with homing or probing then you'll get a crash.

Roxy-3D, 2017-06-07:
@Bob-the-Kuhn   I have a concern.   We need to know if interrupt enabled endstops is causing a problem.  But if people can turn on interrupts for their endstops even if the switches are on the wrong pins, it is going to be hard to get closure on this topic.

Can you use your Arduino Pin knowledge to put sanity checks in place?  Specifically...  If the user has turned on 
```cpp
// Enable this feature if all enabled endstop pins are interrupt-capable.
// This will remove the need to poll the interrupt pins, saving many CPU cycles.
#define ENDSTOP_INTERRUPTS_FEATURE
```
how hard is it to flag an error in SanityCheck.h if they have an endstop on a pin that can't support interrupts????

Bernardo1996, 2018-01-27:
Hello guys,
I have the same problem.
the nozzle crashes on the bed and I can not understand why, is there a solution?

https://www.youtube.com/watch?v=GDLeYqSkSZ8

Roxy-3D, 2018-01-28:
I have a couple of questions:   First, that isn't a Antclabs BL-Touch probe, right?   It looks different than that.

The other question is do you always have the problem at the same point in the probing?   Or does it happen at random places.    Because from watching the video, there doesn't seem to be any reason it should have a problem there....    At least it isn't driving the nozzle into the bed.   And the blinking LED error light is on so the firmware should notice the probe is in an 'error state'.

Bernardo1996, 2018-01-28:
it happens to me in random points. however, the problem is that the bltouch does not activate, ie the rod is not "attracted" and therefore there is no impulse automatically. what have I commented? I do not understand.
Bernardo Carnevale.

Bernardo1996, 2018-01-28:
anyway yes, it is anticlabs.
Bernardo Carnevale.

Bernardo1996, 2018-01-28:
I am forced to do only the G28 (95% of the time it works well), while doing also the G29 certainly the nozzle crashes into a point.

Bernardo1996, 2018-01-28:
however yes, it is anticlabs. (at least this is written on bltouch)

Roxy-3D, 2018-01-29:
When I watched the video...   I'm not sure this is really true...   But at 37 seconds into your video, when the probe is doing the 2nd touch...   The pin re-deploys and clicks the glass.   And it is the next Z-probe that fails.    The very next touch of the next Z-probe succeeds.   But the next Z-probe fails.

Can you capture the another failure for us so we can see if that pattern persists?    And it might be good to try a higher clearance number between probes.   That might give the pin more room.

[6 more replies]

closed 2018-02-02 by Bob-the-Kuhn