MarlinFirmware/Marlin #6808: PSU M80/81 command issue

shane84948, 2017-05-20:
Hello,

I have a Robo R1 plus and just installed a SSR to control my psu. I hooked the positive to the PSU on pin on my Ramps 1.4 and ground it to one of the AUX grounds. Everything works as expected except my commands are backwards.

I have enable the script for PSU 1 and also PSU off at start. When I connect to the printer my psu powers on where it should stay off. I issue the M80 command and my PSU powers down but the screen says that the printer is ready. When I use M81 the PSU powers up and my screen reads Printer off. For some reason everything is backwards. I am currently using version 1.1.1 but have tired this on older and stock versions. I am not sure if this is an issue with my ramps board or the script. 

Using a meter I checked the power from the PSU on pin and it reads 5v with the M81 and 0v with M80.

Any help with this would be greatly appreciated. I have looked through the Marlin sketch but do not want to change anything with proper direction. Can I just swap the state of the pin from HIGH to LOW with the corresponding M commands?

Bob-the-Kuhn, 2017-05-21:
Move your + lead to 5V and the - lead to the pin.  That'll reverse what your SSR sees.

shane84948, 2017-05-21:
Would that work if the psu on pin is configured as an output by default?

Bob-the-Kuhn, 2017-05-21:
The pins on the Atmel are tristate on power up so the PSU won't be triggered by accident.

I don't know what the default state is for that pin. 

M80 will turn on the SSR with the connections I described.

closed 2017-05-21 by shane84948