MarlinFirmware/Marlin #6818: Monoprice Ultimate / Wanhao Duplicator-6

Andrei-Pozolotin, 2017-05-21:
Hello guys.

What does it take to accept new configuration, i.e. : 

Monoprice Ultimate / Wanhao Duplicator-6
https://wanhaousa.com/products/duplicator-6
https://www.monoprice.com/product?p_id=15710

https://github.com/dot-bob/Marlin-Duplicator-6/tree/RC8/Marlin/example_configurations/Duplicator_6

Thanks.

Roxy-3D, 2017-05-21:
Mostly, it helps if it is a very popular machine.   And if it is fairly unique in how it is setup.

The big problem with adding new example machines is it becomes a big burden every time the Configuration.h files change.   Any simple change results in 30 files needing to be updated.  Please allow me to give you an example.  This morning I added (and changed) this in all of the Configuration.h files:
```cpp
  #define UBL_G26_MESH_VALIDATION   // Enable G26 mesh validation
  #define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle
```
Now...  As I'm putting the code in place it would have been wiser to  have a numeric value on the   UBL_MESH_EDIT_MOVES_Z  symbol to control the height.   But I didn't want to go back and edit 30+ files again.   So, I'm going to just use the Z_CLEARANCE_BETWEEN_PROBES as the base height.   That kind of compromise happens again and again because of the number of configuration.h files.

Andrei-Pozolotin, 2017-05-21:
ok, great, thank you for getting back. 

lets apply your logic as follows:

1) "**a very popular machine**"
looking at https://www.3dhubs.com/trends
and treating Monoprice Ultimate  / Wanhao Duplicator-6 / Zortrax M200 as the same
one can see that Wanhao printer made it last, but still made it those top-5 / top-10 lists

_is that popular enough?_

2) "**fairly unique in how it is setup**"

how can this be determined?

3) by staring at the board long enough?
[image]

4) by studying existing working wanhao patches to the marlin:
https://github.com/dot-bob/Marlin-Duplicator-6

_so, based on 3) and 4) - do you see how is it setup?_

5. "**any simple change results in 30+ files needing to be updated**"
that sounds like some long standing code restructuring, and now overdue?
it seems this issue is not "new-platform" specific, and should be addressed separately

6. now, based on the above, is it a) "yes", b) "no", or c) "need more info" - kind of answer?
if you need more information, what that would need to be to help you make your decision final?

Roxy-3D, 2017-05-21:
>> "any simple change results in 30+ files needing to be updated"
> that sounds like some long standing code restructuring, and now overdue? it seems this issue is not "new-platform" specific, and should be addressed separately

If we use a Configuration.h file set to personalize the firmware generation...   I'm not sure how code restructuring helps.  

MonoPrice Ultimate and Wanhao Duplicator would be reasonable machines to add other than my issue 5.   Let's see what the other code maintainers think.    (I'm really not that opposed to it...  I'm just pointing out why there is some resistance.)

dot-bob, 2017-05-22:
My plan is to eventually clean up and submit pull requests to add the remaining functionality needed to support the full hardware.  But again it all comes down to free time.  Currently Marlin supports most of the hardware and can be made functional modifying the configuration.h and configuration_adv.h.

In general to effectively replace the stock firmware the following still needs to be added: The ability to set PWM motor current from the display, save pwm motor current to eeprom, update the case light brightness from the display, save the case light brightness to eeprom, and add support for the PCA9632 to enable control of the RGB LED on the front panel.  These features are currently implemented in the community release of Marlin for the Duplicator 6.  It is just a matter of cleanup and submitting individual pull requests for each feature.

The core functionality and details for a few of these items are currently under active development ( see #6797 ) on Mainline and when they get hashed out I will bring them to the community release of Marlin for the Duplicator 6 until mainline has full hardware support.

jhitesma, 2017-05-22:
Instead of adding new sample configurations it's always possible for someone to maintain their own fork focused on a particular machine.  

Including a handful of common sample configurations makes sense to help people get started, but 30+ gets a little crazy to maintain.  And with the progress on a visual config tool the need for lots of samples is minimized.

Maybe instead of including a bunch of sample configs the Marlin website can provide a place to link to forks dedicated to various machines to help people find appropriate configurations.  Then it would become the responsibility of the person maintaining that particular fork to keep it up to date instead of burdening the core team with supporting a bunch of config files.

Roxy-3D, 2017-05-22:
> Maybe instead of including a bunch of sample configs the Marlin website can provide a place to link to forks dedicated to various machines to help people find appropriate configurations. Then it [...]

I like this idea.  But my guess is the links would die and need to be taken out of the list.

jhitesma, 2017-05-22:
Well, the links would be to github repos so they're more likely to go stale than just disappear.  But either way now that the site is hosted on github it's fairly easy for anyone to issue a PR to correct something they see wrong.  

So people who spot dead links would be able to submit an issue about them being dead or a PR to fix them.  And people who create forks with configs for specific machines could also create PR's to add themselves to the list or issues if they can't figure out how to do that.

Andrei-Pozolotin, 2017-05-23:
@Roxy-3D @jhitesma @dot-bob 
Hello again, guys:

I came up with the following idea:
https://github.com/random-builder/makaira-code
which should resolve most requests similar to mine, and:
* give users actual *.hex downloads
* automatically maintain firmware repository
* build daily against live and current marlin source
* provide consistent build environment as reference
* invite people with minimal programming skills to participate
* empower people to build and share and review multiple flavors 
* free marlin developers from maintenance of specific boards in this repo

Please let me know what you think? Thank you.

mperdue, 2017-05-23:
You guys maintain configuration files for popular machines? I've got a machine that I designed and built. I've got two of them and they are real popular with me. :-)

Andrei-Pozolotin, 2017-05-23:
@mperdue : 
sure, we maintain them here:
https://github.com/random-builder/makaira-code
because you maintain them :-)

mperdue, 2017-05-24:
Just messin' with you.

Andrei-Pozolotin, 2017-05-25:
@thinkyhead Scott: can you please comment on "build system" idea? Thanks, Andrei.

JustAnother1, 2017-05-25:
To help discussion here another Idea:

Restructure the configuration.h and configuration_adv.h files.

Split all the settings in small configuration files according to functionality. So that there might be a cfg_delta.h for the delta printer specific settings. a cfg_bedLevel.h for Bed leveling,..

The example configuration files would then be reduced to just including the cfg_*.h files that apply to the machine. Cartesian printers would not include the cfg_delta.h.

Special machine specific settings could then be done after the includes. That way these machine specific defines could overwrite the defines done in the previously included cfg_*.h files.

Maintenance would be drastically reduced, as the change described by @Roxy-3D would now only be needed in cfg_bedLevel.h. So one edit instead of 30+.

Setting up new machines would be easy as it is basically just copying over the includes from a similar machine and do the hopefully only few settings needed for what makes this printer unique.

Maintenance of these examples files could also be a task for the machine owners.

Just my 2c though.

Andrei-Pozolotin, 2017-05-25:
@JustAnother1 Lars: sounds interesting, thank you.

thinkyhead, 2018-02-06:
> can you please comment on "build system" idea?

Cool! Mind if I give it some publicity on Twitter and Facebook?

thinkyhead, 2018-03-14:
Please don't "+1" old issues. Thanks.

boelle, 2019-02-18:
@Andrei-Pozolotin did you get the configs added?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-12 by Andrei-Pozolotin