MarlinFirmware/Marlin #6819: BLTOUCH Heater Off

DaveAhrendt, 2017-05-21:
Has the BLTOUCH_HEATERS_OFF support been removed recently?  I do have it defined in configuration.h and it had been working.  I just noticed this morning with the most recent snapshot that the heater remained on during probing.  I just did a search for BLTOUCH_HEATERS_OFF and the only place I see it is in the configuration.h file and it's not referenced anywhere else.  I don't see a mention of it being removed.  Did I miss it?

bgort, 2017-05-21:
It's been renamed to PROBING_HEATERS_OFF; we've also added PROBING_FANS_OFF.

Roxy-3D, 2017-05-21:
> It's been renamed to PROBING_HEATERS_OFF; we've also added PROBING_FANS_OFF.

I wonder if we should we add a few lines to SanityCheck.h to watch for this?   I can do this if it is useful. 
 (I just added a few lines to check for G26_MESH_EDITING still being used).
```cpp
#elif defined(UBL_MESH_EDIT_ENABLED)
  #error "UBL_MESH_EDIT_ENABLED is now UBL_G26_MESH_VALIDATION. Please update your configuration."
#elif defined(UBL_MESH_EDITING)
  #error "UBL_MESH_EDITING is now UBL_G26_MESH_VALIDATION. Please update your configuration."
#elif defined(BEEPER)
```

DaveAhrendt, 2017-05-21:
Thanks! I missed the switch.  I guess I need to pay better attention.

bgort, 2017-05-21:
@Roxy-3D I don't see why we shouldn't do that...

Roxy-3D, 2017-05-21:
Done...
```cpp
#elif defined(BLTOUCH_HEATERS_OFF)
  #error "BLTOUCH_HEATERS_OFF is now PROBING_HEATERS_OFF. Please update your configuration."
```

closed 2018-02-06 by thinkyhead