MarlinFirmware/Marlin #6820: Rotary encoder on Reprap smart controller is not working

mikek6, 2017-05-21:
I use ramps 1.4 arduino mega shield with arduino mega and the full graphics discound smart controller. I uploaded to the controller the  bugfix 1.1.x  but the encoder still not working. when i push the encoder's button i get in to menu but the rotary funtion is dead.

Bergerac56, 2017-05-21:
I just tried on a "spare" ramps 1.4 with full graphic encoder flashing the latest bugfix 1.1.x and it is working fine. Are you sure: 1> that it was working with a previous build (issue with the encoder button?) and 2> that you choosed the correct option for the graphic controller ?

mikek6, 2017-05-21:
/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a Scara printer replace the configuration files with the files in the
// example_configurations/SCARA directory.
//
[... 1487 lines cut ...]
 *  34 RAMPS_14    : Analog input 5 on the AUX2 connector
 *  81 PRINTRBOARD : Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 RAMBO       : Analog input 3
 *
 * Note: May require analog pins to be defined for other boards.
 */
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00   // (mm) Diameter of the filament generally used (3.0 or 1.75mm), also used in the slicer. Used to validate sensor reading.

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0    // Index of the extruder that has the filament sensor (0,1,2,3)
  #define MEASUREMENT_DELAY_CM        14    // (cm) The distance from the filament sensor to the melting chamber

  #define MEASURED_UPPER_LIMIT         3.30 // (mm) Upper limit used to validate sensor reading
  #define MEASURED_LOWER_LIMIT         1.90 // (mm) Lower limit used to validate sensor reading
  #define MAX_MEASUREMENT_DELAY       20    // (bytes) Buffer size for stored measurements (1 byte per cm). Must be larger than MEASUREMENT_DELAY_CM.

  #define DEFAULT_MEASURED_FILAMENT_DIA DEFAULT_NOMINAL_FILAMENT_DIA // Set measured to nominal initially

  // Display filament width on the LCD status line. Status messages will expire after 5 seconds.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H

mikek6, 2017-05-21:
i dont see any mistake in configuration

Bergerac56, 2017-05-21:
I compiled with your configuration.h file and it works fine. No problem of rotary encoder. It has to be a hardware problem on your side I think

Bob-the-Kuhn, 2017-05-21:
The pins_RAMPS.h file recently changed.  Could this be the problem?

Bergerac56, 2017-05-22:
Based on the bugfix of the 21th of May, I flashed 2 different ramps based systems. One "real" 3D printer (with my own configuration.h file) and a test hardware (only electronic) with the configuration.h file of mikek6. Both are using a full graphic smart controller (but slightly different version) and were working fine with the release used. This means that I could not reproduce the potential bug from my side. Obviously, that does not exclude a potential problem...

@mikek6: could you reverse to an older version of the firmware and confirm that your hardware is working fine? So we would be sure that it is not a hardware problem?

pabm122, 2017-06-25:
Hi I have the same problem, and my encoder isn't the problem since it is working with repetier firmware, so I don't know how to make it work, since I want to use marlin because I bought a bltouch sensor, and I'm having troubles in repetier with that sensor, and there is no plenty information in the repetier side, so please let me know if someone is dealing with this problem too or if someone already found a solution!

fiveangle, 2017-06-25:
@mikek6 - might be good to close this issue ?

@pabm122 - might be good to open a new issue with all the relevant info so others can better help you ? http://marlinfw.org/docs/basics/reporting_bugs.html

Bob-the-Kuhn, 2017-06-25:
@pabm122 - when you open the new issue please zip up the system that works and the system that doesn't work & drop the ZIP onto that issue.

ronaldo1968, 2017-09-01:
I have the same problem, however, with "REPRAP_DISCOUNT_SMART_CONTROLLER" and not with "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" that are different. I compiled and installed the repetier firmware and it works perfectly, but not works with last marlin. I used the bugfix version today and the problem persists.

I have a folder shared with a video and the souces that I used.

Board: mega2560 with ramps 1.4 and reprap discount smart controller.

Thanks

https://drive.google.com/drive/folders/0B8pLU16dKZyzbUw2T3dHWEpqVEE?usp=sharing

ronaldo1968, 2017-09-02:
I solved the problem, it was due to a problem in the adapter plate that connects the cable to the lcd. I've switched it and it's working.
Thank you.

closed 2017-11-12 by thinkyhead