MarlinFirmware/Marlin #6824: Marlin-1.1.1 AutoBedLeveling

rv4, 2017-05-22:
I have a problem with erratic ABL results.

I looked at the results from a 3x3 G29 command and found out, that the left/front (collum 1 / row 3) and left/middle (collum1 / row 2) most times look strange. (values up to some thousand)

Examples:

08:32:41.119 : -0.03694 +0.09056 +0.07556
08:32:41.119 : +0.08306 +0.08806 -0.01944
08:32:41.119 : **-97.29734** -0.02694 -0.16944
08:32:41.119 : X:289.00 Y:185.00 Z:6.71 E:0.00 Count X:46240 Y:33300 Z:2680

10:59:25.753 : -0.07528 +0.01972 +0.03472
10:59:25.754 : +0.05472 +0.09222 -0.01028
10:59:25.758 : **+12.13330** +0.00972 -0.11528
10:59:25.762 : X:289.00 Y:185.00 Z:7.67 E:0.00 Count X:46240 Y:33300 Z:3060

11:01:15.610 : (0,0)
11:01:15.613 : -0.09389 +0.02361 +0.03611
11:01:15.613 : **+0.13111** +0.07861 -0.01889
11:01:15.618 : **+0.13111** +0.03611 -0.11389
11:01:15.622 : X:289.00 Y:184.99 Z:7.75 E:0.00 Count X:46240 Y:33300 Z:3060

Kind Regards,
Franz

FHeilmann, 2017-05-22:
What happens during probing for that point with the printer itself? Does everything seem normal?

rv4, 2017-05-22:
can't see any abnormal behavior during probing.

rv4, 2017-05-23:
I carried out some more tests in the meantime and found out, that above problem arises if the number of grid points is above 6.

2 x 2 works
2 x 3 works
3 x 2 works

3 x 3 fault
2 x 4 fault
4 x 2 fault

Kind Regards,
Franz

FHeilmann, 2017-05-23:
Do you have EEPROM enabled? There were some bugs in the EEPROM handling that got fixed recently, maybe your issue is also resolved with those fixes.

Load up the latest bugfix-1.1.x and flash it, then perform a

```
M502
M500
```

to load the values from your "Configuration.h" and store them in a safe manner.

rv4, 2017-05-23:
No improvement with the latest bugfix-1.1.x

11:51:43.530 : -0.07250 +0.01500 -0.00000
11:51:43.533 : **-2107340544.00000** +0.08500 -0.02000
11:51:43.533 : +0.13250 +0.04500 -0.12250

rv4, 2017-05-23:
I had AUTO_BED_LEVELING_LINEAR activated when I realized the problem.  With AUTO_BED_LEVELING_BILINEAR activated I don't see abnormal values.

Phaelz, 2017-05-25:
AUTO_BED_LEVELING_LINEAR is not working properly. I was having issues with a very uneven bed.

It isnt't even working. After changing to BILINEAR and maintaining the same settings everything worked fine.

closed 2018-02-07 by thinkyhead