MarlinFirmware/Marlin #6825: Error Z_MIN_ENDSTOP_INVERTING

h4nc, 2017-05-22:
Hi, 

I'm updating RC8 to 1.1.0 stable. I got a error while compiling that says:

SanityCheck.h:521: error: #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING."

#error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires Z_MIN_ENDSTOP_INVERTING to match Z_MIN_PROBE_ENDSTOP_INVERTING." //changed

Thing is if i set those parameters both to false, as the error says the probe goes in the wrong direction while probing.

When I set 

#define Z_MIN_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING false

as before in RC8 (without error) and comment the error section out in the sanitycheck.h file it works.

Is this a problem? Why should those parameters be the same?

Bob-the-Kuhn, 2017-05-22:
I can throw out a guess as to why but why bother.

The real question is: Why do you want them to be different?

h4nc, 2017-05-22:
If I set them both to the same value (false) the probe goes in the wrong direction while probing. It only works when I set it as I mentioned before.

bsabino, 2017-05-22:
By goes in the wrong direcrion do you mean that your bltouch is retracted? At list am having currently this issue.

h4nc, 2017-05-23:
I have fix mounted probe (inductive sensor NPN NO) and I mean the probe toghether with the hotend goes away from the bed and stops after some millimeters because the probe is triggered.

Bob-the-Kuhn, 2017-05-23:
That's the expected behavior.  After the probe triggers the head will move up in order to provide clearance for future X-Y movements.  The amount it moves is set by Z_CLEARANCE_BETWEEN_PROBES.  

Could be that it goes to Z0.00 + Z_CLEARANCE_BETWEEN_PROBES.

h4nc, 2017-05-24:
Ok thank you, so the problem is that it is triggered without "touching" the bed. So I have to invert the probe reading. Where can I find this?

Bob-the-Kuhn, 2017-05-24:
It's the Z_MIN_ENDSTOP_INVERTING and Z_MIN_PROBE_ENDSTOP_INVERTING.  They both need to be set to false for a BLTouch.

After doing that you may want to use the **M43 S** command to check if the BLTouch is setup correctly.  It'll even tell you if you have a BLTouch clone.  The utility can't tell the difference between a real BLTouch and the Geeetech 3DTouch.

To use the command you'll need to enable PINS_DEBUGGING in Configuration_adv.h

h4nc, 2017-05-24:
I don't have a BLtouch. I have a inductive Probe NPN NO.

Bob-the-Kuhn, 2017-05-24:
Never mind ...

Set them both to the same, move the head so it's far enough away from the bed that it won't be triggered and then issue M119.  If M119 says the probe is triggered than you guessed wrong and the two need to be flipped the other way.

Once you get it to say open then test it by bringing a piece of metal close to the probe.  If it still says open then we need to do some trouble shooting.

Past my bed time.  Talk to you tomorrow.

Tannoo, 2017-05-24:
If `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`  is enabled, you cannot have the inverts different. That would be conflicting.

Tannoo, 2017-05-24:
If `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN ` is enabled, the probe becomes the probe AND the endstop.

So, if you invert the probe, you also need to invert the endstop.

As the error from sanity check says, they need to be the same.. inverted or not.

Tannoo, 2017-05-24:
An NPN NO switch is low when it's on. So, the Marlin logic needs to invert it to be recognized as triggered.

Also, to keep that input stable, either an external resistor needs to be applied or enable the probe endstop pullup.

h4nc, 2017-05-24:
Thx for your help. 
I had this settings in RC8 and they worked fine. Your explanation makes sense to me now.

Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is enabled, because I want to use the probe as endstop. 

A external resistor is applied to the probe, because the output would be above 5V and would potentially fry my ramps board. 

To be more specific: I have a Prusa i3 Hephestos with some modifications. 
I attached my configuration.h file.

When I set both to false. The probe is triggered without touching the bed and goes up like bob-the-Kuhn explained.

Isn't it strange that it works when i comment the sanitycheck out and set them both to false and true?

Tannoo, 2017-05-24:
It's best if the sanity file is left alone.

If it stays triggered with both set to false, then set both to true and it should work correctly.

Tannoo, 2017-05-24:
Btw, my sensor is 12vdc and the output is straight to the ramps... no external resistor needed when using the internal pullup.

Tannoo, 2017-05-24:
I am powering from the same 12v supply as the Arduino/ramps is.

h4nc, 2017-05-24:
Thx it works now. I set both to true. I was to confused that is worked with RC8.

closed 2017-05-24 by h4nc