MarlinFirmware/Marlin #6826: Z_DUAL_ENDSTOPS_ADJUSTMENT

remco1123, 2017-05-22:
The weight of the X-axis motor is enough to slightly bend the axis. with Z_DUAL_ENDSTOPS_ADJUSTMENT  i tried to add enough to compensate but nothing happend.
In my setup i use 2 endstops underneath the Z carrier so when the nozzle hits the bed they will continue till the Z carrier and the endstop carrier are separated.

sorry if this is the wrong place to this issue.

vMeph, 2017-05-22:
@Bob-the-Kuhn  sorry to jump in this issue but im also in progress of setting my endstps on Z axis,in my case the bed moves up and down with 2 steper motors nema 17 at this moment i dont know yet if the 2 stepers will be enought to suport a  base with 600 by 450 mm plus a bed size 400 by 300mm aluminum plates of 6mm,im i process of getting them, i been setting my endstops of z to figuer out how all works, sense the bed will be the z i assume the endstops will be on bottom of bed and will be pluged on Zmax pin on ramps so when homes will go to max travel of z axis that i have set in my case is 400mm, and when starts a job will go to Z 0 position.
Now i was wondering about Z_DUAL_ENDSTOPS_ADJUSTMENT, how do you set the 2 switches for Z_DUAL_ENDSTOPS sense i only see 1 Z max pin on ramps board?
Could let me know if  the way im assuming all works is correct and how to connect the 2 Z endstops?

Thanks
Sorry for my english

Bob-the-Kuhn, 2017-05-22:
Your English is better than some of my relatives'.

I'm using the Z_MIN and Z_MAX endstops for my Z dual endstops and one of the AUX pins for my Z probe.

Here's my dual Z configuration:
```
  #define Z_DUAL_STEPPER_DRIVERS

  #if ENABLED(Z_DUAL_STEPPER_DRIVERS)
...
    #define Z_DUAL_ENDSTOPS

    #if ENABLED(Z_DUAL_ENDSTOPS)
      #define Z2_USE_ENDSTOP _ZMIN_
      #define Z_DUAL_ENDSTOPS_ADJUSTMENT -11  // use M666 command to determine/test this value
    #endif

  #endif // Z_DUAL_STEPPER_DRIVERS
```

I just found a note in my pins_xxxx.h file that indicated I needed to swap the Z_MIN and Z_MAX cables to get the Z dual endstops to work properly.  You may or may not need to do this.

----

There's a Marlin utility I like to use when setting endstops.

If you enable PINS_DEBUGGING in Configuration_adv.h and then issue the M43 E1 command you'll enable this utility.

What does is sample the endstops and report changes to the console.  It also toggles the system LED.  Unfortunately on Ramps it's hard to see because it's on the mega board near the USB connector.

remco1123, 2017-05-22:
#define Z_DUAL_STEPPER_DRIVERS

#if ENABLED(Z_DUAL_STEPPER_DRIVERS)
......
  #define Z_DUAL_ENDSTOPS

  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _ZMAX_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  6 (nothing happens with this)  // use M666 command to determine/test this value
  #endif
--------------------------------------------------------------------

i don't use a probe
M119
x_min: open
y_max: open
z_min: open
z2_min: open

This is my Z endstop Z2 is the same on the other side
[image]

Bob-the-Kuhn, 2017-05-22:
It should be _ZMAX_

vMeph, 2017-05-23:
@Bob-the-Kuhn  i have test it last night, i wasnt able to make it work,i have set the driver on the E1 for the Z2 motor, motors move but only 1 switch works, something im missing?

remco1123, 2017-05-23:
@Bob-the-Kuhn Do you know if there is a way to add a Z1 and a Z2 offset.
Z_DUEL_ENDSTOPS_ADJUSTMENT isn't working in my case.
my Z1 or Z2 is like 0.6mm lower ( i switched them so much i'm not sure anymore what Z1 is )

Bob-the-Kuhn, 2017-05-23:
@remco1123 - there's only one adjustment. 

I expect that you have the EEPROM enabled and have forgotten to do the M502, M500, M501 sequence to make your config changes active.  

To be honest with you I think that the 0.6mm difference is great.  I don't see a need for an adjustment.  Let the auto bed leveling take care of that.  

If you want that last 0.6mm then you'll have to play with the stepper and endstop cables.  I don't remember what affect swapping one vs. the other has.  I would suggest making the offset large so it's easy to see which side is getting adjusted.

Bob-the-Kuhn, 2017-05-23:
@vMeph - have you tried swapping the motor and/or the endstop cables?

remco1123, 2017-05-23:
@Bob-the-Kuhn The weight of the X motor is preventing a good G29 auto level. The X-axis bends slightly on the motor end.
i like this design because you don't have to mess with screws
[image]

vMeph, 2017-05-23:
@Bob-the-Kuhn  got it to work apreciate all the help mate
forgot to uncoment the #define USE_ZMIN_PLUG :)

this feature of having 2 Z end stops is very usefull, what sux is that sense i have ramps1.4 board i have to use the E1 driver to add the Z2 this will then not let me have a second extrude :(

was wondering what board you using
i was looking into RUMBA board it has 3 slots for extruders, that would be perfect for the situation that you wanna use 2 extruders and also i like the idea that you can use volatages of 24V, would be perfect to add extra power to the steppers, no idea if RUMBA board is any good

what board you using?

Bob-the-Kuhn, 2017-05-23:
GREAT!!

I'm using an Azteeg X3 pro.  I'm running dual Y and dual Z steppers.  My printer started out life as a CNC machine so I went for power and strength.  Power & strength means heavy which makes for a slow printer.

Yes, 24V will allow you to run your current system faster.

For the most part the quality of a board is determined by the quality of the components used.  Unfortunately there's very little info available on that.  Sometimes you can infer that from the specifications.  If your board is rated for 24V & no higher then chances are the components are just rated for 24V or 25V.  If you run 24V on a board rated for 28V then you know you have some margin.

I'm hoping that within the next year we start seeing affordable 32 bit boards.  I'd really like to have an Ethernet or Bluetooth based controller so I can get rid of my USB cable.

closed 2017-05-27 by remco1123

thinkyhead, 2017-06-07:
I found the bug that broke `Z_DUAL_ENDSTOPS_ADJUSTMENT`. The `Z_DUAL_ENDSTOPS_ADJUSTMENT` value wasn't being put into the global variable because it was concealed by a local variable having the same name. A patch will be merged shortly.