MarlinFirmware/Marlin #6829: Unexpected Z probe height

stevewatts, 2017-05-22:
Hi I've recently nearly-completed a Marlin 1.6.12 Ramps1.4 which I'm trying to use as a CNC for milling pCBs.

I have a normally open Z Min whih doubles up as a a probe using a crocodile clip on the drill bit.  
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

This homws perfectly on the PCB and I can do some milling with it running G-code through.

My boards are not flat so not getting brilliand results so I've been investigating auto-homing.  In order to  configure autohoming I need to select a probe.  the closest I can think of to what I have is:
#define FIX_MOUNTED_PROBE 

Now.  making only that setting, and not even turning on one of the leveling features causes the probe to happily lower down to the base, make contact, but then raise 5mm up and stay there.  The disply shows that the bit is 5mm above home, but then when running G-Code the bit fails to lower back to home before doing its thing.

in addition, although it reports the bit is at +5mm, manually lowering it takes about 5.5mm to get back to touching the surface.

Has anyone got any idea how to set the probe as 'existing zMin'?  or the reason why it would hover 5mm above the surface but not return down to the GCode G90 position?

Thanks in advance
Steve

Roxy-3D, 2017-05-22:
The height is wrong for this explanation...  But do you have the MIN_SOFTWARE_ENDSTOPS turned on.  That will keep you from going below the bed.  And that needs to be turned of for the Mesh Bed Leveling systems because they often times need to dip below the 0.000 point as they track the bed.

```cpp
// If enabled, axes won't move below MIN_POS in response to movement commands.
//#define MIN_SOFTWARE_ENDSTOPS
// If enabled, axes won't move above MAX_POS in response to movement commands.
#define MAX_SOFTWARE_ENDSTOPS
```

stevewatts, 2017-05-22:
Thanks for the quick reply Roxy.

Yes I do have that set, but allowing -1.4 to allow for the full depth of a PCB when drilling:
#define max_software_endstops true  // If true, axis won't move to coordinates greater than the defined lengths below.

// @section machine

// Travel limits after homing (units are in mm)
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS -1.4  //sjw

So the whole thing works and mills OKish when no probe type is defined.

In addition, I didn't put this earlier to keep the story simple:
when I do select one of the bed levelling modes, the probe drops to the bed and appears to probe correctly (not sure how to determine if its right), but when compete leaves the bit at about z+5 again.  milling again takes place up in the air at +5ish.  
Gcode:
G90
G00 X0.0000 Y0.0000 
M03
G04 P3.000000
G00 Z1.5400 
G00 X3.3104 Y7.7779 
G01 Z-0.0800 F30.00 
G01 X3.3192 Y7.8106 F20.00 
G01 X3.3361 Y7.8400 
G01 X3.3600 Y7.8639

The first z1.5  is because the bit is expected to be on the board following the Z homing action.

Roxy-3D, 2017-05-22:
Is there any similarity to what you are seeing and these values:
```cpp
#define Z_CLEARANCE_DEPLOY_PROBE   15 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  6 // Z Clearance between probe points
```

stevewatts, 2017-05-22:
I did wonder about that, so dropped them both down assuming they were an offset of some kind:
#define Z_CLEARANCE_DEPLOY_PROBE   0 // Z Clearance for Deploy/Stow        sjw 10
#define Z_CLEARANCE_BETWEEN_PROBES  0 // Z Clearance between probe points     sjw 5

stevewatts, 2017-06-01:
Sorry i stupidly read the Arduino version...  I was using MarlinRC that time, I've since tested on 1.1.0 and today having the same issues wth 1.1.2.  So it all works as expected (without auto bed leveling) until the fix_mounted_probe is exposed.  tehn it hovers the bit above the workpiece as though that is the new z=0 eventhough it successfully finds z=0 during the homing process.

Is anyone able to help understand this behaviour?  

/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010100

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer replace the configuration files with the files in the
// example_configurations/delta directory.
//

//===========================================================================
//============================= SCARA Printer ===============================
[... 1493 lines cut ...]
 *  34 RAMPS_14    : Analog input 5 on the AUX2 connector
 *  81 PRINTRBOARD : Analog input 2 on the Exp1 connector (version B,C,D,E)
 * 301 RAMBO       : Analog input 3
 *
 * Note: May require analog pins to be defined for other boards.
 */
//#define FILAMENT_WIDTH_SENSOR

#define DEFAULT_NOMINAL_FILAMENT_DIA 3.00   // (mm) Diameter of the filament generally used (3.0 or 1.75mm), also used in the slicer. Used to validate sensor reading.

#if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0    // Index of the extruder that has the filament sensor (0,1,2,3)
  #define MEASUREMENT_DELAY_CM        14    // (cm) The distance from the filament sensor to the melting chamber

  #define MEASURED_UPPER_LIMIT         3.30 // (mm) Upper limit used to validate sensor reading
  #define MEASURED_LOWER_LIMIT         1.90 // (mm) Lower limit used to validate sensor reading
  #define MAX_MEASUREMENT_DELAY       20    // (bytes) Buffer size for stored measurements (1 byte per cm). Must be larger than MEASUREMENT_DELAY_CM.

  #define DEFAULT_MEASURED_FILAMENT_DIA DEFAULT_NOMINAL_FILAMENT_DIA // Set measured to nominal initially

  // Display filament width on the LCD status line. Status messages will expire after 5 seconds.
  //#define FILAMENT_LCD_DISPLAY
#endif

#endif // CONFIGURATION_H

stevewatts, 2017-06-05:
I've found the cause of this unexpected probe raise, but don't understand why it is doing it.

I had allowed the head to travel down below the homing height because I've actually got a drill attached and want to mill.  so for this I thought that  #define Z_MIN_POS -1.5  would provide enough depth I need.

So #define Z_MIN_POS -1.5 works fine, and allwas the head to pass below z=0, but when a probe is selected (tried both manual probe and fix_mounted_probe) this setting causes the head to actually raise by 1.5mm fter its completed homing.

Is anyone able to explain why it does this pelase?

Roxy-3D, 2017-06-06:
> So #define Z_MIN_POS -1.5 works fine, and allows the head to pass below z=0, but when a probe is selected (tried both manual probe and fix_mounted_probe) this setting causes the head to actually raise by 1.5mm fter its completed homing.

> Is anyone able to explain why it does this please?

For starters...  setting:
```cpp
#define MIN_SOFTWARE_ENDSTOPS
// If enabled, axes won't move above MAX_POS in response to movement commands.
#define MAX_SOFTWARE_ENDSTOPS
```
Means the nozzle can not go below 0.000

stevewatts, 2017-06-06:
Not sure I follow you.  I've set min_pos to -1.5 assuming this would allow it to go 1.5mm below home (z=0). Allowing me to drill full depth.
What actually happens is after probing, z=0 actually ends up being 1.5mm above the workpiece.

stevewatts, 2017-06-06:
But also.  Why would that behaviour change because a probe is selected?

stevewatts, 2017-06-11:
Process:
From LCD set auto home
using z- control, lower probe (drill bit) back to workpiece so it reads z=0
start GCode (using CNC-GCode-Controller)

Scenario 1:

#define Z_PROBE_OFFSET_RANGE_MIN 0

// If enabled, axes won't move below MIN_POS in response to movement commands.
//#define MIN_SOFTWARE_ENDSTOPS
// If enabled, axes won't move above MAX_POS in response to movement commands.
#define MAX_SOFTWARE_ENDSTOPS

stops 2mm above  
zmove reports +2

Will not go below z=0, but reports that it is at z=-0.08

-----------------------------------
Scenario 2:

#define Z_PROBE_OFFSET_RANGE_MIN -2 

// If enabled, axes won't move below MIN_POS in response to movement commands.
//#define MIN_SOFTWARE_ENDSTOPS
// If enabled, axes won't move above MAX_POS in response to movement commands.
#define MAX_SOFTWARE_ENDSTOPS

stops 3.5mm above workpiece 
Zmove reports z=0 at 1.5 above

Will not go below z=0, but reports that it is at z=-0.08

----------------------

Scenario 3:

// If enabled, axes won't move below MIN_POS in response to movement commands.
//#define MIN_SOFTWARE_ENDSTOPS     
// If enabled, axes won't move above MAX_POS in response to movement commands.
//#define MAX_SOFTWARE_ENDSTOPS

stops 2mm above  
zmove reports +2 at stopping point.  actually 2mm above

Will not go below z=0, but reports that it is at z=-0.08

Gcode runs, but -0.08 when running shows as .07 on the display, which in reality is about .1 above the workpiece.  This is expected to be within the workpiece (below the probed height for home)

G21
(Absolute Coordinates)
G90
G00 X0.0000 Y0.0000 
M03
G04 P3.000000
G00 Z1.5400 
G00 X3.3104 Y7.7779 
G01 Z-0.0800 F30.00 
G01 X3.3192 Y7.8106 F20.00 
.
.
.

----
Differences from default:
Line Line 130:      #define MOTHERBOARD 33    //sjw  supposedly allows M106/M107  to start and stop fan, this is used for milling drill instead  http://www.buildlog.net/blog/2012/08/hacking-your-cnc-toolchain/
	Line 443: #define USE_XMAX_PLUG  //sjw
	Line 444: #define USE_YMAX_PLUG  //sjw
	Line 445: #define USE_ZMAX_PLUG  //sjw
	Line 464: #define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.     //sjw
	Line 467: #define Z_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.      //sjw
	Line 468: #define Z_MIN_PROBE_ENDSTOP_INVERTING true // set to true to invert the logic of the probe.   //sjw
	Line 472: #define ENDSTOP_INTERRUPTS_FEATURE    //sjw
	Line 499: #define DEFAULT_AXIS_STEPS_PER_UNIT   { 800, 800, 800, 500 }  //sjw { 80, 80, 4000, 500 }
	Line 514: #define DEFAULT_MAX_ACCELERATION      { 80, 80, 30, 10000 }  //sjw{ 3000, 3000, 100, 10000 }
	Line 526: #define DEFAULT_TRAVEL_ACCELERATION   100  //sjw  3000    // X, Y, Z acceleration for travel (non printing) moves
	Line 577: //#define Z_MIN_PROBE_ENDSTOP    //sjw
	Line 578: //#define  Z_MIN_PROBE_PIN  //Z_MAX_PIN    //sjw
	Line 598: #define FIX_MOUNTED_PROBE  //sjw
	Line 684: #define Z_CLEARANCE_DEPLOY_PROBE   0 // Z Clearance for Deploy/Stow    //sjw
	Line 685: #define Z_CLEARANCE_BETWEEN_PROBES  2 // Z Clearance between probe points  //sjw  was 5
	Line 688: #define Z_PROBE_OFFSET_RANGE_MIN 0    //-20  sjw
	Line 689: #define Z_PROBE_OFFSET_RANGE_MAX 0   //20    sjw
	Line 719: #define INVERT_Z_DIR true  //sjw  false
	Line 735: #define Z_HOMING_HEIGHT 4  // sjw (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
	Line 749: #define Z_MIN_POS 0  //sjw  0    - when 0 allows the final home position to be where probed without moving upwards, but then the drill is not able to plunge into the material ideally needs to be -1.5
	Line 750: #define X_MAX_POS 160  //sjw  200
	Line 751: #define Y_MAX_POS 80  //sjw  200
	Line 752: #define Z_MAX_POS 25  //sjw  200
	Line 820: //#define AUTO_BED_LEVELING_BILINEAR  //sjw
	Line 1184: #define ULTRA_LCD   // Character based    //sjw
	Line 1194: #define SDSUPPORT     //sjw
	Line 1202: //#define SPI_SPEED SPI_HALF_SPEED    //sjw
	Line 1219: #define ENCODER_PULSES_PER_STEP 4  //sjw
	Line 1257: //#define INDIVIDUAL_AXIS_HOMING_MENU   //sjw
	Line 1338: #define REPRAP_DISCOUNT_SMART_CONTROLLER   //sjw

thinkyhead, 2017-11-03:
Sorry this issue got abandoned. Did you ever get it resolved?

closed 2017-11-03 by thinkyhead