MarlinFirmware/Marlin #6832: Error compiling marlin 1.1 on teensyduino board

adasko1, 2017-05-23:
Hi, first of all Im really a beginner in this kind of things. Im trying to compile marlin 1.1 on teensy board with standard 4x20 HD44780 lcd

On Marlin 1.1.x. keep getting 'customizedserial' was not declared in this scope.

On Marlin bugfix-1.1.x and older Marlin 1.0.2, when I only change the board for teensylu i can succeed compiling, but the lcd doesnt work. 
If I enable #define ULTRA_LCD keep getting 'LCD_PINS_RS' was not declared in this scope.

On Marlin bugfix-1.1.x, if I enable #define DOGLCD compiling is successful.

FHeilmann, 2017-05-23:
Teensy is a 32bit platform which is not yet ready for prime-time in Marlin. If you want to aid in development and fixup of the 32-bit port (which includes teensy) then you can submit pull-requests. Otherwise, if you want to get up and running with your printer, you should look at other firmwares that fully support teensy or get an Arduino MEGA based controlboard.

teemuatlut, 2017-05-23:
I don't think he's talking about the 32bit Teensies.

adasko1, 2017-05-23:
The chip is at90usb1286 so i suppose its 8bit?

FHeilmann, 2017-05-23:
my mistake, sorry.

fiveangle, 2017-05-23:
you may want to review #6737

adasko1, 2017-05-23:
Tried it yesterday, doesnt work

Bob-the-Kuhn, 2017-05-23:
If you supply the name of your display and how it's hooked up to your Teensylu, I'll try to get a new pins_TEENSYLU.h file late today or tomorrow.

Please fill out the attached Excel file with the hookup info.
[Teensylu_pinout.zip](https://github.com/MarlinFirmware/Marlin/files/1023271/Teensylu_pinout.zip)

adasko1, 2017-05-23:
Pins on my board look different. I will provide the connection scheme between the board and lcd as soon as I come home

[image]
[image]

[image]

Bob-the-Kuhn, 2017-05-23:
I haven't seen a 1286 based controller like this one.  Glad to add it to my library.

I'll need pointers to the schematic for the controller and the LCD.

adasko1, 2017-05-23:
Damn Im sorry, I was wrong with the chip it is 1287, the schematic is added in previous post

Bob-the-Kuhn, 2017-05-23:
1287 - host capable version of the 1286.  Does it use this capability?

I don't see the schematic in this thread.

adasko1, 2017-05-24:
As far as I know it doesnt use host capability.
Isnt it the schematic you need?
[image]

Bob-the-Kuhn, 2017-05-24:
Very nice hookup diagram.  What's missing is the logical pin mapping.

Can you point me to the site where you got this info?  Maybe the info I want is there.

What's the name of your controller?

I see something in the pins.h file from your old software that's encouraging.  There's a chance that combining the new & the old will work.  Don't bet the farm on it.

Worst case is we find out how good a technician you are and how patient you are. Marlin has some utilities that can help identify the logical pin mapping.  Unfortunately the Teensy world has some unusual twists so I'm hoping we don't need to go down that path.

I'd like to see your configuration.h & configuration_adv.h files you're using with 1.1.x

adasko1, 2017-05-24:
The board is Unique one rev a, what is a small production board. All info I have is from Polish forum, there is no other schemes or files. I can contact the producer and ask him for more info about the board. 

Im more like a mechanical guy then a programming one :/ , though Im really patient.

Bob-the-Kuhn, 2017-05-24:
Try this file & see what happens. [pins_TEENSLU.zip](https://github.com/MarlinFirmware/Marlin/files/1027389/pins_TEENSLU.zip)

If that doesn't work then we'll need to use some of the utilities to determine the logical pins.

You might also try enabling just REPRAP_DISCOUNT_SMART_CONTROLLER.  The pictures look very similar.

adasko1, 2017-05-25:
Ok, I still dont have the answer from the producer, though i did some testing today. With your pins_TEENSYLU I could compile the firmware, But the lcd didnt really work two rows were black and no any text. By the host software I could move the axis, but when i didnt disable motors, on x and y axis it slowly keeps going, ticking like a clock. 

Later I tried to put pin numbers for LCD from the original firmware to Your file and now the LCD works fine, but I cant enter to menu. Using host software, heating works fine, X axis doesnt work, Y and Z axis seem to work. Files:
[pins_TEENSYLU.zip](https://github.com/MarlinFirmware/Marlin/files/1029241/pins_TEENSYLU.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1029242/Configuration.zip)

No changes in configuration_adv.h

Bob-the-Kuhn, 2017-05-25:
On the X axis issue - the file has X_STEP and LCD_PINS_D4 both attached to logical pin zero.  Try commenting out LCD_PINS_D4 .

----

On the encoder button won't work problem - try these values.
```
    //buttons are directly attached
    #define BTN_EN1 29  //5
    #define BTN_EN2 31  //7
    #define BTN_ENC 7   //39  //the click
    
    //#define BLEN_C 2
    //#define BLEN_B 1
    //#define BLEN_A 0
```

If this doesn't get the encoder press button to work then we'll need to identify the logical pin that the ENC C pin is attached to.   To do that that you'll need to attach an LED to the pin in question and run a utility that pulses every pin.

1. Enable PINS_DEBUGGING in Configuration.h
2. Attach an LED between +5V and ground.
3. Once the LED is on then move the + lead to ENC C
4. Enter this command **M43 T I1 R5**. This will toggle each pin in sequence.  
5. The pin number you get from this needs to be translated from Teensy numbering to FASTIO numbering.  Use the following table to do the translation.

Might as well check the ENC 1 and ENC 2 while you're at it.

```
//  Marlin  Teensy             Marlin  Teensy  
//  0       28                  24      0       
//  1       29                  25      1       
//  2       30                  26      2       
//  3       31                  27      3       
//  4       32                  28      4       
//  5       33                  29      5       
//  6       34                  30      6       
//  7       35                  31      7       
//  8       20                  32      8       
//  9       21                  33      9       
//  10      22                  16      10      
//  11      23                  17      11      
//  12      24                  18      12      
//  13      25                  19      13      
//  14      26                  20      14      
//  15      27                  21      15      
//  16      10                  22      16      
//  17      11                  23      17      
//  18      12                  38      18      
//  19      13                  39      19      
//  20      14                  8       20      
//  21      15                  9       21      
//  22      16                  10      22      
//  23      17                  11      23      
//  24      0                   12      24      
//  25      1                   13      25      
//  26      2                   14      26      
//  27      3                   15      27      
//  28      4                   0       28      
//  29      5                   1       29      
//  30      6                   2       30      
//  31      7                   3       31      
//  32      8                   4       32      
//  33      9                   5       33      
//  34      *                   6       34      
//  35      *                   7       35      
//  36      36                  36      36      
//  37      37                  37      37      
//  38      18                  40      38      
//  39      19                  41      39      
//  40      38                  42      40      
//  41      39                  43      41      
//  42      40                  44      42      
//  43      41                  45      43      
//  44      42                  46      44      
//  45      43                  47      45      
//  46      44                  34      *       
//  47      45                  35      * 
//
//     * not assigned in Teensyduino but is available in Marlin FASTIO
```

adasko1, 2017-05-25:
Ok, for now I figured out that:
1. When i commented out LCD_PINS_D4 I got error with it when compiling, so i turned off the LCD and all axis work fine so maybe the pin number in LCD should be changed?
2. Then I tried changing the encoder pin numbers for what you wrote. Turning the encoder works (the FR value changing), but the click doesnt work.

I will try to do this pins debugging today, hope it works

adasko1, 2017-05-25:
Ok this is not good, I think I made a short circut between 5V and some different pin trying to connect led, because now my computer cant find the board in any host and flip. Is there any way to check if it is fried beside just looking if something is damaged?

Bob-the-Kuhn, 2017-05-25:
OUCH!!!

By chance do you have something that connects to the ISP connector that can be used to burn a new bootloader?

Do you have +5V and +12V?

Have you tried powering it from just the USB port?

adasko1, 2017-05-25:
I dont really know about your first question. The board has a power from usb and 12V, but the LCD doesnt display text anymore (just 2 black rows and 2 white).

Bob-the-Kuhn, 2017-05-25:
It'll only cost $30 to get a 2560 Arduino + RAMPS + RepRap graphics display & it'll be there in a week.  http://www.ebay.com/itm/302268617535

adasko1, 2017-05-26:
I know ill probably do that, the thing is should i buy  arduino+ ramps or maybe mks gen 1,4?

Bob-the-Kuhn, 2017-05-26:
I don't have a good reason for disliking MKS.  It's probably because they do a poor job of documenting their products.  

I haven't heard anything negative about their quality or functionality.  An internet search is probably the better way of looking into it.

A single board solution is inherently more reliable than a 2 or 3 board solution like a RAMPS system.

When looking for a board consider the LCD controller you'll be using.  If the LCD has the two 10 pin cable interface then look for a controller that has the two connectors on it.

adasko1, 2017-05-26:
Is this the ISP connector you asked for?
[image]

Bob-the-Kuhn, 2017-05-26:
Yes.  If you have something like the USBtinyISP then you can program the CPU without going through the USB port.  In fact you can remove all the cables from the controller and power it via the USBtinyISP.

I have one just because I've shot myself in the foot enough times to warrant it.

adasko1, 2017-05-26:
I think I will buy another board and in the meantime, buy the isp programmer trying to unbrick the teensy one. 
The question is what are the chances that the board will work? Do You know where I can find a new bootloader for at90usb1827?

Bob-the-Kuhn, 2017-05-26:
I believe all you need to do for the bootloader is select the board in the Arduino IDE, select USBtinyISP in the programmer list and then click Burn bootloader. The bootloader that comes with the Teensyduino system will be programmed into you board.  I don't think you need to select a port.  Arduino will automatically find the programmer.

It always complains that there is a data mismatch.  Ignore that, reconnect the USB cable & see if you can download Marlin.

I'm using this style of [programmer](http://www.ebay.com/itm/USBTiny-USBtinyISP-AVR-ISP-Programmer-6-10-Pin-Bootloader-for-Arduino-UNO-MEGA-/191821167214?hash=item2ca96eba6e:g:hUgAAOSwTA9X4Oxq).  I see other styles out there.  I have no info on them.

What are the chances it'll work?  Not good.  I shorted an analog input to +24V on my 2560 & now I only have 3 or 4 useable analog ports.

Roxy-3D, 2017-05-26:
> What are the chances it'll work? Not good. I shorted an analog input to +24V on my 2560 & now I only have 3 or 4 useable analog ports.

I'm not sure what I did, but I think I bumped a 12 v line onto my E0 Enable line.   So my extruder quit working.  But other than that...  My Atmega2560 board works.  I did move the extruder to E1 and it worked.    So...  I pulled that processor board and just use it for simple code and have it clearly labeled as a not fully functional board.   It is possible to blow out the various output pins but not damage the rest of the processor.

fiveangle, 2017-06-06:
Hard to believe it would be possible to blow out the USB connection that built-in to the MCU by accidentally frying some other input. And even more crazy to thinking frying an input might erase PROGMEM. The MCU is likely toast.

I'd definitely jump to a board with 256k. I'd stay Way from Makerbase as they're always the last to get support (because they provide zero documentation).

closed 2018-02-07 by thinkyhead