MarlinFirmware/Marlin #6834: Problem with enqueue_and_echo_commands_P();

JanVelleman, 2017-05-23:
Hi,

I'm adding 2 custom menu items (load and unload filament) and in the past I just added these as Gcode commands in a function.

With the latest version I'm using the enqueue_and_echo_commands_P(); function as follows:
```
static void lcd_load_menu_PLA_go()
  {
    lcd_return_to_status();
    //enqueue_and_echo_commands_P(PSTR("G28\nM109 S200\nG92 E0\nM83\nG1 E640 F8000\nG1 E60 F200\nM82\nM104 S0"));
  }
```
So this sequence is just homing, heating, moving and then cooling again.

The problem now is that after the nozzle has reached 200°C the extruder motor starts turning as expected but the last "M104 S0" also gets executed... I do not get why?? The result is the nozzle has cooled down when the filament reaches it... 

Does anybody know why that M104 gets triggered just after the M109 even though it's last in line?

Any ideas on how I can fix this or achieve the same end goal?

Tannoo, 2017-05-24:
There is no command to wait, only to execute as soon as the buffer allows.

Roxy-3D, 2017-05-24:
@JanVelleman  Can you use the M600 to do what you need?   It also retracts...  And waits for confirmation, etc.   But it handles the full filament change cycle.

JanVelleman, 2017-05-24:
Hi, thank you for your replies.

Well i thought of using the m600 command but when you want to load filament when there is none already loaded it seemd stupid to do the unload cycle first... also when you want to change from abs to pla for example there is no way to change temperatures between unload and load routines. When you have seperate load and unload routines for abs and pla that is possible...

If i would use flags in the m600 routine to choose between only load/unload, how would i go about doing that? Can you help me a bit there? 

If that is not an option is there a way to achieve what i want with the "enqueue_and_echo_command()" routine instead of with "enqueue_and_echo_commands_P()"?

Thank you

Bob-the-Kuhn, 2017-05-24:
Put a "M400" command just before the "M104 S0" command.

M400 waits until all the motor movements have been completed.  This keeps any later commands in the queue from executing.

To expand on @tannoo's comment, Marlin executes the commands as soon as it can.  Movement commands are put into a queue and then Marlin fetches the next command and executes it.  Processing will stop if the movement queue is full or if a command is received that has a wait/synchronize function in it.

JanVelleman, 2017-05-24:
Ok, i will try that. Thanks for the explanation!

JanVelleman, 2017-05-24:
Works! Thanks :)
I'll close this issue.

closed 2017-05-24 by JanVelleman

loceur, 2018-10-16:
It would be nice to add a quick comment in the configuration_adv about this issue, as it's not intuitive and there's no easy debugging methodology.

AnHardt, 2018-10-17:
Come on. Isn't that common knowledge? Next time we explain them how to clean their nose.

loceur, 2018-10-17:
It wasn't common knowledge for me. Maybe I missed the common knowledge
thread somewhere?

I essentially copy/pasted gcode generated from Cura, and Cura didn't use
M400s in it's code.

Instead of inserting a comment about m400s, maybe insert an example of
moving, waiting, and doing another action?  Or maybe some other more
complex set of actions?

loceur, 2018-10-17:
> How about something like:
> "Use M400 after movement commands to wait for them to complete."

That's nice and easy and works for me.

loceur, 2018-10-17:
User level debugging is probably the more general fix.  Getting feedback of
script done is not useful. Outputting the gcode debug responses to a file
on the SD with debug enabled, for example, would have helped me solve the
issue without lucking into this thread.

thinkyhead, 2018-10-17:
Just so you understand, when you send a G1 command, the movement is queued and the next command is then executed right away. If Marlin waited for every G1 command to complete, well, you can see that simply wouldn't work.

AnHardt, 2018-10-17:
Of cause we could delay about everything, like we do with M106 ( fan_speed) - but the cost on RAM would be enormous.
If that would be better? I forsee the issues: "Feature ??? delayed!", "Birthday party ruined because Marlin did not beep before pause at the end!", "Climate ruined because Malin shut down the heaters 5 seconds to late."  (Ok. Maybe i stood up with the wrong foot this morning.)

loceur, 2018-10-17:
Hah!  No, don't change the delay. The way it works is fine. Just need a way
to troubleshoot and understand how it works.