MarlinFirmware/Marlin #6835: Cross platform primitive types

p3p, 2017-05-23:
Marlin uses both platform dependant types (int, long) and constant bitwidth types (int16_t, int32_t), having just rebased the 32bit branch again this caused a bit of frustration with the new ultralcd code, could there be an addition to the coding standards that require constant bitwidth types?

thinkyhead, 2017-05-23:
I definitely patch up primitive types whenever I have a chance. I'll add a note about this to the coding standards page shortly. I have some other things I want to include there too, now that we've been doing this a while.

closed 2017-05-23 by thinkyhead

Roxy-3D, 2017-05-23:
@thinkyhead   @p3p    Your thoughts on:  https://github.com/MarlinFirmware/Marlin/pull/6228#issuecomment-303504855