MarlinFirmware/Marlin #6838: MarlinMesh.scad incompatible with UBL

040medien, 2017-05-23:
The mesh output table from UBL shows the XMIN/YMIN value in the lower left cell. When reformatting that into a matrix that MarlinMesh.scad accepts, the script expects the XMIN/YMIN value to be in the top left cell. This results in a mirrored output of the mesh points.

040medien, 2017-05-23:
I just found out about  SCAD_MESH_OUTPUT - but it seems that's not compatible with UBL either.

Roxy-3D, 2017-05-23:
Well...  I haven't used MarlinMesh.scad.    I've only done it a couple of times, but a G29 T1 will output the mesh in a .CSV format that can be pulled into any spreadsheet.   And at that point you just highlight the data and tell it to plot it.

I think a topography map with the (X-Min,Y-Min) in the lower left corner makes sense.  Very often, the (X-Min, Y-Min) is (0,0) and it is natural for this to be in the lower left corner for most people.

I think MarlinMesh.scad might need to be tweaked to handle that formatting of the data.  We did you acquire MarlinMesh.scad ?

040medien, 2017-05-23:
It's in your git repository:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/buildroot/share/scripts/MarlinMesh.scad

And at least as of 1.1.1, G29 T1 doesn't output a CSV file, it probes three points and then provides no other feedback. Maybe a recent change?

Roxy-3D, 2017-05-23:
No...   Please recheck that...  The current code is using T for a Topographical map.   The older code used O.   So if you don't want to reflash your firmware to see what I'm talking about, do a G29 O1 instead.

closed 2018-02-06 by thinkyhead