MarlinFirmware/Marlin #6841: z_safe_homing does not use my z_probe

gacek4000, 2017-05-23:
after couple houers testing i have no ide what to do
its looks like z safe homing does not use my z probe
G28 makes X and Y homing but Z axis doesn't react on z proble, its homing only on z min endstop
G29 works well - using a z probe
M119 shows that z probe is triggered by a bed as it should be
on my old Marlin 1.1.0-RC4 from 24 March 2016 everything is working corectly
my z probe (inductive sensor) is connected to Pin 19
please help me guys...
[configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1023832/configuration.txt)

Bob-the-Kuhn, 2017-05-23:
To use the Z probe for homing you have to use the Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and disable Z_MIN_PROBE_ENDSTOP.

gacek4000, 2017-05-24:
doesnt work. 
i set as you said and reconnect probe to z min endstop pin
now its works

Bob-the-Kuhn, 2017-05-24:
GREAT!!

closed 2017-05-24 by Bob-the-Kuhn