MarlinFirmware/Marlin #6842: Bed leveling after starting the printer does not go back to park position

Lyonidaspay, 2017-05-24:
Hello.
I noticed that the nozzle did not go back to the start/park position after fishing probing.
I have a BLTouch installed and working fine till now ( no crashes for me), I just noticed that when doing a bed leveling with the LCD menu, directly after starting the printer, the procedure goes well, but at the end the nozzel stays at the last probing position and does not come back the start/park position.

I have a BLTouch not connected to Z_MIN_ENDSTOP, it's connected to dedicated pin, I also still have the Z_MIN_ENDSTOP working fine. When homing Z axis, it's the Z_MIN_ENDSTOP that actually stops the Z mouvement not the BLTouch ( pin ), But when probing it's the BLTouch ( pin) that stops the Z mouvement.

Bob-the-Kuhn, 2017-05-24:
All the above is as expected.

If you want to use the Z probe to home and to probe then:
1. It MUST be connected to Z_MIN_PIN
2. the current Z_MIN sensor must be disconnected
3. disable Z_MIN_PROBE_ENDSTOP
4. enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Lyonidaspay, 2017-05-24:
Thanks, but I need to probe with Z_MIN_PROBE_ENDSTOP, and Home Z axis with Z_MIN_ENDSTOP.
For now I only need the nozzel to go back to it park position after leveling .

fiveangle, 2017-05-24:
G29 does not home afterward as far as i am aware, and never has on my system.

Bob-the-Kuhn, 2017-05-24:
There's a method to automatically execute commands after G29.  For your case I think it would be similar to the following:

#define Z_PROBE_END_SCRIPT "G0 X0 Y0"

Lyonidaspay, 2017-05-25:
Thank you. I Will try it and give you a feed back. 
Just to let you know, I have the nozzle parked with the probe away from the bed on X Y  HOMING so I can't  use the probe as a Z min endstop. 
I'm working on a FILAMENT JAM sensor and I hope it would replace the RUNOUT sensor.  
A FILAMENT JAM sensor would detect both RUNOUT and nozzle JAM . It is a retary encoder touching the FILAMENT. 
When the encoder is  turning the FILAMENT is moving and when stoped it will pause ,park the nozzle and show a menu on the LCD for two options : 
1 - FILAMENT Change ( call FILAMENT RUNOUT M600)
2 - FILAMENT JAM (  retract and extrude with the panel encoder)

Work in progress.
If you can help with this !

Bob-the-Kuhn, 2017-05-25:
If you enable Z_SAFE_HOMING then G28 will do the Z probe at the position you specify with Z_SAFE_HOMING_X_POINT and Z_SAFE_HOMING_Y_POINT.

closed 2018-02-06 by thinkyhead