MarlinFirmware/Marlin #6843: Z raise before homing

hackqiang, 2017-05-24:
when I update Marlin to v1.1, I found when I execute homing,
the Z will raise(go down) for a little distance before XY homing.
but my Z is at the bottom in the first time, so the Z can not go down any more.
Is there any config for disable the little Z go down before auto homing?
thanks

Tannoo, 2017-05-24:
Do you have a Z_MAX limit?

hackqiang, 2017-05-24:
thanks for you reply,
yes, I have Z_MAX set, here is my configre:
https://github.com/hackqiang/Marlin/blob/RC/Marlin/Configuration.h

BTW:
my corexy using the double Z with belt， the default position of Z is on the bottom.

Bob-the-Kuhn, 2017-05-24:
Z_HOMING_HEIGHT is intended to provide clearance when doing homing so that the nozzle doesn't run into anything like clamps or clips.

If you're homing Z in the - direction, the default G28 sequence is to first increase the distance between the bed and the nozzle by the amount specified in Z_HOMING_HEIGHT.  After that it homes the X & Y axis and then it homes Z.

You can home Z first but it requires that you home Z in the + direction by setting Z_HOME_DIR to 1.  Default is -1.  That means you'd need a Z_MAX endstop.  Currently in your config it's disabled.

Does the Z axis move in the correct direction?  

The raise at the beginning of G28 will be in the same direction as if you issued G0 Z0 followed by G0 Z5.  If that's not the case then there is definitely something wrong.

FYI - since your config has Z_HOMING_HEIGHT disabled then Marlin will use Z_CLEARANCE_BETWEEN_PROBES as the amount to raise at the beginning of G28.  Currently that is set to 5mm.

hackqiang, 2017-05-24:
Thanks !
I have no switch for Z_MAX endstop, the Z direction is correct, the printing is OK.
the only problem is the process of G28:
step1: Z goes down for a little distance(maybe now it's Z_CLEARANCE_BETWEEN_PROBES==5mm)
step2: XY homing
step3: Z homing
So, I just need set the  Z_HOMING_HEIGHT to 0, so the step 1 won't happen, right?
(I will test and feedback on this issue)

hackqiang, 2017-05-24:
After I set the Z_HOMING_HEIGHT to 0 in Configuration.h,
the Z won't go down anymore when homing.
Thanks to all guys~

closed 2017-05-24 by hackqiang