MarlinFirmware/Marlin #6847: Makerbase MKS MINI12864 v1.0 with RAMPS_14 Configuration Marlin 1.1

dvseliteus, 2017-05-24:
Has anyone gotten an MINI12864 LCD to work with Ramps 1.4 on the new Marlin 1.1?

I have tried many different things and referenced https://github.com/makerbase-mks/Marlin but could not figure out how to properly configure dogm_lcd_implementation.h in the old or new versions of marlin.

Bob-the-Kuhn, 2017-05-24:
Try enabling just REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.  It's based on the 12864.

dvseliteus, 2017-05-24:
@Bob-the-Kuhn tried enabling REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
After firmware uploaded the speaker would beep every half second with no backlight. I swapped EXP 1 & 2 which showed backlight but no display.

dvseliteus, 2017-05-25:
The one I have is a little different.
Here is the one I have [HERE](https://www.aliexpress.com/store/product/3d-printer-Reprap-LCD-MKS-MINI12864LCD-mini-12864-smart-display-Reprapdiscount-controller-Full-Graphic-for-mother/1047297_32808844839.html?spm=2114.12010612.0.0.kCEwGI)

Bob-the-Kuhn, 2017-05-25:
There's some decent interface instructions in that link.  I take it that they don't work.

How are you connecting the display to your RAMPS board?

dvseliteus, 2017-05-25:
:smile: I tried the custom pin settings with the new version of marlin and no dice.
Maybe I need to try it with an older version?

I am connecting the EXP 1 & 2 Cables to a with one of these [Adapter Cards](https://images-na.ssl-images-amazon.com/images/I/71-cYLzXhvL._SL1500_.jpg)

Bob-the-Kuhn, 2017-05-25:
Are the connectors on your adapter card the same orientation as the picture?  My card has the key on the other side of the connectors.

Do you at least have the backlight coming on?

dvseliteus, 2017-05-25:
I am able to get the back-light to come on but there is no text. I think you might be on to something here.
Tomorrow I'll cut the headers on the adapter so I can invert the cables. Will followup on the result. Thanks for the help!

Bob-the-Kuhn, 2017-05-25:
Don't do that.  If the headers were backwards then you wouldn't get a light.

Does the encoder, SD card and button(s) work?

fiveangle, 2017-05-25:
I have this exact same display that just arrived today from the big rock candy mountain, which I felt forced to get in order to try out UBL on my Printrbot (although now I hear LCD is no longer an absolute requirement - blech).

I have a feeling getting it to work on your RAMPS will be much easier than on the rats nest of pin voodoo that is the Printrboard, but I'm curious how it goes for you :)

Bob-the-Kuhn, 2017-05-25:
Try this file.  I've made a couple of changes in it compared to the instructions in the link.  On my RAMPS system DOGLCD_A0 should be connected to 16 and DOGLCD_CS should be connected to 17.  

There one more pin I have a question on.  I expect that it's a no connect for this LCD.  I've sent a couple of emails trying to get the pinout of the connectors.

[MKS MINI12864 pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/1028031/MKS.MINI12864.pins_RAMPS.zip)

Bob-the-Kuhn, 2017-05-25:
@fiveangle - Dave - if you find the connection diagram for a RepRap Discount Full Graphics controller then you're 90% of the way there. [This ](https://www.aliexpress.com/store/product/3d-printer-Reprap-LCD-MKS-MINI12864LCD-mini-12864-smart-display-Reprapdiscount-controller-Full-Graphic-for-mother/1047297_32808844839.html?spm=2114.12010612.0.0.kCEwGI) says to use the MINIPANEL LCD controller option & modify some pin assignments.

dvseliteus, 2017-05-25:
@Bob-the-Kuhn 
I tried the new pins.RAMPS.h file and had the same result. I tried defining the following LCS;
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
MINIPANEL
SAV_3DGLCD

All of those definitions have similar results.
When I connect the EXP 1 & 2 cables I get no back-light but I do get some buzzer noise randomly.
When I connect the EXP 1 & 2 cables inverted I get a back-light with no text and no buzzer noise.

Here are some photo's of my actual parts: [Adapter](https://ibb.co/j5xFSa) | [LCD](https://ibb.co/cqm9na)
Also I did some digging and found the Manufactures website, which is zero help: https://www.makerbase.com.cn/home/index

Bob-the-Kuhn, 2017-05-25:
When you had the cables inverted, were you able to see the SD card?

Was the screen blank or did it have garbage on it?

I received an email from MKS but it basically translated to the pinout in the link you provided earlier.

Try the mods in the link but this time use:
```
DOGLCD_CS  27
DOGLCD_A0  25
```

If that doesn't work then it's back to basics.  Let's make sure of the pins on your adapter.

Here's the pinout I expect to see.  This is what the Full Graphic interface is.
```
           EXP1                             
  BEEPER   D37   1  2   D35  BTN_ENC        
  LCDE     D17   3  4   D16  LCDRS          
  LCD4     D23   5  6   D25  LCD5           
  LCD6     D27   7  8   D29  LCD7           
           GND   9  10  VCC                 
                                            
                                             
           EXP2                             
  MISO     D50   1  2   D52  SCK            
  BTN_EN2  D31   3  4   D53  SD_CSEL        
  BTN_EN1  D33   5  6   D51  MOSI           
  SD_DET   D49   7  8   D41  STOP/KILL BTN  
           GND   9  10  NC                  
```

To identify the logical pin that the each connector pin is attached to you'll need to attach an LED to the pin in question and run a utility that pulses every pin.
1. Enable PINS_DEBUGGING in Configuration.h
2. Attach an LED between +5V and ground.
3. Once the LED is on then move the + lead to ENC C
4. Enter this command M43 T I1 R5. This will toggle each pin in sequence.

If you want to save some time then here's the full syntax for the M43 T command.
```
   *  M43 T       - Toggle pin(s) and report which pin is being toggled
   *                  S<pin>  - Start Pin number.   If not given, will default to 0
   *                  L<pin>  - End Pin number.   If not given, will default to last pin defined for this board
   *                  I<bool> - Flag to ignore Marlin's pin protection.   Use with caution!!!!
   *                  R       - Repeat pulses on each pin this number of times before continueing to next pin
   *                  W       - Wait time (in miliseconds) between pulses.  If not given will default to 500
```

----

I did run across a [thread](https://github.com/repetier/Repetier-Firmware/issues/658#issuecomment-303945397) where they were able to get the MIN12864 to work.  The nasty is they did hardware mods to the LCD.  Sure hope it doesn't come to that.

GarrethX, 2017-05-25:
> I did run across a thread where they were able to get the MIN12864 to work. The nasty is they did hardware mods to the LCD. Sure hope it doesn't come to that.

My HW mod is just required for DUE, as the HWSPI in u8glib v1 for that display is not working at 32 bits:) 
At rumba I managed to run this display with HWSPI.

I even found an old code I used for rumba (not in 3d printer firmware, just one of my projects needed a temporary display:) ), I used there simply:

```
#define UI_DISPLAY_DOGLCD_CS 38
#define UI_DISPLAY_DOGLCD_A0 41
U8GLIB_MINI12864 u8g(UI_DISPLAY_DOGLCD_CS, UI_DISPLAY_DOGLCD_A0,U8G_I2C_OPT_NONE);
```

Bob-the-Kuhn, 2017-05-25:
@GarrethX - many thanks   😊

Combined with your last post in the original thread I think I see a light at the end of the tunnel.

> #define MINIPANEL
> But you must go through the firmware I posted, in file ultralcd_impl_DOGM.h I modified line 176:
> #elif ENABLED(MINIPANEL)
>   // The MINIPanel display
>   //U8GLIB_MINI12864 u8g(DOGLCD_CS, DOGLCD_A0);  // 8 stripes
>   U8GLIB_MINI12864_2X u8g(53,51,DOGLCD_CS, DOGLCD_A0,U8G_PIN_NONE);

Someone has already made a similar mod to ultralcd_impl_DOGM.h.  Not quite the same so if the current one doen't work then we'll try 
U8GLIB_MINI12864_2X u8g(DOGLCD_CS, DOGLCD_A0,U8G_PIN_NONE);

dvseliteus, 2017-05-25:
@GarrethX 
Where can I define?
#define UI_DISPLAY_DOGLCD_CS 38
#define UI_DISPLAY_DOGLCD_A0 41
I tried modifying 
```
#define MINIPANEL
But you must go through the firmware I posted, in file ultralcd_impl_DOGM.h I modified line 176:
#elif ENABLED(MINIPANEL)
// The MINIPanel display
//U8GLIB_MINI12864 u8g(DOGLCD_CS, DOGLCD_A0); // 8 stripes
U8GLIB_MINI12864_2X u8g(53,51,DOGLCD_CS, DOGLCD_A0,U8G_PIN_NONE);
```
But was unable to get it working...
I came across the [post](http://www.fabrykator.pl/board/viewtopic.php?f=67&t=859&sid=595a9f6c664be3b51ef19ea1aba186fc&start=20) where I think you got this from and the photo's on page 3 look like I may need to connect to more pins, but I'm not sure since this was for the rumba board?

@Bob-the-Kuhn 
I also tried the 
> U8GLIB_MINI12864_2X u8g(DOGLCD_CS, DOGLCD_A0,U8G_PIN_NONE);
But was not able to get it working. This may be beyond my technical abilities.

Bob-the-Kuhn, 2017-05-25:
Different thread, same white knight.

I can do the modifications.

But first ... is the SD card portion working?

Please identify the pins on your connector.  I expect that we'll find some differences.  Once we know for sure what we're working with then we can move forward to modifying pins_RAMPS.h and possibly ultralcd_impl_DOGM.h.

GarrethX, 2017-05-25:
Well, we start to have a little confusion because of amount of threads :)

First, for 8 bit processors:
u8glib v1 works fine in hw spi together with the card reader. Pins for
rumba I posted above which I used for one project succesfully (lcd+card
reader).  I guess they are the same on other boards like ramps?

For 32 bit processors:
u8glib v1 works only in SW SPI
u8glib v2 works in both SW and HW spi

Pin defines I copied from my project and their position was not relevant
there.
In firmware of the printer you want to put these defines in pin file for
your board.

I posted the drawing on the mentioned thread how the pins are located on
the exp1 and exp2 ports, but since different boards can have those ports
with different pinout, you have to check and compare pinout of your board
and adjust accordingly.

dvseliteus, 2017-05-25:
@Bob-the-Kuhn 
I'll grab an SD card tomorrow to test. But doing more research Leap 3D on their [ebay listing](http://www.ebay.com/itm/Mini-MKS-12864-Graphics-LCD-Display-Controller-3D-Printer-w-Side-Load-SD-Card-/122415138875?hash=item1c8082c83b:g:t5sAAOSwCU1Y27D5) is saying;

> This LCD controller similar to Full Graphic LCD Smart Controller and most of information found in in [RepRapDiscount Full Graphic Smart Controller WIKI](http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller) will apply here as well. <
Still working to find the pinout for the actual board.

Bob-the-Kuhn, 2017-05-26:
Looks like EXP2 is the same as the FULL Graphic & EXP1 is similar but needs different signals mapped to the pins.

I'm expecting (hoping) to see some pin mapping differences on your adapter from mine.  

Oh - please use those jumpers that plug onto the connector pins when hooking up the LED.  I just heard of one where he apparently shorted +12V to +5V and fried his controller.

Bob-the-Kuhn, 2017-05-26:
I believe that we'll end up with the following pinout for the MKS MINI12865

```
EXP1                         
  FUNCTION    PIN #   FUNCTION
    BEEPER     1  2     BTN_ENC    
    NC         3  4     NC         
    NC         5  6     DOGLCD_CS  
    DOGLCD_A0  7  8     NC         
    GND        9  10    VCC        
  
  
EXP2                         
  FUNCTION    PIN #   FUNCTION
    MISO       1  2     SCK        
    BTN_EN2    3  4     SS         
    BTN_EN1    5  6     MOSI       
    SD_DET     7  8     KILL_PIN   
    GND        9  10    NC         
  
EXP2 - same as RepRapDiscount FULL GRAPHIC Smart Controller
  
AT90USB controllers - use FASTIO logical pin numbers
  
Hardware implemented functions (software versions available)
        logical pins
        2560  2561  1284  AT90USB  
   SCK   D52   D10   D7    D9       
   MOSI  D51   D11   D5    D10      
   MISO  D50   D12   D6    D11      
   SS    D53   D16   D4    D8  
```

dvseliteus, 2017-05-26:
@Bob-the-Kuhn what's this pinout mean in layman's terms?
Also what did you mean by  jumping the connector pins when hooking up the LED?
and another question how can I tell if the SD Card is working with no screen text?

Bob-the-Kuhn, 2017-05-26:
OK - need to do a real graphic on the pinout.

On the SD card - use the host interface (Repetier Host, Octoprint) to see if the SD card shows up.

On the LED - I have some single conductor female-female jumper cables that are meant to slip over the square posts on the controller board connectors/pins.  I'm just saying it's safer to use those than to try to maneuver the LED pins directly.

fiveangle, 2017-05-26:
These are the jumper wires @Bob-the-Kuhn is referring to, which make testing out different pin connections much easier:

https://www.adafruit.com/product/266

dvseliteus, 2017-05-26:
Thanks @fiveangle I have these on hand. I was hoping for a digital pin change than a wire harness adapter. Maybe I'm better of getting another display ☺️? I'll try to follow @Bob-the-Kuhn instructions and get it working.

fiveangle, 2017-05-27:
There's no CPLD on any 3D printer I've ever seen, so sometimes you need to just cozy up to the diagnostic basics ;)

I'll be working to get this LCD working on a Printrboard RevF sometime over the long weekend so I'll let you know if I find out anything interesting (which I doubt will help you, since it sounds like the real mystery for you is on the board-size.  Good luck !

-=dave

Bob-the-Kuhn, 2017-05-28:
@fiveangle - I just committed Teensyduino compatible M43 code on PR #6737.  I expect it'll make finding the pin numbers easier.

Also, I noticed that the SD card reader software uses both FASTIO & Teensy pin number routines.  You may not be able to get the card reader portion working.

fiveangle, 2017-05-28:
Yeah, I wasn't expecting the SD card to work.  I'll try it immediately and report back. Thx !

Bob-the-Kuhn, 2017-05-28:
Looks like all the pins use FASTIO except for SDSS & LCD_SDSS.  Those use the Teensy numbers

Bob-the-Kuhn, 2017-05-29:
@fiveangle - how's it coming?

If you give me your connection list and a copy of your pins_xxx.h file I'll see if I can come up with some suggestions.

fiveangle, 2017-05-29:
> ```
> EXP1
>   FUNCTION    PIN #   FUNCTION
>     BEEPER     1  2     BTN_ENC
>     NC         3  4     NC
>     NC         5  6     DOGLCD_CS
>     DOGLCD_A0  7  8 [...]

Looks the one I have is the side-access SD slot and has the keyways (and silkscreen) the wrong way around, so technically comes out to this:

```
EXP1                         
  FUNCTION    PIN #   FUNCTION
    GND        1  2     VCC    
    DOGLCD_A0  3  4     NC         
    NC         5  6     DOGLCD_CS  
    NC         7  8     NC         
    BEEPER     9  10    BTN_ENC       
  
EXP2                         
  FUNCTION    PIN #   FUNCTION
    GND        1  2     NC        
    SD_DET     3  4     KILL_PIN         
    BTN_EN1    5  6     MOSI       
    BTN_EN2    7  8     SS   
    MISO       9  10    SCK         
```

I'm making a custom cable for the Printrboard anyway, but it's probably best for others to just shave the keyway off and plug it in the other way around (with red tracer on ribbon closest to the "E" side of the port labels)

fiveangle, 2017-05-29:
@Bob-the-Kuhn - let me know if you'd be up for a slack or hangouts session to help me with a few questions to get me started. Nothing is plugged in for Printrboards except the viki2 panel so I've got to put it all in and I just want some pointers on the "best" way to  do it.

fiveangle, 2017-05-29:
Found this from another vendor and matches closer to what I'm observing:
(From http://m.globalsources.com/gsol/I/3D-printer/p/sm/1148779928.htm)
[image]

Bob-the-Kuhn, 2017-05-29:
It's a busy day so I don't know when I'll be available.

Looks like you've made good progress.  

Try hooking up MOSI, SCK, and defining DOGLCD_CS & DOGLCD_A0.  Maybe you'll get something on the display.

The only remaining signals to hook up are MISO & SS (also called SDSS & SDCS).  

I'll have to do more research on what to do about SS.  On Printrboard it only goes to the SD card.  I've heard of software SPI being used on Marlin.  It sounds like it's an option that's already there.  

I'll mod the files & send them later.

fiveangle, 2017-05-29:
I think I'm done.  The Printrboard just wasn't meant to be run with an LCD ;)

I've wasted enough of my holiday on this failed project so someone else will have to pick up the torch.

I enabled MINIPANEL and set options in RevF pins file (attached) to patch to EXP1 header like so (which PB fork recently enabled VIKI for their new printers on same pins, so is probably correct):

```
#if ENABLED(MINIPANEL) 
  #define DOGLCD_A0  42 // Non-FastIO
  #define DOGLCD_CS  43 // Non-FastIO
  #define LCD_SCREEN_ROT_180

  // (FastIO Pins)
  #define BTN_EN1 26
  #define BTN_EN2 27
  #define BTN_ENC 47

  #define BLEN_C 2
  #define BLEN_B 1
  #define BLEN_A 0

  // encoder rotation values
  #define encrot0 0
  #define encrot1 2
  #define encrot2 3
  #define encrot3 1
#endif

```

It appears defining MINIPANEL defines ULTRAPANEL, which defines ULTRA_LCD, which already contained pin definitions in the RevF pins file, resulting
[garbage.zip](https://github.com/MarlinFirmware/Marlin/files/1036493/garbage.zip)

 in a bunch of redefines.  Commenting those out, this was the closest it came to compiling:
And this is the closest I've gotten to getting it to compile...

```
"/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/../tools/avr/bin/avr-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -MMD -fno-exceptions -felide-constructors -std=c++0x -mmcu=at90usb1286 -DTEENSYDUINO=136 -DARDUINO_ARCH_AVR -DARDUINO=10802 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/cores/teensy" "-I/Users/speedster/Documents/Arduino/libraries/U8glib/src" "-I/Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/libraries/Wire" "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_243020/sketch/ultralcd.cpp" -o "/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_243020/sketch/ultralcd.cpp.o"
ultralcd.cpp: In function 'void lcd_synchronize(const char*)':
ultralcd.cpp:493: error: invalid conversion from 'void (*)(const char*)' to 'screenFunc_t {aka void (*)()}' [-fpermissive]
     lcd_goto_screen(lcd_synchronize);
                                    ^
/var/folders/q1/g9l9mqmn319fh27k43yqpy440000gr/T/arduino_build_243020/sketch/ultralcd.cpp:441:8: note: initializing argument 1 of 'void lcd_goto_screen(screenFunc_t, uint32_t)'
   void lcd_goto_screen(screenFunc_t screen, const uint32_t encoder = 0) {
        ^
Using library U8glib at version 1.19.1 in folder: /Users/speedster/Documents/Arduino/libraries/U8glib 
Using library Wire at version 1.0 in folder: /Applications/Arduino 1.8.2-teensy.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
invalid conversion from 'void (*)(const char*)' to 'screenFunc_t {aka void (*)()}' [-fpermissive]
```

When you begin just flailing, you know it's time to call off the dogs...

Bob-the-Kuhn, 2017-05-30:
@fiveangle 

Dave - take a look at PR #6896.  Looks like I'm well on the way to having a system that uses "Teensy only" pins_XXX.h files.  

As part of the testing I'm going to need to interface an LCD to my Teensy.  95% of that should translate directly to Printrboard.  Don't throw the MKS MINI12864 away yet.

Bob-the-Kuhn, 2017-06-03:
@fiveangle

Dave - we're getting close to releasing the Teensy only system.  Could you do us a favor & see if it runs on your system?

Just drop these [files](https://www.dropbox.com/s/cntdcrd48jqms8d/unify%202.zip?dl=0) into the latest bugfix-1.1.x and rename the Printrboard file. You shouldn't notice anything different except the pin numbers.

Also, I've fixed the M43 issues I created.  It should be a useful tool again.

Bob-the-Kuhn, 2017-06-04:
My LCD is working!!!

I had to add the following lines to my Configuration.h file.  Apparently on the AT90USB boards more delay is needed on the bit to clock setup & hold times.  Haven't done any experimenting to see if shorter delays will work.  DELAY_0_NOP is the shortest, 5 is the longest.
```
#define ST7920_DELAY_1 DELAY_5_NOP
#define ST7920_DELAY_2 DELAY_5_NOP
#define ST7920_DELAY_3 DELAY_5_NOP
```

Only three signals are needed to drive the LCD.
```
CLOCK   LCD_PINS_D4			
DATA    LCD_PINS_ENABLE			
SELECT  LCD_PINS_RS	
```

Bob-the-Kuhn, 2017-06-04:
I've been doing some digging on the MINIPANEL setup in Marlin.  The line of interest is:
U8GLIB_MINI12864_2X u8g(DOGLCD_CS, DOGLCD_A0);

In the U8GLIB documentation I found this:
```
class U8GLIB_MINI12864_2X : public U8GLIB 
{
  public:
    U8GLIB_MINI12864_2X(uint8_t sck, uint8_t mosi, uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) 
      : U8GLIB(&u8g_dev_uc1701_mini12864_2x_sw_spi, sck, mosi, cs, a0, reset)
      { }
    U8GLIB_MINI12864_2X(uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) 
      : U8GLIB(&u8g_dev_uc1701_mini12864_2x_hw_spi, cs, a0, reset)
      { }
};
```

It looks like there are four lines to control the LCD (sck, mosi, cs, a0) and that Marlin assumes the LCD is hooked up to a hardware SPI.  I think you can get it to work with the following connections:
SCK - get it off the ISP connector
MOSI - get it off the ISP connector
DOGLCD_A0 - any convenient I/O pin
DOGLCD_CS - any convenient I/O pin

An alternative method is to use a software SPI.  I think that is built into the U8GLIB.  I think this will also work:
```
#define MINI_SCK   // any convenient I/O pin
#define MINI_MOSI   // any convenient I/O pin
#define DOGLCD_A0  // any convenient I/O pin
#define DOGLCD_CS   // any convenient I/O pin
replace U8GLIB_MINI12864_2X u8g(DOGLCD_CS, DOGLCD_A0); with U8GLIB_MINI12864_2X u8g(MINI_SCK, MINI_MOSI, DOGLCD_CS, DOGLCD_A0);
```

If you use the hardware SPI signals I think you'll have access to the SD card.

If you use the software SPI signals then I doubt you'll have access to the SD card.

fiveangle, 2017-06-04:
Will try to test the new teensy pins setup for you tomorrow night if I can get some time

Bob-the-Kuhn, 2017-06-04:
Thanks

My LCD is now fully functional.  The SD card software also has a software SPI.  The result is I'm using two software SPIs.  That's OK for the RepRap full graphics because the LCD and the SD don't share any signals.

The MKS MINI12864 appears to use MOSI and SCK for both the display and the SD card which means there can be only one SPI.  Right now I only know of the hardware SPI that can do this.  There is a MSPIM hardware SPI on the chip but I have not been able to find the software to run it.

The EXP2 connector and the ISP connector have all the signals needed to run the MKS MINI.  

I have a cable that should connect your MKS MINI to the Printrboard.  If you give me your address I'll send it to you.

To use the cable you'll need the following defines:
```
#define DOGLCD_A0      4  // D4  EXP2-5
#define DOGLCD_CS      5  // D5  EXP2-6
#define BTN_ENC        6  // D6  EXP2-7
#define BEEPER_PIN     7  // D7  EXP2-8
#define KILL_PIN       8  // E0  EXP2-10
#define SD_DETECT_PIN  9  // E1  EXP2-11
#define BTN_EN1       10  // C0  EXP2-12
#define SDSS          11  // C1  EXP2-13
#define BTN_EN2       12  // C2  EXP2-14
```

----

Were you the one that asked for my email so long ago? It's
bob.kuhn@att.net

Bob-the-Kuhn, 2017-06-05:
@fiveangle - I just noticed your post.  Sorry about the delay.

Thanks for testing.  Make me a whole lot more comfortable with the change.

Nice to know that Teensduino can now handle 128k images.  Now if I could only get it to install I'd be a happy camper.

Yes, removing the pointer to the Printrboard IDE extension makes sense given the 128k capability.

I know we've gone over this fan thing before.  With the old system I'm sure that the physical output pin changed depending on the options that were enabled.  I can't lay my hands on that discussion.

It makes sense to me to change it so the fan is always on the fan connector.  I can't tell what the original developers meant to happen.  Al I know for sure is the results didn't match the comments.

Again - thanks for the testing.

fiveangle, 2017-06-05:
Nice work @Bob-the-Kuhn !  Reviewing your recent comments added in your pins*.h config files against my previous work, I had pretty much made it this far before but looks like I hit the HW SPI hurdle that I was unware of and beyond my capabilities to figure out at the time, which you've managed to overcome. 

Printrboard has in-built SD on HW SPI plus you I can't enable SD + ABL + LCD anyway due to 128k prog mem limit, so ditching SD entirely makes sense for anyone with Octoprint (although with other optimizations like using only 3PL, no workspace offsets, etc, and every other trick in the book, it might be possible to just squeeze SD in along with graphics-based LCD but I doubt it).

I'll change my defines to match yours for simplicity.  If you give me the complete pinout connections and other changes, I can makeshift the connections here now and confirm.

Using your defines, I have:

Configuration.h:
```
/**
 * SD CARD: Multiple
 * 
 * Set to use always use internal SD card slot instead of SD on your LCD panel if your board has one
 */
#define USE_INTERNAL_SD

//
// MKS MINI12864 panel with graphic
// controller and SD support - https://github.com/makerbase-mks/Marlin
//
#define MKSMINI12864
```

pins_PRINTRBOARD_REVF.h:
```
#if ENABLED(MKSMINI12864)
  #define MINIPANEL
  #if ENABLED(USE_INTERNAL_SD)
    //      PIN       FASTIO PIN#  ATUSB90 PIN# Teensy2.0++ PIN# 
    #define SDSS               20  //        10               B0  
    #define SD_DETECT_PIN      -1  // no auto-detect SD insertion on built-in Printrboard SD reader
  #else
    //      PIN       FASTIO PIN#  ATUSB90 PIN# Teensy2.0++ PIN#  Printrboard RevF Conn.   MKSLCD12864 PIN#  
    #define SDSS               11  //        36               C1                EXP2-13             EXP2-07
    #define SD_DETECT_PIN       9  //        34               E1                EXP2-11             EXP2-04
  #endif
  
    //      PIN       FASTIO PIN#  ATUSB90 PIN# Teensy2.0++ PIN#  Printrboard RevF Conn.   MKSLCD12864 PIN#  
    #define DOGLCD_A0           4  //        29               D4                EXP2-05             EXP1-04      
    #define DOGLCD_CS           5  //        30               D5                EXP2-06             EXP1-05
    #define BTN_ENC             6  //        31               D6                EXP2-07             EXP1-09
    #define BEEPER_PIN          7  //        32               D7                EXP2-08             EXP1-10
    #define KILL_PIN            8  //        33               E0                EXP2-10             EXP2-03
    #define BTN_EN1            10  //        35               C0                EXP2-12             EXP2-06
    #define BTN_EN2            12  //        37               C2                EXP2-14             EXP2-08
    #define LCD_BACKLIGHT_PIN  43  //        56               F5                EXP1-12       [unknown yet]
  //#define SCK                21  //        11               B1                ICSP-04             EXP2-09
  //#define MOSI               22  //        12               B2                ICSP-03             EXP2-05
  //#define MISO               23  //        13               B3                ICSP-06             EXP2-10            
  
  // encoder connections present
  #define BLEN_A 0
  #define BLEN_B 1
  #define BLEN_C 2

  // encoder rotation values
  #define encrot0 0
  #define encrot1 2
  #define encrot2 3
  #define encrot3 1

  // increase delays to max
  #define ST7920_DELAY_1 DELAY_5_NOP
  #define ST7920_DELAY_2 DELAY_5_NOP
  #define ST7920_DELAY_3 DELAY_5_NOP
#endif
```

We'll see how it goes...

fiveangle, 2017-06-05:
[image]

Bob-the-Kuhn, 2017-06-05:
Here's the pinout for the EXP2 connector on the Printrboard.  
```
PIN FUNCTION    PORT
1               GND
2               5v
3               nc
4               nc
5   DOGLCD_A0   4 (D4)
6   DOGLCD_CS   5 (D5)
7   BTN_ENC     6 (D6)
8   BEEPER      7 (D7)
9               nc
10  KILL_PIN    8 (E0)
11  SD_DET      9 (E1)
12  BTN_EN1     10 (C0)
13  SDSS        11 (C1)
14  BTN_EN2     12 (C2)
```

I've got a flat ribbon cable that connects to EXP2 & ISP connectors on the Printrboard, does a few twists & turns and terminates in the MKS MINI connectors.

fiveangle, 2017-06-05:
@Bob-the-Kuhn - yeah, that matches what I have above.  When I send M300 S1 the beeper, it beeps alright, but the entire display goes blank. Could this a power delivery issue with the 5v reg (hard to believe the buzzer takes that much) the or could something else be wired wrong, you think ?

Bob-the-Kuhn, 2017-06-05:
The beeper works but the display goes blank ... very strange.

I'm drawing a blank.  

Could be power.  Could be hash caused by the beeper on power & ground getting the controller upset.

My RepRap doesn't have that issue.

Now that I have the delays set properly, the display is rock solid.

caothai, 2017-06-21:
@Bob-the-Kuhn - how can i use MKS MINI 12864 with ramps 1.4, where do i change the pinout for the EXP2 ?
Thanks

fiveangle, 2017-06-26:
I'm pretty sure you just shave the key off the ribbon plugs and plug them in 180-deg around.  I appoligize for derailing the discussion here around Printrboard, but other than a few items not working (no contrast or backlight control implemented on MINIPANEL) it should plug into your RAMPS adapter after shaving off the keys on the IDC connectors.

dvseliteus, 2017-06-26:
@fiveangle 
Thanks for the update and having the thread turn to the printboard was pretty interesting. I picked up a RepRapDiscount Smart Controller LCD and have been using that. Now that I know my Ramps and Firmware are working I will try to shave the IDC's and invert them when connecting. Let's see what happens...

Bob-the-Kuhn, 2017-06-27:
The RAMPS to MKS MIN 12864 setup is documented in this [WIKI page]( http://reprap.org/wiki/MKS_MINI_12864).

fiveangle, 2017-06-27:
@Bob-the-Kuhn - page looks good.  Only part that I think might be improved to help people is more specifically calling out that the cable keys are opposite of typical graphics controllers:

> Note that the housings are rotated 180 degrees in this graphic.

Would it be better to be more direct like:

> Note that the ribbon cable housings are rotated 180 degrees from most other graphic controllers.  Some users have had success by removing the alignment key so the cable can be plugged in 180 degree so a custom cable is not required.

Or something between yours (terse) and mine (verbose) ?

chexov, 2017-11-28:
Thanks for help!
Just 5 cents. Instead of shaving key from the cable just remove the plastic part of the socket (leaving just the pins on the MKS board) and put it back rotated 180deg.

fiveangle, 2017-11-28:
@dvseliteus - good to close ?

dvseliteus, 2017-11-28:
@fiveangle yup good to close. Great info on this thread. I hope it helped others like it helped me. Thanks all who commented!

closed 2017-11-28 by dvseliteus