MarlinFirmware/Marlin #6849: Missing ; in gcode.h

DaveAhrendt, 2017-05-25:
Compile error on line 230 in gcode.h when TEMPERATURE_UNITS_SUPPORT is defined.
Offending line is:
        return input_temp_units == TEMPUNIT_K ? PSTR("Kelvin") : input_temp_units == TEMPUNIT_F ? PSTR("Fahrenheit") : PSTR("Celsius")

Roxy-3D, 2017-05-25:
So...   If GCC can have 1300 different ways to optimize the code,  how about they add an option to put in missing semi colons on your source file?

Tannoo, 2017-05-25:
More of a travis test is needed for EVERY option.

DaveAhrendt, 2017-05-31:
The ; is still missing on line 230 in 1.1.2 build

Bob-the-Kuhn, 2017-05-31:
This was just fixed in bugfix-1.1.x via PR #6915.

thinkyhead, 2017-06-02:
> More of a travis test is needed for EVERY option.

I'd like to see that, but it has to be done without significantly impacting the build time. If we wanted to methodically test every possible option on every possible board, with every possible LCD controller, it would make the Travis test time ridiculously long. So we have been selective. The best way to add more tests is to still do a small number of test builds, but to enable more options under each one.

The other thing we don't have in the bundle is a helper script that allows us to test intentional build failures, but I'm sure that could be added.

Roxy-3D, 2017-06-02:
> I'd like to see that, but it has to be done without significantly impacting the build time.

110% agreed....

boelle, 2019-02-18:
@DaveAhrendt is the problem gone?

DaveAhrendt, 2019-02-18:
As far as I know, yes.  I moved on to RRF on Duet some time ago so I’ve not kept up with Marlin.

From: Bo Herrmannsen 
Sent: Monday, February 18, 2019 5:06 PM
To: MarlinFirmware/Marlin 
Cc: DaveAhrendt ; Mention 
Subject: Re: [MarlinFirmware/Marlin] Missing ; in gcode.h (#6849)

@DaveAhrendt is the problem gone?

boelle, 2019-02-18:
dare i ask why you got the Duet?

DaveAhrendt, 2019-02-19:
Really needed 32 bit for my larger delta and at the time Marlin 2.0 was not really ready for use.
I also really appreciate the “gcode for everything” approach of Duet and RRF.  Never have to recompile
to make changes.  Just edit a config file and reboot and you’re done.

From: Bo Herrmannsen 
Sent: Monday, February 18, 2019 6:12 PM
To: MarlinFirmware/Marlin 
Cc: DaveAhrendt ; Mention 
Subject: Re: [MarlinFirmware/Marlin] Missing ; in gcode.h (#6849)

dare i ask why you got the Duet?

thinkyhead, 2019-02-20:
I've been using the Duet Wifi and RRF on the E3D Toolchanger beta, and it's pretty cool. I'm told the browser-based UI isn't as good as OctoPrint. I will have to try OctoPrint next!

closed 2019-02-20 by thinkyhead