MarlinFirmware/Marlin #6851: esp8266 with marlin?

vMeph, 2017-05-25:
Hi
Is there any suport or plans on marlin for add esp8266 wifi funcionality in Marlin?

Thanks

CONSULitAS, 2017-05-25:
you should read this: https://github.com/MarlinFirmware/Marlin/issues/6524#issue-225427305

If this answers your question, please close the issue.

simon-jouet, 2017-05-26:
HI @vMeph 

Depends what you're looking for, if you just want to use your atmega + ramps for instance with an ESP8266 as a wifi to serial bridge, then it's very straightforward just use something like this: https://github.com/jeelabs/esp-link.

If you want to use the ESP as the main controller board then at the moment it's not supported, I have been doing some work in the issue @CONSULitAS linked (well not so much in the last two weeks) on porting Marlin to the ESP32 I hope the basic firmware will work soon, and you're more than welcome to contribute.

Using the ESP8266 as the main controller is probably not a great idea, quite a limited number of peripherals and low pin count especially when the ESP32 is now out with much more power at a very cheap price.

vMeph, 2017-05-26:
Thanks for info, i will look into it

closed 2017-05-26 by vMeph