MarlinFirmware/Marlin #6854: Last version of BugFix 1.1.x downloaded today

Bergerac56, 2017-05-25:
Hello,
I just gave a try to the last bugfix version of today (05/25). I see at least 2 differences in behavior (and potential problems)
.
1> When I specify the " #define Z_PROBE_END_SCRIPT "G1 X155 Y105 Z0 F8000" (in my specific case) to end the G29 command, this release seems not to execute correctly the command given. The printer head finishes at X=0 Y=0 Z=0 which is not expected. With the release of 2 days ago, this was correctly executed.

2> When I start to print a file from the SD card, the previous behavior was: Heating the bed / heathing the Nozzle / G28 / G29/start printing. With the last release, the LCD indicates immediately that Heating is done (but still at room temperature) and starts with G28 and G29. Precision: I use CURA 2.5. The "pre" code does not mention the heating part. Only G28 and G29. Heating, using the parameters from the file was "automatic" before. Now, even if I add a M109 S60(bed) and M140 S200(Hotend), it still says immediately "heating done" and does a G28 and G29. It indicates "ENDSTOPS X Y" then " File printing" and... does nothing (whitch is good as the nozzle is cold). But the bargraph on the LCD is progressing...

Reversion to the build of yesterday solves those "issues".

I did not change anything in the configuration files, just copied from one release to the other one. Did I miss something?

Bergerac56, 2017-05-26:
After hours spent to look why, I found the cause of the problem. It is in the very end of the configuration_adv.h file. Starting some days ago, a  little section appeared:
/**
 * Spend 28 bytes of SRAM to optimize the GCode parser
 */
#define FASTER_GCODE_PARSER

And this is all the difference. I did not notice this right away. 

I use  quite deeply modified configuration.h and configuration_adv.h files. Copying those files "as is" to the last versions of the bugfix 1.1.x release did not respect this little section.

With the #define FASTER_GCODE_PARSER, all is fine. Without, the behavior of the firmware is ... interesting ;)

Bob-the-Kuhn, 2017-05-26:
I've also seen this issue.

We should have it fixed today.

Bergerac56, 2017-05-26:
Thanks a lot

fiveangle, 2017-05-26:
#6634

Bob-the-Kuhn, 2017-05-26:
This has been fixed on PR #6866 .  The corrected code should be available now.

Bergerac56, 2017-05-26:
Tried with and without "#define FASTER_GCODE_PARSER". It works as expected. Nice job :) Thanks

fiveangle, 2017-05-27:
Close ?

Bergerac56, 2017-05-28:
Yes, this fix the problem. I made various verifications. Now, activation the "parse" option or not gives a logic result. Thanks again.

closed 2017-05-28 by Bergerac56