MarlinFirmware/Marlin #6856: [video] Manual Mesh Bed Leveling: lcdDrawUpdate not working

bealparslan, 2017-05-25:
Bug video: [https://youtu.be/ptzMVecYLdQ](url)
I'm using last 1.1.x marlin version. My config:

```
...
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
//#define Z_MIN_PROBE_ENDSTOP
//#define Z_MIN_PROBE_PIN Z_MAX_PIN
//#define PROBE_MANUALLY
//#define FIX_MOUNTED_PROBE
...
#define Z_HOMING_HEIGHT 0
...
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
#define MESH_BED_LEVELING
...
#define LCD_BED_LEVELING
...
```
Look: **ultralcd.cpp** line 1642
```
/**
 * Step 3: Display "Homing XYZ" - Wait for homing to finish
 */
void _lcd_level_bed_homing() {
 if (lcdDrawUpdate) lcd_implementation_drawedit(PSTR(MSG_LEVEL_BED_HOMING), NULL);
 if (axis_homed[X_AXIS] && axis_homed[Y_AXIS] && axis_homed[Z_AXIS]) lcd_goto_screen(_lcd_level_bed_homing_done);
 lcdDrawUpdate = LCDVIEW_KEEP_REDRAWING;
}
```
That's the only problem: "Homing XYZ" and "Moving.." messages not shown.
I don't think the condition **lcdDrawUpdate** works correctly.

edautz, 2017-05-25:
I reported this earlier. See https://github.com/MarlinFirmware/Marlin/issues/6790

bealparslan, 2017-05-26:
Well, video was needed. The bug is a little strange.

closed 2018-02-07 by thinkyhead