MarlinFirmware/Marlin #6857: Feature request: adding entry to lcd

batata004, 2017-05-25:
Hi,

I read the entire configuration and _adv file and I found no place where I can add a custom item to the LCD menu. So I ask you: where should I change the code of marlin to add a new menu entry right in the very first option? I want to create an entry labeled (for example) XXX and when I click it I want Marlin to start preheating the bed and start the bed level procedure. 

I have a medium knowledge of arduino and I can change it if you just tell me where I could do that.

Tannoo, 2017-05-25:
In `ultralcd.cpp`, Main Menu starts at about LN 805.

batata004, 2017-05-26:
@Tannoo thanks man! I looked into the line you said and I did this:

1) added the line below to the top of the code:

`void lcd_iniciar();`

2) added the line below to the right position (right above PREPARE):

`MENU_ITEM(submenu, _UxGT("Iniciar"), lcd_iniciar);`

3) added the line below to the bottom in the right position:

```
  void lcd_iniciar() {

  }
```

Of course, I am still just trying to compile and make this option appear in my menu. After that I will try to make it execute some arbitrary GCODE. BUT When I try to compile the code I got many errors in my console. I attached ultralcd.ccp file so you can check it. I didnt change any other thing at all, just added the 3 lines above cause they made sense.

```
sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `lcd'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

z:/desktop/copie somente o atalho - arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `lcd_sdcard_pause()'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `lcd_sdcard_resume()'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `watch_temp_callback_E0()'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `watch_temp_callback_bed()'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `_reset_acceleration_rates()'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `_planner_refresh_positioning()'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `lcd_sd_updir()'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here
[... 1026 lines cut ...]

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `lcd_control_temperature_preheat_material1_settings_menu()'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `lcd_control_temperature_preheat_material2_settings_menu()'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\ultralcd.cpp.o (symbol from plugin): In function `lcd_iniciar()':

(.text+0x0): multiple definition of `_lcd_move_distance_menu(AxisEnum, void (*)())'

sketch\ultralcd - Copy.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

```

batata004, 2017-05-26:
@Tannoo with your help I could finally add an entry to the very first position of my LCD \O/

Now I have one final question: I declared a function to be executed when this option is selected and I want to know what code should I put inside this function in order to it run a specific GCODE?

Tannoo, 2017-05-26:
This will cause the maching to home, wait for bed to get to temp, then start bed leveling.
```
      void my_function() {
        char some_variable[16];
        snprintf_P(some_variable, sizeof(some_variable), PSTR("G28\nM190 S50\nG29"));
        enqueue_and_echo_command(some_variable);
      }
```

batata004, 2017-05-26:
@Tannoo thanks your code works great! But why cant I just use the code below?

```
  void lcd_iniciar() {

    enqueue_and_echo_commands_P(PSTR("M140 S105\nG29"));

  }
```

I still dont know how to make the LCD comes back to the info screen as soon as I select the optoin in the lcd.

Tannoo, 2017-05-26:
I think this will work.

`snprintf_P(some_variable, sizeof(some_variable), PSTR("G28\nM190 S%i\nG29"), PREHEAT_2_TEMP_BED);`

Tannoo, 2017-05-26:
>   enqueue_and_echo_commands_P(PSTR("G28"));

This does not display anything on the screen. This outputs to the command buffer and the serial output.
There is nothing to display. Not even to draw a blank screen.

batata004, 2017-05-26:
@tanoo thanks! But its really strange cause I only managed to make the LCD screen to come back to the info screen if I use this:

lcd_return_to_status();

If I dont use this and I only use the enqueue_and_echo_commands_P() the screen will get completely blank and after 20 seconds it comes back to the status screen.

Also, I am using this:

`enqueue_and_echo_commands_P(PSTR("M140 S105\nG28\nG29"));`

You told me I shoudl use this below? Isnt anything wrong?

`enqueue_and_echo_commands_P(snprintf_P(some_variable, sizeof(some_variable), PSTR("G28\nM190 S%i\nG29"), PREHEAT_2_TEMP_BED))`

Tannoo, 2017-05-26:
Post your full function.

batata004, 2017-05-26:
hi, it is:

  ```
void lcd_iniciar() {

    lcd_return_to_status();
    enqueue_and_echo_commands_P(PSTR("M140 S105\nG28\nG29"));

  }
```

But I would like the 105 temperatur to be retrieved from the PREHEAT_2_TEMP_BED.

Tannoo, 2017-05-26:
void lcd_iniciar() {
  enqueue_and_echo_commands_P(PSTR("M140 S%i\nG28\nG29"), PREHEAT_2_TEMP_BED);
  lcd_return_to_status();
}

Tannoo, 2017-05-26:
If that doesn't work. Use the snprintf_P() I posted earlier.

Tannoo, 2017-05-26:
Why don't you use this instead?

snprintf_P(some_variable, sizeof(some_variable), PSTR("G28\nM190 S%i\nG29"), PREHEAT_2_TEMP_BED);

```
void lcd_iniciar() {
  char temp_gcode[20];
  snprintf_P(temp_gcode, sizeof(temp_gcode), PSTR("M140 S%i\nG28\nG29"), PREHEAT_2_TEMP_BED);
  enqueue_and_echo_command(temp_gcode);
  lcd_return_to_status();
}
```

batata004, 2017-05-27:
@Tannoo the code below worked perfectly:

````enqueue_and_echo_commands_P(PSTR("M140 S%i\nG28\nG29"), PREHEAT_2_TEMP_BED);```` 

Thanks, you were very kind and helpful! Hope other people find this thread and be able to add a menu entry easily, it's very useful being able to customize the LCD display adding some macros/shortcuts!

closed 2017-05-27 by batata004

Tannoo, 2017-05-27:
We could put a `Custom Commands` on the main menu for just this type of thing.
(Enabled in `configuration_adv.h`, of course.)

emartinez167, 2017-05-27:
That is a brilliant idea! Great work guys!

Regards,

Ernesto.

batata004, 2017-05-27:
@Tannoo if you add this option I would love to use it!! You could just add an option to the user add a custom menu entry that runs some arbitrary GCODE. I think you guys putted so much effort in autog0 file structure and it would be much easier to allow a menu to be inserted and to run some gcode. Placing this menu entry right on the top of the others would be great and of easy acccess.

Roxy-3D, 2017-05-27:
How would this work out?
- Have a #define CUSTOM_USER_MENUS in Configuration_adv.h  With out this enabled, everything is just as it is today.
- With that enabled, At the very bottom of the first menu page, there is a 'User Commands' entry
- Within the 'User Commands' page, we present up to 4 or 5 different sequences that are defined in Configuration_adv.h (along with titles)

fiveangle, 2017-05-27:
Seems like it would be super useful but only if it could do things like return the text output of the command back to the LCD.  I'm always wondering what the current K value is for LIN_ADVANCE so I have to go back and check on the slicer I used.  Being able to report it with an M900 in my custom menu would be great.  I'm sure users would come up with a 1,000,0001 uses for such a feature 👍

Roxy-3D, 2017-05-27:
What I'm hearing you say is you want the LCD Panel to be available for generic output.   And possibly echo the standard output that is sent to the host machines in a portion of the LCD Window.   This will take some thought to work through.

Right now...  There is some thinking going on how to allow the main part of Marlin to use the Encoder Wheel and buttons in a clean fashion.  We should probably expand that thinking to also allow the main parts of Marlin to have easier access to the LCD Display.  (UBL grabs control of them and takes them away from the LCD Code.  This would be redone in a cleaner form.)

What would help is some clever thinking how the output of  SERIAL_ECHO() macros gets temporarily displayed on the LCD in addition to being sent to the host program.  Maybe it is something like when a custom command is executed, the LCD also displays all SERIAL_ECHO() data for 3 seconds, and then reverts to the Status Screen?   I don't know the answer.   But we would want some simple algorithm that was both backwards compatible and future safe.  (Obviously...  that might be too much to hope for.)

batata004, 2017-05-27:
wow that would be awesome! Just let me add one thing, if the user defines "CUSTOM_USER_MENUS " I think the "USER COMMANDS" should be displayed at first, for easy/shortcut access. Otherwise the user would have to scroll down the LCD till the very last option is reached. If the user does not want to use this CUSTOM_USER_MENUS it will keep everything as it is today.

Of course, adding the capability to display the output of the commands is great but I think you could start this in a easier way and then improve it.

My suggestion is this: when the user define the CUSTOM_USER_MENUS, he should assign a pair value of LABEL and GCODE. The label will the be the option label that will be displayed (some of you suggested 5 options max) and the gcode would be the gcode sequence that should be executed. I dont know if Arduino allows multidirectional arrays, but if so, this would be my suggestion:

`#define CUSTOM_USER_MENUS array(array("BED LEVEL","M109\nG28\nG29"),array("OTHER STUFF","GCODE STUFF"))`

After the user chooses an option, Marlin should show the status screen (the main screen where marlin shows temperatures, XYZ...) and at the very botom/left of the screen you could show the LABEL of the current GCODE that is being execute OR the current GCODE that is being executed.

Only later, you could add an option to display the result of the GCODE in the screen because I believe this would require much more thinking and clever solution to display data without outputing it to serial.

Tannoo, 2017-05-27:
Here is something to start with. #6875

Tannoo, 2017-05-27:
No arrays, yet... I don't know how to mess with multidirectional arrays.

I think that since this is an end user configuration item, maybe simplicity is best?

batata004, 2017-05-28:
@Tannoo right, simplificty is better! Maybe you could do this:

```
#define CUSTOM_USER_MENUS_LABEL_1 BED LEVEL
#define CUSTOM_USER_MENUS_GCODE_1 G28\nG29

#define CUSTOM_USER_MENUS_LABEL_2 OTHER STUFF
#define CUSTOM_USER_MENUS_GCODE_2 M114

...
...
```

So the user can comment/enable only the custom menus he/she wants. This is the easiest it can get.

Tannoo, 2017-05-28:
Yes.. here:

https://github.com/Tannoo/Marlin/tree/user_command_menu

batata004, 2017-05-28:
Why cant I use this -> https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x ?

Tannoo, 2017-05-28:
You can, but it won't have this PR in it.

Tannoo, 2017-05-28:
Once (or if) the PR is merged, then everyone will have it in `bugfix-1.1.x`.

batata004, 2017-05-28:
Thanks man! I am starting to learn about git recently! Could you tell me why didnt you merge it to the main branch of bugfix-1.1.x?

And if I download your PR, will it contain other's people most recent contribution or just your new feature on top of the latest stable version?

Tannoo, 2017-05-28:
My brach is a "snapshot" of when I copied the latest `bugfix-1.1.x`, then I made changes. I can update that branch with any new changes.

The PR branch is based on `bugfix-1.1.x as of yesterday afternoon.

I don' have the permisstions to merge any PR into Marlin.

PRs are usually generated to get feedback and testing before merging.

batata004, 2017-05-28:
Great explanation! So I took the latest RCBUGFIX and migrate all my settings to the latest RCBUGFIX. After that I copied the file ultralcd.cpp, language_en.h and configuration_adv (from your repository) and added them to my RCBUGFIX. I compiled and it worked. 

I will try it now on my printer. Just one advice: if I enable custom menus I need to enable all 5 custom menus! Cant I only use one custom menu and not use the others? Cause if I comment all the other 4 menus and keep only the first one the compiler says errors. This way I could save memory to use with other things instead of loading 4 menus that I will never use.

batata004, 2017-05-28:
@Tannoo I just tested your version and it worked perfectly! Amazing! I added a custom command like this:

```
  #define USER_DESC_2 "Second Script"
  #define USER_GCODE_2 "M140 S105\nG28\nG29"
```

And it worked! But now I have to ask you again (cause I think the solution you pointed earlier will not work here): how do I replace the 105 value above with PREHEAT_2_TEMP_BED ?

I liked that the custom commands appeares at the TOP! I liked that when I clicked the option after a few seconds it comes back to the status page! Is it possible to make it come back right away to the status page after I select a custom command to be executed? Cause sometimes it takes 5 seconds, other times it takes 10 or even 20 seconds to come back to status page.

Tannoo, 2017-05-28:
That would take more code... I was trying to keep this as a simple custom script option.

But, evolution does happen...

Tannoo, 2017-05-28:
>But now I have to ask you again (cause I think the solution you pointed earlier will not work here): how do I replace the 105 value above with PREHEAT_2_TEMP_BED ?

You are correct... give me some time.

batata004, 2017-05-28:
thanks, I hope you could have taken that nap you said in another thread! You are doing a great job man, thanks a lot!

AnHardt, 2018-09-25:
ultraldc.cpp calls `lcd_implementation_status_screen()` in, depending on the display type included `ultralcd_impl_HD44780.h` or `ultralcd_impl_DOGM.h`. There in you can find cals to `lcd.setCursor` and `_draw_axis_value();`. Therein you'll find `lcd.print()` and `lcd_print()` what is a UTF8 wrapper for `lcd.print()`.
So finally everything ends in one of the lcd-libraries. For the graphic displays in u8glib.
If you display has a 20x4, 16x4 or 16x2 HD44780 type display, or a 128x64 graphic display, all you need is a wrapper for your displays library. For others you have to start at the (`ultralcd_impl_HD44780.h` or `ultralcd_impl_DOGM.h`) level.

luckyonshu, 2018-09-26:
Thanks for helping AnH :) The function I was actually looking for is lcd_move_get_x_amount, y, z, etc... :) It's nice when a community does it's best to help out :) Thanks again. Lucky.