MarlinFirmware/Marlin #6861: new feature request: PID nozzle output consider object fan PWM

khschmidt, 2017-05-26:
if the object cooling fan increases from for example 0% to 100% (for bridges) the nozzle Temperatur drops down about 7-9 degrees. The PID needs very long to compensate this influence. But it is no problem to consider this influence. The output of the nozzle PID-PWM adds a correcting part Kp*FanPWM; the sum of both is the new PWM for the nozzle heating.

oysteinkrog, 2017-05-26:
One could have two PID tunings, one for 0% fan and one for 100% fan, and then simply interpolate between them depending on the fan speed?

Roxy-3D, 2017-05-26:
Yes.  That would be a  good start to correcting the switch over.   But the thing about PID is it has a 'history' of what the errors are.  Is there a way to easily map the already acquired history to the new settings?   Because if we need to acquire new data to get the algorithm in alignment, we still have an error during the transition.

fiveangle, 2017-05-27:
The Derivitive and Integral could be jettisoned on PWM change so the PID loop will begin aquiring on the new dataset (using the new interpolated PID<sub>0</sub> - PID<sub>255</sub> loop drivers.  Interesting idea but I wonder how much more effective these hints would be ?  I guess it might possibly be an improvment for seriousluy over-fanned printers, and not much at all for anemic ones (like my Printrobot Simple Metal ;)

What I've found is that the auto-tuning process doesn't work that well and most issues people have with PID not working right is more likely poor auto-tune PID values.  For some reason it always seems to want to keep P and I too low, and D too high.

boelle, 2019-03-12:
@thinkyhead this one needs the feature request label

boelle, 2019-06-22:
missing correct label

Haschtl, 2019-08-13:
I just added a PID_FAN_SCALING feature in my Marlin fork.
Currently it just increases the PID-Output with a new factor Kf. (pid_output += fan_speed * Kf)

Here is the main commit with the needed changes:

https://github.com/Haschtl/Marlin/commit/83359768a68e4a796a0ce1e22b5f08aedd39704a

But due to some errors (while copying from other repo), it needed two small fixes:

https://github.com/Haschtl/Marlin/commit/6264d669b0ea08e804e85ac35514ff2a04249b3b
https://github.com/Haschtl/Marlin/commit/816583156589d649afb55789831c7118c5104fec

This is not tested at all yet!! I will test it in the next time and post the results here. 

The main purpose of this is to feed-forward the power-losses due to the fan. 
I currently assume, that the correlation of power-losses and fan-speed is linear without a time-delay. Therefore it should be fine to increase the output-power linear too.

Example calculation:
- Heating cartridge with 30W
- Increase of losses due to fan speed at 100% (255) is 5W (estimated)

Heating needs to be increased by 5W:
- Kf = 5W/30W = 0.167 

Again: This is not tested at all, but feel free to test it.

Haschtl, 2019-08-27:
I have now tested this feature and it's working pretty good!
[image]

I just increased the Kf-value slowly and switched the fan on (100%) and off until the temperature drop is almost zero. For me Kf=0.3, but my two partfans are pretty aggressiv. Kf will be lower, if the partfan has less influence on the hotend-temperature.

My hotend for reference:
- 40W 24V Hotend (Ultimaker2)
- 2x 12V Fans 
- PT100 Thermistor with E3D Amplifier

Haschtl, 2019-08-28:
The only problem with this method is that the airflow of the object is not linear dependend to the PWM-value. The Kf-value is only valid for one fan-speed. 
But as the normal usecase is to enable the fans after the first layer to a fixed speed, i will not take any non-linear effects into account.

AnHardt, 2020-07-08:
This is done.

closed 2020-07-08 by ellensp