MarlinFirmware/Marlin #6862: G29 not working correctly?

Xserxes, 2017-05-26:
Hi,

i am using a hictop 3dp11 with marlin 1.1.1. I have attached an probe (LJ18A3-8-Z/BX -Yellow one from china) and it was working good long time. I have now updated to marlin 1.1.1 and getting issues when leveling (.
My normal procedure is to set the z-offset first. Therefore iam doing an G28 followed by G1 Z0, measure with paper and set m851 z-x.y (x+y are numbers). When it fits, i am finisched.
My gcode contains an g28 followed by g29.

Is ther anything else i must do newly?

Because if i drive in the front left corner (G28->G29->G1 x50 y40 z15-> G1 z0) , it drives into the bed. At the G28 messurement point, everything is ok.It seems not to apply the G29 results or in a wrong way. 

Can anybody confirm this or can tell me what i am doing wrong?

Thank you

Kind regards

Olaf

Roxy-3D, 2017-05-26:
I think we need the Bi-Linear label.

Xserxes, 2017-05-26:
It is already configured for bi-linear.

Roxy-3D, 2017-05-26:
I meant to label this thread as a bi-linear issue.

JurikRistjouw, 2017-05-28:
I had a similar issue, i found out that the printer was using EEPROM settings, which overrule the ones set in the config.h. If that is the case, try to clear those settings after flashing with M502 and then save with M500. After this, use caution when lowering or homing. Because the new settings could be way off compared to the EEPROM ones. Good luck

junglegymninja, 2017-06-11:
Are there any updates on this issue? I am also experiencing this with bi-linear leveling. I just increased my mesh to be 5x5 and it's still the same issue. I tired M502 and M500 immediately following a flash and it still persists. Similar to Xserxes my homing point in the center is ok, as I go left towards X0 it drives the extruder into the plate and as I go right towards X200 it lifts significantly. I've tried 3x3, now 5x5 grid. Manual leveling then G29. I still get a slope. Without G29 I can manual level just fine, with G29 I get a slope. Am I doing something wrong?

closed 2018-02-07 by thinkyhead