MarlinFirmware/Marlin #6863: [Request]LIN_ADVANCE K on eeprom

sarf2k4, 2017-05-26:
Hi, would it be possible to have the lin_advance K variable to be in the eeprom as well?

From the previous topic I've read, the K value differs from each filament type to another.

I also opened a feature request at [here](https://github.com/prusa3d/Slic3r/issues/319) so that the K values can be adjusted according to filament type to achieve a good print quality.

tcm0116, 2017-05-26:
> Hi, would it be possible to have the lin_advance K variable to be in the eeprom setting display as well?

That should be in 1.1.0 as it was implemented in #6368. It should be available under the Control/Filament menu when LIN_ADVANCE in enabled in Configuration_adv.h

> the K values can be adjusted according to filament type to achieve a good print quality

Keep on mind that Marlin doesn't know what filament you're printing with. That's the slicer's deal. However, I agree with you and had actually submitted a similar feature request with Prusa for their Slic3r fork (https://github.com/prusa3d/Slic3r/issues/265).

fiveangle, 2017-05-26:
Probably good to try the lastest version before submitted requests ;)

From bugfix-1.1.x:

```
Send: M501
Recv: echo:V37 stored settings retrieved (453 bytes)
Recv: echo:  G21 ; Units in mm
Recv: 
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X20.00 Y20.00 Z507.00 E96.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X250.00 Y250.00 Z8.00 E80.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1400 Y1400 Z200 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1000.00 R3000.00 T1500.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_ms> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Y8.00 Z1.00 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z5.00
Recv: echo:PID settings:
Recv: echo:  M301 P30.00 I5.00 D50.00
Recv: echo:  M304 P270.00 I30.00 D550.00
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-0.90
Recv: echo:Linear Advance:
**Recv: echo:  M900 K0.00 R0.00**
```

I still need to get to submitting PR to change default K to be 0 (like I've set in mine) because choosing some arbitrary K (current default i think  is 50 ?) is no better than off.

Please close if this resolves your question !

tcm0116, 2017-05-26:
@fiveangle I agree that having the default K set to 75 is rather arbitrary and that 0 would be better. I've done the same as you in my Configuration_adv.h file.

sarf2k4, 2017-05-26:
Thank you, i was still using 1.1.0 and havent checked on the latest 1.1.1
yet. I will close this thread when i have the chance to do so.

So to disable the lin advance, just set the k to 0 right?

tcm0116, 2017-05-26:
>> So to disable the lin advance, just set the k to 0 right?

Yes

sarf2k4, 2017-05-26:
Ok, thank you for the help and feedback, i should migrate over to 1.1.1 as
soon as i can.

A bit of off topic, i am also wondering if it is possible to have an extra
persistentconfig.h that will overide the default config.h if a variable
were defined in persistentconfig.h?

fiveangle, 2017-05-27:
No, but would be trivial to write an awk, sed, perl, etc script to pre-pre-process your Configuration*.h, but unless you are tracking Bugfix at a monthly cadence or so, the time savings seems dubious compared to the potential issues that could arise.  Best to cozy up to Diff/BeyondCompare/WinMerge/etc to migrate Configuration*.h's on new build install.

sarf2k4, 2017-05-27:
Hopefully this is the last one, ive been using npp compare but the compare
plugin seems broken now. Any recommended visual comparison software to use
aside from what ypu have listed?

fiveangle, 2017-05-27:
Those are the best.  vim -d also is nice when you're stuck in a terminal :)  I personally have purchased BeyondCompare myself.  Remember to close this if there's no more action for anyone.  Thx

tcm0116, 2017-05-27:
+1 for BetondCompare

closed 2017-05-28 by sarf2k4