MarlinFirmware/Marlin #6864: Crashprobe and Homing

token47, 2017-05-26:
Short question: Is a crashprobe dangerous?

Long question:

Hi! I'm thinking about implementing a crashprobe on my Z endstop/probe. The kind that stays retracted until the printer goes to a certain spot / corner where it presses a pin which releases the probe from the magnet, the it goes probing and in the end the probe naturally retracts as the nozzle gets near the bed.

But suddenly I realized: it will become very dangerous to try to home the printer from the LCD or even sending a G28 or G29 from the host software (or clicking those "home" buttons), because the probe will not get deployed first. Is that right? The only safe way to home is through a script (be it a macro button or a start script on the begin of a print).

Searching for solutions to this problem I have found patches do LCD code to introduce that functionality on the homing menu option (but have found none to the sending of naked G28/G29 commands).

On the Configuration.h there is a z probe end script which would be perfect.... if it was a START script:

/**
 * Commands to execute at the end of G29 probing.
 * Useful to retract or move the Z probe out of the way.
 */
//#define Z_PROBE_END_SCRIPT "G1 Z10 F12000\nG1 X15 Y330\nG1 Z0.5\nG1 Z10"

Is there a best practice on this?

Thank you

Roxy-3D, 2017-05-26:
I believe this will work just fine.   The lower level probe code knows the state of the probe and handles that stuff independent of whether it is a G29  or G28 needing the probe.

You can raise the nozzle halfway up your Z-Axis and pull the power to force a clean reset.  And then do a home via the LCD panel and I think you will find it raises the Z enough to deploy the probe and then tries to probe.    You should have plenty of time to pull the power  (or press Reset) if the wrong thing is happening.

token47, 2017-05-26:
That was the kind of behavior that I was expecting. But I'm not sure where in the configuration I tell how to deploy the probe. Just to be clear, this is the kind of probe I'm talking about (in fact, it will probably be this very project):

https://www.youtube.com/watch?v=hqBF5s-aBCQ

The probe will only deploy if someone/something drives the carriage to the border in a specific location. It's easy to program that on the start script of a print, but I'm not sure where in the config I teach the firmware to do that.

Roxy-3D, 2017-05-26:
Oh!  After watching the video, I'm not sure my answer is correct.   How do you deploy (and retract) your probe?   The lower level probe code kind of assumes it can do things via the servo code.  And this isn't controlled that way.

token47, 2017-05-26:
To deploy: drive carriage to specific location, the pin will be pressed and the probe will "crash" (drop).

To retract: do nothing. Just trying to get close to the bed will press it back to it's resting position (a magnet will pick it up after a certain point).

As you see:

- it seems to be dangerous if not made through a script. So you can say that there is some support for it as "you can make it work if you want", but my initial question was more on the line of the dangerousness of the firmware not knowing anything about it, just the slicer (and thus leaving the possibility of you trying to home it from the LCD or a direct g28 command). My guess is this is how delta's "live with it".

- it will not use the servo functions. That's why the BL-Touch use the servo signals to drive a solenoid (even not being a servo), because that's about the only option they had without patching the code.

- therer is a Z_PROBE_END_SCRIPT on the configuration, but there is not an equivalent of "before G28 script", which would be the ideal place for putting G-Code commands for that.

I haven't really implemented the probe yet, I'm just researching how much of a trouble it could be.

Roxy-3D, 2017-05-26:
yeah...  It looks like if you declare yourself to be a  Z_PROBE_ALLEN_KEY type probe,  the  void run_deploy_moves_script() gets called...

But of course...  You are not really an Allen Key probe.

token47, 2017-05-26:
Interesting... very close! Not sure if there are other side effects. Will do some reading.

token47, 2017-05-26:
It seems to be only for deltas. But in general, mechanically speaking, it's the same kind of probe that I'm after. It deploys and retracts almost exactly the same way (in fact there is a variation of the one on the video I posted, that uses a real allen key as the pin, so I'm guessing that's it).

Yes.... reading on: http://marlinfw.org/docs/configuration/configuration.html#probe-type

""A retractable z-probe for deltas that uses an Allen key as the probe""

Any chances this gets allowed on cartesians? (in my case, a corexy).

Roxy-3D, 2017-05-26:
> Any chances this gets allowed on Cartesians? (in my case, a corexy).

Yes...   If tested and debugged code showed up...   It wouldn't be real hard to find somebody to merge it.
Ideally, it should conform to existing precedents.

closed 2018-02-07 by thinkyhead