MarlinFirmware/Marlin #6865: Aligning Bed left/right with only one endstop/probe

token47, 2017-05-26:
Hi

Is there a way to align the bed left/right using the two motors using only one endstop/probe?

It would go this way:

- home normally (G28)
- probe for bed leveling normally (G29)
- use the two motors to compensate for the horizontal differences, effectively correcting / zeroing it (on the Y axis there's not much you can do, but on the X you can leverage the two motors)
- adjust the matrix for the corrected bed or probe the bed again to get a new matrix

This would be WAY easier to implement than to put two endstops on the Z axis.

Bob-the-Kuhn, 2017-05-26:
I take it that you're running dual Z motors.  I really like the dual Z endstop function.

In UBL there's a grid command to do what you want.  Just send **G29 J** and it'll do a 3x3 probe, calculate a tilted plane and then tilt the entire mesh accordingly.  This assumes an existing mesh.

Roxy-3D, 2017-05-26:
> Just send G29 J and it'll do a 3x3 probe, calculate a tilted plane and then tilt the entire mesh accordingly. This assumes an existing mesh.

Or...  for backward compatibility...  You can do a G29 P0 to create a mesh full of 0.00's.  And just tilt that like the earlier version of G29 (Linear) still do...

For physically leveling the bed...  I do a G29 P1 U T    The U option  tells it to acquire points far away from the nozzle's starting point.   So especially if you start in the center of the bed, you get outside points probed first on your bed.  When you have enough points to see what needs to be done, you stop the probing and make adjustments.   Most beds are not really flat.  So it is a judgment call how to 'level' the bed.   The good news is, what ever you decide, when you probe the mesh for real...  It works very well.

token47, 2017-05-26:
Ok, I saw the "U" option:

U     Unlevel    Perform a probe of the outer perimeter to assist in physically leveling the bed.

It seems that it will only print the information, and will not act on the motors to correct the table. Is that right? I'm talking about doing the physical correction the same as with dual Z endstops, but using the probe instead (you would not even need the whole grid, just two probes on left andd right would be enough to align the bed, and then go on to the real G28 and G29).

The problem is that the motors dis align too easily when motors are off. Trying to remove the print from the glass is enough to do it. So the idea is to re-align it every new print, not just once manually.

boelle, 2019-02-18:
@token47 i think this one was added recently in bugfig 2.0

revilor, 2019-02-19:
I would support @boelle and say: isn't this the Z_STEPPER_AUTO_ALIGN feature?

boelle, 2019-02-19:
ahh that was the name, i tried to use it once but stopped using it since i have dual z endstops (top mounted) and also a inductive probe so it made less sense for me, but i knew there was an option

boelle, 2019-06-22:
we should close, feature has been implemented

InsanityAutomation, 2019-10-15:
@boelle @shitcreek this is done as well, please close. G34 works quite well.

boelle, 2019-10-15:
holy smokes, wonder if the issue list goes empty :-D

closed 2019-10-15 by boelle