MarlinFirmware/Marlin #6868: BT7272A board for FLSUN Prusa clones

hominidae, 2017-05-26:
Hello,

Any timeline on when the board/pin configuration for the pull request submitted here will be integrated?

https://github.com/MarlinFirmware/Marlin/pull/6033

fiveangle, 2017-05-26:
I think the team is too busy dealing with the influx of spurious do-nothing issue submissions that it's hard to find time to focus on the actual work ;)

thinkyhead, 2018-02-07:
```cpp
#define BOARD_RAMPS_PLUS_EFB   143    // RAMPS Plus 3DYMY (Power outputs: Hotend, Fan, Bed)
#define BOARD_RAMPS_PLUS_EEB   144    // RAMPS Plus 3DYMY (Power outputs: Hotend0, Hotend1, Bed)
#define BOARD_RAMPS_PLUS_EFF   145    // RAMPS Plus 3DYMY (Power outputs: Hotend, Fan0, Fan1)
#define BOARD_RAMPS_PLUS_EEF   146    // RAMPS Plus 3DYMY (Power outputs: Hotend0, Hotend1, Fan)
#define BOARD_RAMPS_PLUS_SF    148    // RAMPS Plus 3DYMY (Power outputs: Spindle, Controller Fan)
```

closed 2018-02-07 by thinkyhead