MarlinFirmware/Marlin #6869: Bug with LCD entry option

batata004, 2017-05-26:
Hi,

@Tannoo helped me in another thread to add a menu entry to the LCD screen and it worked perfectly. Now I have a problem that sounds like a bug.

In the submenu that I inserted in the LCD I made it execute this:

```
    lcd_return_to_status();
    enqueue_and_echo_commands_P(PSTR("M140 S105\nG28\nG29"));
```

After I click in the submenu item the LCD gets back to the status screen and the printer starts homing and heating. Ok! BUT now if I select PRINT FROM SD and select a file (while the bed is still heating) the printer waits, waits... till the temperature (105C degrees) is reached so it start printing.

This should not happen since the M140 command should not wait the temperature reach the desired value to execute other codes.

The behaviour that I want is nothing "weird", it already happens in marlin: turn on your pirinter, go to PREPARE and PRE HEAT ABS BED. Ok, the printer will start preheating! While that happens try to print any file in your SD card, it will start immediatly executing the GCODES inside the file, it will not wait for the printer to reach the final temperature.

Is this a bug or am I doing any thing wrong?

Tannoo, 2017-05-27:
I belive it was intentional to have the preheat to not hold up any other operations by the user. So, printing from SD before the bed has reached temp is normal.

If you are "pre-heating", you would wait before you start a print, correct?

I don't put cookies in the oven while it's pre-heating.
Some people might.

batata004, 2017-05-27:
@Tannoo  I agree with you and what you said in your first paragraph is what happens BUT that should also happens when using the GCODE -->> enqueue_and_echo_commands_P(PSTR("M140 S105\nG28\nG29")); <<--, it should stop executing M140 command when the user selects to print a file from the SD, BUT THAT IS not what is happening, it's blocking the printing till the preheat is done. I want to be able to stop this preheating if another preheating operation is told to the printer (like the one inside my GCODE file).

Tannoo, 2017-05-28:
It's possible that the SD routine is looking for activity and wait for it to be done.. heating, moving, etc... before it starts executing a file.

M140 can't be stopped because it executes once to change the setpoint of the bed and is done.

batata004, 2017-05-28:
with the new implementation of @Tannoo this bug does not happen. I have no idea why but this bug is solved :)

closed 2017-05-28 by batata004