MarlinFirmware/Marlin #6870: G30 appears to not use the probe offset in M851

jeremyherbert, 2017-05-27:
Hi,

I am having an issue with Marlin 1.1.0-1 (zipfile download). My probe offset is set using M851 to 4.8, and it prints 4.8 when Marlin starts up. If I run the following commands:

```
G28
G30
G30 S-1
``

I expect the machine to home, do a single probe and set the probe position to Z=4.8, and then move exactly to the probe point and print it. The probe point should be Z=4.8 at the end if I understand this correctly. However, when I do this, the first G30 appears to leave the Z axis unchanged (I think it should be doing something like `G92 Z4.8` implicitly), and the `G30 S-1` does not contain my probe offset. Am I using this command correctly?

jeremyherbert, 2017-05-28:
Just tried with the latest 1.1.1 release, same problem

fiveangle, 2017-06-04:
https://github.com/MarlinFirmware/Marlin/issues/5319#issuecomment-263590975

Consider doing what the last commenter in that thread said and submitting PR to fix the docs, since G30 looks to not work exactly like reprap wiki G30 (which focuses on other FWs beside Marlin)

closed 2018-02-07 by thinkyhead