MarlinFirmware/Marlin #6872: SINGLENOZZLE does not allow multiple extruders; sanity checks fail

gertmenke, 2017-05-27:
I am trying to get Marlin to work with my Diamond Hotend. I defined SINGLENOZZLE, because the Diamond uses a single nozzle, heater and thermistor for three extruders.
However, when compiling, I get:

<code>
In file included from MarlinConfig.h:39:0,
                 from Marlin.h:36,
                 from watchdog.cpp:23:
SanityCheck.h:1173: error: static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.
 static_assert(COUNT(sanity_arr_1) <= XYZE_N, "DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.");
 ^~~~~~~~~~~~~
SanityCheck.h:1174: error: static assertion failed: DEFAULT_MAX_FEEDRATE has too many elements.
 static_assert(COUNT(sanity_arr_2) <= XYZE_N, "DEFAULT_MAX_FEEDRATE has too many elements.");
 ^~~~~~~~~~~~~
SanityCheck.h:1175: error: static assertion failed: DEFAULT_MAX_ACCELERATION has too many elements.
 static_assert(COUNT(sanity_arr_3) <= XYZE_N, "DEFAULT_MAX_ACCELERATION has too many elements.");
 ^~~~~~~~~~~~~
static assertion failed: DEFAULT_AXIS_STEPS_PER_UNIT has too many elements.
</code>

However, I need to specify DEFAULT_AXIS_STEPS_PER_UNIT for all three extruders, because they are indeed different types of extruder (MK9 and Titan). I plan to switch the MK9 to a Titan soon, but it should still be possible to configure them separately, imo.

gertmenke, 2017-05-27:
PS: Some more infomation:
I am using Arduino 1.6.4 and Marlin's bugfix-1.1.x branch (currently 55c700d537d50bb06b6ced7c2ca61ffcaba7735a) for testing.
If you have any further questions, please ask.

gertmenke, 2017-05-27:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1033686/Configuration.h.txt)

Uploading .txt is allowed, but .h is not...

Tannoo, 2017-05-28:
Don't you need this for that type of setup?
```
/**
 * "Mixing Extruder"
 *   - Adds a new code, M165, to set the current mix factors.
 *   - Extends the stepping routines to move multiple steppers in proportion to the mix.
 *   - Optional support for Repetier Firmware M163, M164, and virtual extruder.
 *   - This implementation supports only a single extruder.
 *   - Enable DIRECT_MIXING_IN_G1 for Pia Taubert's reference implementation
 */
//#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
#endif
```

Tannoo, 2017-05-28:
Sanity check is correct, you have too many values set for these:
```
/**
 * Default Axis Steps Per Unit (steps/mm)
 * Override with M92
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_AXIS_STEPS_PER_UNIT   {160.0, 160.0, 2133.333333, 418.5, 418.5, 98.5}

/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_FEEDRATE          {120, 120, 20, 500, 500, 500}    // (mm/sec)

/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_ACCELERATION      {3000,3000,100,10000,10000,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for Skeinforge 40+, for older versions raise them a lot.
```

Tannoo, 2017-05-28:
I haven't messed with multiextruder setups, but I remember reading somewhere that the mixing part is handled elsewhere... not in the stepper defines.

Tannoo, 2017-05-28:
These should only have X, Y, Z, E. Or will have sanity check errors.

gertmenke, 2017-05-28:
Strictly speaking, the Diamond does not do mixing, at least not very well. (Or that's what I heard anyway, I have not tried that myself, yet.) So a setup where only one extruder is active at a time should do fine. It still has three separate extruders, though. And these extruders can have different DEFAULT_AXIS_STEPS_PER_UNIT.

As I understand it, there are currently four types:
Standard: One nozzle per extruder.
SINGLENOZZLE: One nozzle, multiple extruders, no mixing - use only the currently active extruder
MIXING_EXTRUDER: One nozzle, multiple extruders which can be active at the same time
SWITCHING_EXTRUDER: One nozzle, one extruder, one servo to switch filament

Slic3r does support multiple colors, but I don't know if it supports mixing. If it does, that might solve the problem for me. However, just selecting the color with T0, T1, T2, as you would with a traditional multiextruder setup (without having to bother with mixing ratios) would be a nice thing to have. In fact, a quick test shows that it does seem to work when I disable the sanity checks. Only the different values for STEPS_PER_UNIT on my extruders break things for me right now.

gertmenke, 2017-05-28:
PS: There's another sanity check:
<code>
/**
 * Only one type of extruder allowed
 */
#if (ENABLED(SWITCHING_EXTRUDER) && (ENABLED(SINGLENOZZLE) || ENABLED(MIXING_EXTRUDER))) \
  || (ENABLED(SINGLENOZZLE) && ENABLED(MIXING_EXTRUDER))
    #error "Please define only one type of extruder: SINGLENOZZLE, SWITCHING_EXTRUDER, or MIXING_EXTRUDER."
#endif
</code>

So it's either SINGLENOZZLE or MIXING_EXTRUDER; not both.

gertmenke, 2017-05-28:
D'oh, I didn't realize there's DISTINCT_E_FACTORS, apparently exactly for this purpose.
At least, Marlin compiles with this setting; I will check if it works as expected and then close this issue.

gertmenke, 2017-05-28:
Okay, that worked. Sorry for the confusion.

closed 2017-05-28 by gertmenke