MarlinFirmware/Marlin #6880: Heatbed temp in LCD?

sarf2k4, 2017-05-28:
Hi,

I noticed that the bed temperature control has been removed, was it intentional?

Without that, it would be impossible to change the bed temperature during mid-printing or to adjust the bed temperature according to the user, not relying on the presets from PLA/ABS (60/100)

If there is an option to enable this, where would they be?

christianh17, 2017-05-28:
Hi,
in my setting the heatbed is still displayed. So it can't be a general problem it must be a problem of your config. Can you post it here?

sarf2k4, 2017-05-28:
Here is my config.h. What i meant was in Control>Temperature>Bed, it was missing

[Configuration_2.txt](https://github.com/MarlinFirmware/Marlin/files/1034161/Configuration_2.txt)

christianh17, 2017-05-28:
Sorry, did not find the problem - no idea why it does not show the temp. I hope somebody else can help.

sarf2k4, 2017-05-28:
here is the picture telling that the bed temperature adjustment from Control>temperature>bed doesn't exist on my end
http://share.pho.to/Ai8sP

mtowara, 2017-05-29:
which version? Bugfix-1.1.x?
If yes do you know the specific commit?

sarf2k4, 2017-05-29:
I am using the base 1.1.1 release

mtowara, 2017-05-29:
try #define THERMAL_PROTECTION_BED
I don't know why, but the Code deciding to show this menu item seems to depend on this being set.

sarf2k4, 2017-05-29:
I'll try it first, I often disable the above option because my bed has a rather high resistance value compared to quality heatbed out there, so any changes in the ambient would trigger the bed protection.

If I were to use the bed temp adjustment feature from control>temperature>bed by at least enabling the thermal protection, it isn't a good idea.

sarf2k4, 2017-05-29:
By the way, I am trying to use the hex file and flash the board via
octoprint. I need some help in flashing the new firmware since I am using
ch340 mega2560 and kept getting timeout error

I was trying to flash via ssh on my windows platform

mtowara, 2017-05-29:
>I was trying to flash via ssh on my windows platform

flash via avrdude-ssh?

I don't know if octoprint handles this itself, but be sure to disconnect octoprint from the printer, so that the serial interface is available for flashing.

sarf2k4, 2017-05-29:
Actually use octoprint as a medium where I upload .hex into the octoprint, then use ssh for avrdude command to flash the board. I got it working now. It is actually due to the service didn't stop (had to issue another service stop for octoprint once again).

I also confirmed that by enabling the thermal bed protection option would have the bed temperature adjustment in the Control>Temperature>Bed.

Now, to my question, was this intentional, accidental or to force user to turn on the bed thermal runaway protection?

My heatbad has a pretty high resistance value and it can be quite hard to reach to target temperature depending on the surrounding ambient

mtowara, 2017-05-29:
I would think bug... 
@thinkyhead, could you have a look at Conditionals_post.h (WATCH_BED) and ultralcd.cpp?

fiveangle, 2017-05-29:
> I am using the base 1.1.1 release

And if using rcbugfix-1.1.x ?

sarf2k4, 2017-05-30:
I also tried on the latest bugfix-1.1.x, just downloaded and flashed with no bed protection. 

Still no bed temp adjustment

carlonb, 2017-06-05:
I tried to comment the lines at row 2361 and 2363 in file ultralcd.cpp and seems solved, but I do not know why this was related to the THERMAL_PROTECTION_BED.
I'm using 4x20 LCD.
`//    #if WATCH_THE_BED 
      MENU_MULTIPLIER_ITEM_EDIT(int3, MSG_BED, &thermalManager.target_temperature_bed, 0, BED_MAXTEMP - 15);
//    #endif 

Anyone have some more suggestion?

sarf2k4, 2017-06-05:
I think its better to change if 'bed_thermistor_type' != 0

With this, the bed option will always be there whenever you define bed
thermistor other than 0 which is not used

igortomiatti, 2017-06-27:
I have the same problem.

The option to change bed temperature on the LCD only appears after activating Thermal Runway.

Will this be fixed in the next versions?

Bob-the-Kuhn, 2017-06-28:
@igortomiatti - What flavor of Marlin are you running?

I just downloaded bugfix-1.1.x.  My LCD shows the bed temperature change menu whether `THERMAL_PROTECTION_BED` is enabled or not.

closed 2018-02-06 by thinkyhead