MarlinFirmware/Marlin #6883: enable DEACTIVATE_SERVOS_AFTER_MOVE with switching nozzle

christianh17, 2017-05-28:
I just tried my first PR. 

In my setting it is possible to deactivate the servo after nozzle switching. It kind of "snaps in" after switching so I do not need the power of the servo to keep the nozzle in place. As I can imagine there are others who do not want to have the noise of the servo all the time keeping the power I suggest to enable this option. 

Of course one has to know if it is a good idea for his / her printer.

It is PR: #6882 
Hope it gets approved,
regards
Christian

p3p, 2017-05-28:
Hey Christian, You have a lot of unrelated changes in the pull request so it can't to be excepted as it is, when submitting a PR its best to create a new branch with specifically what you want changed, there are a lot of pull request to review so keeping it as concise as possible would be very much appreciated.

christianh17, 2017-05-28:
Hey p3p. I just made a new PR, see #6887 It has the same content. Perhaps this one can be approved.
I learned a lot about github and unterstood I chose the wrong way before...

p3p, 2017-05-28:
That's much better, I can't say whether the change will be accepted but they will at least understand what you intended now. If you can close this issue and move any discussion about the PR into that PR comments that would be good and stop thinkyheads head exploding from the increasing issue count.

christianh17, 2017-05-28:
OK, I will close this issue and enter my explanation into the comment of the PR. Thanks for your hints!

closed 2017-05-28 by christianh17