MarlinFirmware/Marlin #6885: Redefinition errors after defining TEMP_SENSOR_1_AS_REDUNDANT

SturmGhost, 2017-05-28:
After defining TEMP_SENSOR_1_AS_REDUNDANT I get two redifinition errors. Latest release.

In temperature.h 183-186

` #if ENABLED(TEMP_SENSOR_1_AS_REDUNDANT)
      static int redundant_temperature_raw;
      static float redundant_temperature;
    #endif`

Redefinition of static float redundant_temperature; (commenting out fixed the problem)

And in temperature.cpp line 128-131

`#if ENABLED(TEMP_SENSOR_1_AS_REDUNDANT)
  int Temperature::redundant_temperature_raw = 0;
 float Temperature::redundant_temperature = 0.0;
#endif`

Redefinition of float Temperature::redundant_temperature = 0.0; (commenting out fixed the problem)

Bob-the-Kuhn, 2017-06-03:
It's been this way since at least RC8.

Strange seeing it in both the public & private areas.

I've initiated PR #6948 to make it only private.

Roxy-3D, 2017-06-03:
How can it be both public and private?  Shouldn't that cause a compiler error?

Bob-the-Kuhn, 2017-06-03:
It definitely is causing a compiler error now.

closed 2018-02-07 by thinkyhead