MarlinFirmware/Marlin #6890: Delta printer auto level problem

mbribeiro, 2017-05-29:
Hi, I am using Marlin 1.1.0 with ramps board in a delta printer. I am using an inductive proximity sensor.
I think it works because I have really bad value for height  (210mm when the correct is around 245) and after a (g29) probe the command g1 x0 y0 z0 works perfectly. 
The problem is all other distances away from the center has 1 mm or 2mm offset (the probe should compensate this). In some places (x-75 Z2) almost touch the bed and X75 Z0 is away for about 2mm. If after the g29 command, just above the last probed point, if I send a g1 z2 it almost hit the bed (strange because this point was already probed, it is not even interpolated). 
I am using 9 points of probe, double touch with really low speed in the first approach otherwise the hotend crashes the bed.
I am out of ideas and I would appreciate any help.

Roxy-3D, 2017-05-29:
Which bed leveling system are you using?

mbribeiro, 2017-05-29:
> Which bed leveling system are you using?
None, I just screwed the bed into the frame. I thought with auto level I would not need one.

Roxy-3D, 2017-05-29:
G29 is used to do software bed leveling...  And you are probing 9 points.   So you have  one of these enabled:
```cpp
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
```

mbribeiro, 2017-05-29:
G29 is used to do software bed leveling... And you are probing 9 points. So you have one of these enabled:

> //#define AUTO_BED_LEVELING_3POINT
> //#define AUTO_BED_LEVELING_LINEAR
> //#define AUTO_BED_LEVELING_BILINEAR
> //#define AUTO_BED_LEVELING_UBL
> //#define MESH_BED_LEVELING

Yes, for a Delta printer the only possible is AUTO_BED_LEVELING_BILINEAR. There is a sanity check for this.

Roxy-3D, 2017-05-29:
Ah....   https://github.com/MarlinFirmware/Marlin/pull/6695

mbribeiro, 2017-05-29:
> Ah.... #6695 
Ok I'll give it a try. Oh, BTW, in config.h for Kossel, the define
 #define FIX_MOUNTED_PROBE  is missing (was missing in 1.0.0 too). I had to use a diff tool from a previous version to figure it out what was preventing it to work.

I'll publish here the result of my test.

Thanks.

mbribeiro, 2017-05-30:
I tried Marlin 1.1.1 and still the same. I don't know if it is normal to use springs to adjust the bed level because I think mine is a little lifted for one side only (and it seems auto leveling does not compensate this).
I will attach 2 files, sending the command g1 z2 with x and y zeroes and a g1 z2 with y 50. As can be seen z2 with y 50 almost touch the bed.

[image]

[image]

LVD-AC, 2017-06-26:
give g33 a try

mbribeiro, 2017-06-26:
> give G33 a try

Sure, I will post here the results.
Thanks.

mbribeiro, 2017-06-26:
I tried G33. First time I sent g33 C -1 (as shown in the firmware) BTW I am using v 1.1.1
I lost the count by after this command the printer repeated the calibration process (homming and probing) a lot after stopping. This produces the best results, after that g1 x0 y0 z1 almost hit the bed but z0 does.
With only g33 the same g1 x0 y0 z1 hit the bed. For 0,0,0 g29 was better, the problem was on the edges. G33 fails on 0,0,0. 
I don't know if I did something wrong, I just uncommented 
#define DELTA_AUTO_CALIBRATION
and nothing more.

closed 2018-02-09 by thinkyhead