MarlinFirmware/Marlin #6892: Delta printer auto level problem

amizera, 2017-05-29:
I have delta with bltouch sensor and marlin 1.1.1, after marlin auto calibration i have printer leveld with paper and i can sucesfully print first layer

[image]

but when i use auto bad leveling before print results are strange

READ: Bilinear Leveling Grid:
READ:       0      1      2      3      4      5      6
READ:  0 -0.122 -0.122 -0.081 -0.087 -0.113 -0.097 +0.091
READ:  1 -0.048 -0.062 -0.101 -0.151 -0.202 -0.232 -0.187
READ:  2 +0.062 +0.043 +0.017 +0.018 -0.001 +0.022 +0.078
READ:  3 +0.072 +0.012 -0.014 -0.049 -0.079 -0.073 +0.048
READ:  4 +0.140 +0.089 +0.083 +0.052 +0.031 +0.059 +0.147
READ:  5 +0.123 +0.089 +0.010 -0.003 -0.056 +0.011 +0.237
READ:  6 +0.298 +0.177 +0.092 +0.099 +0.078 +0.227 +0.191

and print is not good
on bottom is to low, on top is to high (lines are not connected)
why ?

[image]

Roxy-3D, 2017-05-29:
You need to edit some of your mesh points.    I don't know how your bed is orientated...  But it looks like you have a row (or at least a line) of mesh points that are too low.

You should be able to move the nozzle to that very thin region (where the mesh point is too high) and do a G42 to go to the closest mesh point.   You can then use M421 Q -.1 to lower the mesh point.   (You will need I & J parameters on those commands)

amizera, 2017-05-29:
[image]
mesh points have Z values ?
or i should do it after G29 ?
measure point are:
READ: Bed X: -67.000 Y: -67.000 Z: -0.122
READ: Bed X: -45.000 Y: -67.000 Z: -0.122
READ: Bed X: -23.000 Y: -67.000 Z: -0.081
READ: Bed X: -1.000 Y: -67.000 Z: -0.087
READ: Bed X: 21.000 Y: -67.000 Z: -0.113
READ: Bed X: 43.000 Y: -67.000 Z: -0.097
READ: Bed X: 65.000 Y: -67.000 Z: 0.091
READ: Bed X: 65.000 Y: -45.000 Z: -0.187
READ: Bed X: 43.000 Y: -45.000 Z: -0.232
READ: Bed X: 21.000 Y: -45.000 Z: -0.202
READ: Bed X: -1.000 Y: -45.000 Z: -0.151
READ: Bed X: -23.000 Y: -45.000 Z: -0.101
READ: Bed X: -45.000 Y: -45.000 Z: -0.062
READ: Bed X: -67.000 Y: -45.000 Z: -0.048
READ: Bed X: -67.000 Y: -23.000 Z: 0.062
READ: Bed X: -45.000 Y: -23.000 Z: 0.043
READ: Bed X: -23.000 Y: -23.000 Z: 0.017
READ: Bed X: -1.000 Y: -23.000 Z: 0.018
READ: Bed X: 21.000 Y: -23.000 Z: -0.001
READ: Bed X: 43.000 Y: -23.000 Z: 0.022
READ: Bed X: 65.000 Y: -23.000 Z: 0.078
READ: Bed X: 65.000 Y: -1.000 Z: 0.048
READ: Bed X: 43.000 Y: -1.000 Z: -0.073
READ: Bed X: 21.000 Y: -1.000 Z: -0.079
READ: Bed X: -1.000 Y: -1.000 Z: -0.049
READ: Bed X: -23.000 Y: -1.000 Z: -0.014
READ: Bed X: -45.000 Y: -1.000 Z: 0.012
READ: Bed X: -67.000 Y: -1.000 Z: 0.072
READ: Bed X: -67.000 Y: 21.000 Z: 0.140
READ: Bed X: -45.000 Y: 21.000 Z: 0.089
READ: Bed X: -23.000 Y: 21.000 Z: 0.083
READ: Bed X: -1.000 Y: 21.000 Z: 0.052
READ: Bed X: 21.000 Y: 21.000 Z: 0.031
READ: Bed X: 43.000 Y: 21.000 Z: 0.059
READ: Bed X: 65.000 Y: 21.000 Z: 0.147
READ: Bed X: 65.000 Y: 43.000 Z: 0.237
READ: Bed X: 43.000 Y: 43.000 Z: 0.011
READ: Bed X: 21.000 Y: 43.000 Z: -0.056
READ: Bed X: -1.000 Y: 43.000 Z: -0.003
READ: Bed X: -23.000 Y: 43.000 Z: 0.010
READ: Bed X: -45.000 Y: 43.000 Z: 0.089
READ: Bed X: -67.000 Y: 43.000 Z: 0.123
READ: Bed X: -67.000 Y: 65.000 Z: 0.298
READ: Bed X: -45.000 Y: 65.000 Z: 0.177
READ: Bed X: -23.000 Y: 65.000 Z: 0.092
READ: Bed X: -1.000 Y: 65.000 Z: 0.099
READ: Bed X: 21.000 Y: 65.000 Z: 0.078
READ: Bed X: 43.000 Y: 65.000 Z: 0.227

Roxy-3D, 2017-05-29:
We need to know the index numbers for I & J at the bad spot.   Move the nozzle over to the thinnest part of the print using G42 I # J #    Probably that thin line is a row of mesh points and the nozzle will center over the thin area.   Once you know an index pair, you can move the mesh point up by doing M421 I# J# Q .1

amizera, 2017-05-30:
when i send e.g.  G42 I1 J1 nothing hapen on printer

I see that M421 Q option is for AUTO_BED_LEVELING_UBL, but i have AUTO_BED_LEVELING_BILINEAR
http://marlinfw.org/docs/gcode/M421.html - 
"[Q<linear>] A value to add to the existing Z value (Requires AUTO_BED_LEVELING_UBL)"

when i try chacnge to UB i have this error
#error "AUTO_BED_LEVELING_UBL does not yet support DELTA printers."

Roxy-3D, 2017-05-30:
I don't think you have the latest version of the code.   UBL is working on Delta's.   There are some corner cases that need more work like interactive editing of mesh points outside of the printable radius.  But  it is working...

closed 2018-02-07 by thinkyhead