MarlinFirmware/Marlin #6898: Where to set offset for Z-Endstop (Not Z-Probe)?

Ianmcmill, 2017-05-30:
I am confused. As so often when reading code comments.
Marlin 1.1.1.
In configuration.h
```
#define X_PROBE_OFFSET_FROM_EXTRUDER 10  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 10  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]
```
Does this also apply to an ordinary mechanical z-endstop when no z-probe is installed at all?
I don't see any other value but above cleary says "PROBE" and not endstop.
If endstop and probe offsets are different where am I supposed to set ENDSTOP offsets?

Bob-the-Kuhn, 2017-05-30:
They do not apply to endstops.

Endstops are wonderful!  Once you get them setup you'll love them.

Not quite sure what you mean by "set ENDSTOP offsets" so I'll go over the two most likely scenarios.

----

One scenario is when your bed is smaller than the area that the nozzle can reach.   In my case the bed is 65mm to the right of the home position and 60 mm behind the home position.  I also home the Z axis in the + direction.  In this situation X_MIN_POS, Y_MIN_POS & Z_MAX_POS set my offsets so that 0,0,0 is the lower left front corner of my bed.  Here's my settings.
```
  #define X_MIN_POS -65        // G28 sets X position to this value after homing
  #define Y_MIN_POS -60       // G28 sets Y position to this value after homing
  #define Z_MIN_POS 0        
  #define X_MAX_POS 280
  #define Y_MAX_POS 280
  #define Z_MAX_POS 425  // G28 sets Z position to this value after homing
```

----

The other scenario is getting the endstops in the correct location.  

To do this I like to use one of the M43 utilities.  I use the M43 E1 command.  You need to enable PINS_DEBUGGING in Configuration_adv.h to get to this command.

M43 E1 sits in the background and monitors the enabled/defined endstops for changes.  When a change happens it sends a message to the console and toggles the system LED.

What I do is 
1. Move the nozzle to where I want the home position to be.
2. Move the endstop until the LED toggles.
3. Back the endstop off until the LED toggles.
4. Slowly move the endstop until the LED toggles
5. Clamp/tighten the endstop in place.
6. Repeat for the other endstops.

The direction of the last endstop move is very important.  If I home my X axis to the left side then my final endstop move is to the right.

You should be able to get the endstop to within 5mm of ideal with this method.

Hmmm - I think I'll add a P option to the command so that the user can specify what pin the LED is attached to.

Ianmcmill, 2017-05-30:
Thank you. The first scenario is the scenario I am in.

Ianmcmill, 2017-05-30:
When my z endstop is set to min and z is triggered, the tip of the nozzle it is below the print surface, outside of the print bed. 
I would then need to set 
`  #define Z_MIN_POS 2` where 2 is the distance the tip of the nozzle is below the print bed?

Bob-the-Kuhn, 2017-05-30:
It would be -2 because you want Z0 to be above the home position.

vMeph, 2017-05-30:
Good to know this,im myself are starting to build my 3d printer from scratch, so all is new to me, while im still on the progress of building im already trying to understand things and sometimes i fell that its missing information on how things works with exemple of senarios for people that is starting, would be great to have a wiki for diferent subjects that would explain things  in a easy way that all could understand better , that would avoid also replying alot of times to the same questions i would say
thanks  @Bob-the-Kuhn for this senario explaining makes more sense now how to off set the z endstop

Bob-the-Kuhn, 2017-05-31:
@vMeph - if you haven't already, check out [Open Builds](http://www.openbuilds.com).  They're dedicated to home built machines.  An excellent source of info on the build portion of the machine.   That's where I got my start.

Ianmcmill, 2017-05-31:
My travel limits are now set like this
```
// Travel limits after homing (units are in mm)
#define X_MIN_POS -15
#define Y_MIN_POS -24
#define Z_MIN_POS -2.51 //updated
#define X_MAX_POS 200
#define Y_MAX_POS 200
#define Z_MAX_POS 200
```

After issuing **G28** and **M119** it returns this:
```
13:54:20.729 : Reporting endstop status
13:54:20.729 : x_min: TRIGGERED
13:54:20.729 : y_min: TRIGGERED
13:54:20.729 : z_min: TRIGGERED
```

However when issuing **M114** I get this:
`X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0`

Ianmcmill, 2017-05-31:
No no no. I still have a problem with the travel limit set in firmware that compiles and is being uploaded and the M206.

So I have set the travel limit in the firmware in the ARDUINO IDE and uploaded it.
When homing all axis and issuing M114 while having all three endstops triggered it says the tip is at 0,0,0 but it should know that it is at x-15, y-24, z-2,51. 
Therefore I said `M206 X-15.0 Y-24.0 Z-2.51` and now after homing M114 says "I am at -15, -24, -2,51".
When I now print something the whole print area is shifted by +15 +24 +2,51 which means it prints in the air. So it takes into account that after homing it is outside of the working area but when printing it ignores this and adds to what should be the real physcial 0,0,0 these values. Albeit I do not know from which part it takes these. From the compiled firmware sketch or from the issued M206 command.

Which of the setting should I remove? The travel limits inside Configuration.h or the issued M206 command.

Bob-the-Kuhn, 2017-05-31:
Do you have EEPROM support enabled?  If yes, do M502, M500, M501 & see if that fixes it.

We've all been bitten by the "made a config change but forgot to update the EEPROM" problem.

Ianmcmill, 2017-06-01:
EEPROM is enabled.
EEPROM settings were saved. Surprisingly not forgotten before. 
But I got the Z offset wrong. It is 0.6mm instead of 2.51mm. Now everything is fine.
Thank you.

closed 2017-06-02 by Bob-the-Kuhn