MarlinFirmware/Marlin #6900: Marlin 1.1.x Random kill printer halt

WOOKIE111, 2017-05-30:
Hello,

I have recently changed from Marlin Dev UBL to the new marlin 1.1.X from may 

and i seem to get random kill/halt happening.

also the button pin for the viki2 seems to be miss mapped as the click wheel is set as printer kill for some reason.

Electronics
Azteeg X3
Viki 2 LCD ( not the new style cable)

attached is the log from pronterface of what i saw. im not sure how to get more information

https://pastebin.com/tpBfK4SK

WOOKIE111, 2017-06-05:
Is it preferred that i edit with updates? or make new comments for each update?

**Update**
Been getting a temp halt in the exact same spot of a print in its first loop. temp halt only rises when start print is active, and doesnt activate when just holding temp outside of a print. and i have sliced the same file many times in different slicers and i get identical result.

will be swapping thermosister to be positive. but it multimeters just fine, but the fiberglass insulation on the v6 looks a little frayed from when i removed a glob of abs

thinkyhead, 2017-11-03:
Is this resolved? The helpful folks who handle issues have apparently been busy! Re-open if it persists.

closed 2017-11-03 by thinkyhead