MarlinFirmware/Marlin #6901: 1.1.2 error compiling w/MESH and LCD_BED_LEVELING enabled

WheresWaldo, 2017-05-30:
Compiles fine without LCD_BED_LEVELING, as soon as you enable LCD_BED_LEVELING  the following errors appear

> sketch\ultralcd.cpp: In function 'void lcd_bed_leveling()':
> ultralcd.cpp:1755: error: 'lcd_refresh_zprobe_zoffset' was not declared in this scope
>          MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
>   
> sketch\ultralcd.cpp:244:30: note: in definition of macro '_MENU_ITEM_PART_2'
>          menu_action_ ## TYPE(__VA_ARGS__); \
>  
> sketch\ultralcd.cpp:296:53: note: in expansion of macro 'MENU_ITEM'

There is more of the same within the error, only difference is enabling LCD_BED_LEVELING. Compiler was v1.8.2 on Windows

Bob-the-Kuhn, 2017-05-31:
I couldn't get it to fail.

Please ZIP your config files & post them.  I have a much better chance of reproducing the problem if I have the config files.

Roxy-3D, 2017-05-31:
So I don't know the answer.... But does turning off
```cpp
  //#define BABYSTEP_ZPROBE_OFFSET // Enable to combine M851 and Babystepping
```
help?   Your error messages sound suspiciously close to what that does...

WheresWaldo, 2017-05-31:
@Bob-the-Kuhn Zipped and attached. Please note that there are four different printers all named the R1 so there are extra entrees in the steps section, I just uncomment the one I need depending on the flavor of R1.
Also comment the show bootscreen or you will get an error.[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1040275/Configuration.zip)

@Roxy-3D Babystepping was disabled, if I enable it I get a different error. 
If babystepping is enabled but not LCD_BED_LEVELING or Babystep to Z offset, it compiles fine.
If babystepping is enabled and BABYSTEP_ZPROBE_OFFSET is enabled I get the following error

>sketch\ultralcd.cpp: In function 'void lcd_babystep_zoffset()':
> ultralcd.cpp:972: error: 'class Planner' has no member named 'abl_enabled'
>              if (planner.abl_enabled)
>  
> exit status 1
> 'class Planner' has no member named 'abl_enabled'

If Babystepping and LCD_BED_LEVELING is enabled and BABYSTEP_ZPROBE_OFFSET is not the error is 

> sketch\ultralcd.cpp: In function 'void lcd_bed_leveling()':
> ultralcd.cpp:1755: error: 'lcd_refresh_zprobe_zoffset' was not declared in this scope
>          MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
> 
> sketch\ultralcd.cpp:244:30: note: in definition of macro '_MENU_ITEM_PART_2'
>          menu_action_ ## TYPE(__VA_ARGS__); \
> 
> sketch\ultralcd.cpp:296:53: note: in expansion of macro 'MENU_ITEM'
>    #define MENU_ITEM_EDIT_CALLBACK(type, label, ...) MENU_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)
> 
> sketch\ultralcd.cpp:1755:9: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'
> MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
> 
> ultralcd.cpp:1755: error: 'lcd_refresh_zprobe_zoffset' was not declared in this scope
>           MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
> 
> sketch\ultralcd.cpp:248:102: note: in definition of macro '_MENU_ITEM_PART_2'
>          lcd_implementation_drawmenu_ ## TYPE(encoderLine == _thisItemNr, _lcdLineNr, PSTR(LABEL), ## __VA_ARGS__); \
> 
> sketch\ultralcd.cpp:296:53: note: in expansion of macro 'MENU_ITEM'
>    #define MENU_ITEM_EDIT_CALLBACK(type, label, ...) MENU_ITEM(setting_edit_callback_ ## type, label, PSTR(label), ## __VA_ARGS__)
> 
> sketch\ultralcd.cpp:1755:9: note: in expansion of macro 'MENU_ITEM_EDIT_CALLBACK'
>          MENU_ITEM_EDIT_CALLBACK(float32, MSG_ZPROBE_ZOFFSET, &zprobe_zoffset, Z_PROBE_OFFSET_RANGE_MIN, Z_PROBE_OFFSET_RANGE_MAX, lcd_refresh_zprobe_zoffset);
> 
> exit status 1
> 'lcd_refresh_zprobe_zoffset' was not declared in this scope

If all are enabled the error is 

> sketch\ultralcd.cpp: In function 'void lcd_babystep_zoffset()':
> ultralcd.cpp:972: error: 'class Planner' has no member named 'abl_enabled'
>              if (planner.abl_enabled)
> 
> exit status 1
> 'class Planner' has no member named 'abl_enabled'

Bob-the-Kuhn, 2017-05-31:
I finally got it to fail with the same message.  I turned on a Z probe in addition to LCD_BED_LEVELING and MESH_BED_LEVELING but with PROBE_MANUALLY disabled.

Turning off the Z probe and enabling  PROBE_MANUALLY give you a different error.

Turning off both the Z probe and PROBE_MANUALLY compiles when both LCD_BED_LEVELING and MESH_BED_LEVELING are enabled.

@Roxy-3D - oh guru of leveling ...  does having  PROBE_MANUALLY disabled in order to use LCD_BED_LEVELING and MESH_BED_LEVELING make sense to you?

Tannoo, 2017-05-31:
Is there a future edition ya'll are debugging?

WheresWaldo, 2017-05-31:
@Tannoo 1.1.2 was in Releases

WheresWaldo, 2017-05-31:
@Bob-the-Kuhn It never worked that way before, you can't enable FIXED_MOUNTED_PROBE and PROBE_MANUALLY at the same time and these printers have a FIXED_MOUNTED_PROBE.

BlackLightUK, 2017-05-31:
Possibly a related issue (at least it's a different compile error relating to PROBE_MANUALLY) 

When LCD_BED_LEVELING and MESH_BED_LEVELING are enabled, Marlin 1.1.2 compiles without error. 

When I enable PROBE_MANUALLY (with no other probe option enabled) I get a compile-time error of 
`lcd_wait_for_move not defined in this scope`
and points at ultralcd.cpp at line 1642

WheresWaldo, 2017-05-31:
@BlackLightUK & @Fixx  My understanding is that PROBE_MANUALLY is used if you don't have a probe. My machine uses the nozzle as a probe but it is unreliable for anything but homing, but since it is used for homing I have to enable FIXED_MOUNTED_PROBE. I then use MESH since it is easier to manually create a mesh with it than with UBL. So in my case PROBE_MANUALLY isn't what I want enabled. But it is interesting that the common denominator in all of these errors is LCD_BED_LEVELING.

BlackLightUK, 2017-05-31:
Just noticed this on Probes Configuration page ( [http://marlinfw.org/docs/configuration/probes.html](http://marlinfw.org/docs/configuration/probes.html) ) 

> “Mesh Bed Leveling” works in the same manner as “Bilinear Grid” but takes different G29 parameters. (This feature is superseded by combining the manual probe option with bilinear leveling, and will not be included in future versions of Marlin.)

Should we be moving away from Mesh and start using Bilinear Grid w/ PROBE_MANUALLY ?

WheresWaldo, 2017-05-31:
@BlackLightUK I think the desire is that everyone move to UBL, but like I said in my case UBL is very clunky since I don't want to auto-generate the mesh I just want to manually create it. Auto mesh generation makes double work since I have to manually edit most of it after the fact anyway. MESH is easier to use this way. I don't think that statement applies to any version of 1.1.x and may be 1.2.x and future versions.

I have tried a bunch of combinations of those three parameters and it errors out in most cases, as a point of reference, Bugfix-1.1.x from May 17th (the last set of files I downloaded) does compile without error. So whatever bug this is was introduced between May 17 and yesterday.

Tannoo, 2017-05-31:
UBL -- `G29 P4 R` to edit all points manually.

WheresWaldo, 2017-05-31:
@Tannoo Good to know, I will try it on my machine, but would still like to get to the bottom of this current issue.

BlackLightUK, 2017-05-31:
@Fixx - Good to know - thanks :)

Roxy-3D, 2017-05-31:
> does having PROBE_MANUALLY disabled in order to use LCD_BED_LEVELING and MESH_BED_LEVELING make sense to you?

No...   I think the future direction is a machine can have both a real Z-Probe and a 'Manual' Z-Probe.  That makes sense for a number of reasons.  So probably we want to start relaxing these constraints.

UBL brute forces the existence of a Manual Probe for G29 P2.  But we want to move that out G29 and into the Z-Probe sub-system.

> UBL -- G29 P4 R to edit all points manually.

UBL -- G29 P4 R to edit all points interactively.

WheresWaldo, 2017-05-31:
@Roxy-3D if G29 P4 R works the same way as the LCD choice for MESH then even though I want to see this issue addressed I will tell the people using these configurations to switch to UBL immediately. What I am asking is if I enable UBL and enable LCD_BED_LEVELING can I use the menu choice and the LCD to actually create the MESH manually?

thinkyhead, 2017-06-01:
> Should we be moving away from Mesh and start using Bilinear Grid w/ PROBE_MANUALLY ?

Yes, it replaces `MESH_BED_LEVELING` which is now marked deprecated.

thinkyhead, 2017-06-01:
`PROBE_MANUALLY` and `MESH_BED_LEVELING` are mutually-exclusive. 

`PROBE_MANUALLY` is a companion to `AUTO_BED_LEVELING_3POINT`, `AUTO_BED_LEVELING_LINEAR`, and `AUTO_BED_LEVELING_BILINEAR`.

Ideally, the code should just ignore `PROBE_MANUALLY` and `MESH_BED_LEVELING` being set together, since `MESH_BED_LEVELING` is the old version of `PROBE_MANUALLY`, but a sanity check could also be added to simply prevent these options being set together.

thinkyhead, 2017-06-01:
#6922 prevents untenable combinations of these options.

closed 2017-06-01 by thinkyhead

WheresWaldo, 2017-06-01:
I am not sure why this was closed, somewhere in the discussion the actual problem got sidetracked and the solution is not apparent. At no time did I test with PROBE_MANUALLY Forget everything written after the second response.

This is what causes the issue:
FIXED_MOUNTED_PROBE + MESH_BED_LEVELING + LCD_MENU_LEVELING
The result is the first error in the original post. If you remove LCD_MENU_LEVELING is compiles fine.

I don't see how setting setting PROBE_MANUALLY + AUTO_BED_LEVELING_BILINEAR + LCD_MENU_LEVELING is a direct replacement  since it doesn't compile either.

Fixx, 2017-06-01:
@WheresWaldo BTW; LCD_BED_LEVELING, BILINEAR_LEVELING, PROBE_MANUALLY compiles fine for me.

WheresWaldo, 2017-06-01:
@Fixx can you send me your Configuration.h so I can compare it to mine? Still though, I am trying to figure out how to duplicate the method of manually creating a mesh that worked just two weeks ago but doesn't work now.

Roxy-3D, 2017-06-01:
@WheresWaldo    Did you notice the new option:
```cpp
  #define UBL_MESH_EDIT_MOVES_Z     
```
But independent of that @Tannoo  has all of the important UBL commands available from the LCD Panel now.  He is trying to get an interactive LCD Mesh Editor workings...  (Instead of moving the nozzle or specifying on the command line where to edit...  He is working to get that done entirely from the LCD Panel.)

And I'm working on the bug where the nozzle goes way up when it is moved off the mesh.

WheresWaldo, 2017-06-01:
@Roxy-3D I saw that, 
I think you can see where I am going with this. I can't rely on auto-generating the mesh. Robo was clever in what they wanted to do with auto-leveling, it is the implementation that is poor, As a result the probing method makes the entire middle of the bed unusable if you rely solely on auto-probing. Hence the double work. Allow a completely manual approach similar to the way MESH and LCD_MENU_LEVELING work and it is usable and easily understood. No one in the group I deal with want to input a bunch of numbers. I don't even want to do that.

The technical reason (optional reading) the bed is big and thin, so when probing anywhere other than the four corners, where there is direct support for the bed, causes the bed to flex under the weight of the entire carriage. The flex is not equivalent and not even linear. So Robo users cannot rely on the auto-probing to get an even first layer. MESH solved this by allowing a grid to be created using the LCD and a feeler gauge. That was what I had hoped wouldn't be lost in the move to UBL. I understand that you want UBL to be totally automatic but please don't take away functionality some have come to expect. Also since the corner is a good place to home, I set up homing to go X0, Y0 and then max X/10 & max Y/10 is the location that it homes Z. Right in the corner where the support is and essentially no bed flex.

In the past I could use FIXED + MESH + LCD and it worked. That combination doesn't seem to work now.

Roxy-3D, 2017-06-01:
> I understand that you want UBL to be totally automatic but please don't take away functionality some have come to expect. Also since the corner is a good place to home, I set up homing to go X0, Y0 [...]

Actually...  You can do exactly what you want right now...    
- Center the nozzle in the middle of the bed.   Or else use the X & Y Parameter on the next command.
- Do a G29 P1 U     That will probe the outside region of the bed.   Allow the probing to go as far as you think is OK and it is not causing deflections.
- Do a G29 T   to see the mesh map
- Move the nozzle back to the center of the bed if it is not there.
- Do a G29 P3 C  +/-#.##   R ?? to fill in the center values with 'reasonable' values
- Do a G26 C P O2.5 to see what mesh points need to be adjusted.
- Move the nozzle to any area that needs correction.
- Do a G29 P4 R to edit those mesh points
- Do a G29 S1 to save the mesh.   
- Once the mesh is perfect...  Just use it...

WheresWaldo, 2017-06-01:
@Roxy-3D I am not trying to be contrary but please don't overlook:

> Allow a completely manual approach similar to the way MESH and LCD_MENU_LEVELING work and it is usable and easily understood. **No one in the group I deal with want to input a bunch of numbers. I don't even want to do that.**

I select a menu choice, press the encode button, it moves to point one I turn the encoder, when I am happy with the feel on the feeler gauge I press the encoder again. It moves to the next point and the process starts all over again. I don't see how your sequence duplicates that.

Roxy-3D, 2017-06-01:
Well....   Any mesh bed leveling system needs a mesh.   I was trying to explain how to get one in a fairly quick way and save it for future use.   

If you want to probe every point manually...  You can do that today with UBL by doing a G29 I 999 and then do a G29 P2 R 999

But manual probing is so painful...  It is better to let the firmware build as much of the mesh as it can for you.

WheresWaldo, 2017-06-01:
Yes I know manual probing is a pain, thank god you only do it once, Can I do exactly what you just said to do entirely on the LCD?

That is the whole issue with user experience. I have a controller and I want to use it, I shouldn't have to deal with any g-codes to make it work.

Please look at this from the perspective of a completely novice end user and not a programmer. I know it's your baby and I know you think it is just so simple why doesn't he get it, but see my question in the first sentence. FIXED + MESH +LCD does this, no numbers, no create mesh and try figure out where it off then edit, no remembering g-codes or parameters, no fuss no muss and just a slight inconvenience.

Roxy-3D, 2017-06-01:
I have not used all of @Tannoo  's  LCD work for UBL.   But he did a pretty extensive command set.  

In the build mesh menu...  One of the options is Fill-in Mesh.  It assumes you will be using a B-usiness card or shim under the nozzle.   You should be able to build the whole mesh that way if you want.   
- Home  (G28)
- Invalidate entire mesh.
- Fill in entire mesh manually...

```cpp
  void _lcd_ubl_build_mesh() {
      START_MENU();
      MENU_BACK(MSG_UBL_TOOLS);
      #if (WATCH_THE_BED)
        MENU_ITEM(gcode, MSG_UBL_BUILD_PLA_MESH, PSTR("G28\nM190 S" STRINGIFY(PREHEAT_1_TEMP_BED)
          "\nM109 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND) "\nG29 P1\nM104 S0\nM140 S0"));
        MENU_ITEM(gcode, MSG_UBL_BUILD_ABS_MESH, PSTR("G28\nM190 S" STRINGIFY(PREHEAT_1_TEMP_BED)
          "\nM109 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND) "\nG29 P1\nM104 S0\nM140 S0"));
      #else
        MENU_ITEM(gcode, MSG_UBL_BUILD_PLA_MESH, PSTR("G28\nM109 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)
          "\nG29 P1\nM104 S0"));
        MENU_ITEM(gcode, MSG_UBL_BUILD_ABS_MESH, PSTR("G28\nM109 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)
          "\nG29 P1\nM104 S0"));
      #endif
      MENU_ITEM(submenu, MSG_UBL_BUILD_CUSTOM_MESH, _lcd_ubl_custom_mesh);
      MENU_ITEM(gcode, MSG_UBL_BUILD_COLD_MESH, PSTR("G28\nG29 P1"));
      MENU_ITEM(submenu, MSG_UBL_FILLIN_MESH, _lcd_ubl_fillin_menu);
      MENU_ITEM(gcode, MSG_UBL_CONTINUE_MESH, PSTR("G29 P1 C"));
      MENU_ITEM(function, MSG_UBL_INVALIDATE_ALL, _lcd_ubl_invalidate);
      MENU_ITEM(gcode, MSG_UBL_INVALIDATE_CLOSEST, PSTR("G29 I"));
      MENU_ITEM(submenu, MSG_WATCH, lcd_status_screen);
      END_MENU();
    }
```

WheresWaldo, 2017-06-01:
@Roxy-3D Thank you for that simple explanation, that is something I can try and promote. I wasn't trying to be dense or difficult just wanted the machine to do work that it is capable of doing. I want the process to be idiot proof whether you can auto-generate or need to manually create everything. I will try this approach and see if it meets those needs. I have high hopes for UBL as I am certain you do to.

Tannoo, 2017-06-01:
>That is the whole issue with user experience. I have a controller and I want to use it, I shouldn't have to deal with any g-codes to make it work.

That is why I put together the UBL menu.

brucehvn, 2017-06-10:
In my case, I'm using an optical probe, but the printbite surface I'm using doesn't seem uniformly reflective so I don't know if the auto probing would work well.  Even just homing using the probe, I usually have to tweak the z-offset for each session.  So, I also prefer to create the mesh manually.  I'll try the UBL method Roxy described from the LCD and see if that works ok for me.  I had just updated to the 1.1.x branch and had everything set to start doing mesh bed leveling.   Then I saw that I didn't have the option on the LCD, so I enabled the define for lcd bed leveling and got the compile error described here.

WheresWaldo, 2017-06-10:
@brucehvn The current setup using the LCD doesn't work because of a small coding error in one of the menu choices. Until that is fixed and the files merged into 1.1.3 or whatever it is decided on to call it. Use the commands that @bgort provided, they would be something like this:
```
G28
G29 I 999
G29 P2 B
G29 S1
M420 S1
M500
```
These commands using the serial terminal, will let you create the mesh manually as well as save the mesh values and enable UBL mesh adjustment.

Roxy-3D, 2017-06-10:
@Tannoo has been working hard...  He has the original mesh leveling equivalent at the very top of the UBL menu now:  https://github.com/Tannoo/Marlin/tree/UBL_Mesh_Plotting    But we are trying to get the 20x4 displays to work with the interactive mesh editing when selected from the LCD Panel.  (It works fine from the host computer.)   

We have not merged his improvements yet.  But he has been doing frequent re-base operations against bugfix-v1.1.x so you are not going to be in uncharted water.    You might want to give that branch a try and soon, when we merge you can go back to bugfix-v1.1.x

adamsd5, 2017-09-17:
https://github.com/Tannoo/Marlin/tree/UBL_Mesh_Plotting
Dead link or the privacy settings don't let me see it.

Can't wait for UBL to be usable.  I spent 5 hours today and kept running into bugs or things that don't make any sense to me.  For example, the interactive adjustment of mesh points was impossible.  It seems like bugs, and I'll be happy to relate my experiences if they help (i.e. someone thinks I've found something new), but if not, I will leave @Tannoo to keep going with what seems like it will be a truly excellent set of features.  Until then, I hope I can auto-probe in Bilinear mode and do something to adjust the mesh manually.

Tannoo, 2017-09-17:
UBL is usable. The "Mesh Plotting" is already part of Marlin now. My branch has been deleted long ago.

Tannoo, 2017-09-17:
@adamsd5, you need to open a new issue and follow the template for the issue at hand.

adamsd5, 2017-09-19:
I will do so.  My apologies... I was very frustrated in my last post.  Thanks for your hard work and patience.  I expect I have found some bugs or just need to understand it all better.

Tannoo, 2017-09-19:
Use Marlin 1.1.x.