MarlinFirmware/Marlin #6902: Error compiling in gcode.h

BlackLightUK, 2017-05-30:
Hi there. 

I've just tried compiling 1.1.2 after tweaking the configuration.h and configuration_adv.h for my Tevo Tarantula - and arduino 1.6.8 threw up an " invalid conversion from 'const char*' to 'char*' [-fpermissive] " on line 138 of gcode.h

BlackLightUK, 2017-05-30:
There also appears to be a missing ) at the end of line 103 of the same file.

p3p, 2017-05-31:
Hey, I think you misread at line 103 (the preprocessor makes it less clear) though I can confirm the warning at line 138 with FASTER_GCODE_PARSER disabled, It's probably just a missing typecast.

Bob-the-Kuhn, 2017-05-31:
Try replacing line 138 with this:
`if (b) value_ptr = (char*) (DECIMAL_SIGNED(p[1]) ? &p[1] : NULL);`

That took care of a compile error in 1.1.x for me.

BlackLightUK, 2017-05-31:
Many thanks, @Bob-the-Kuhn - that sorted the compiler error. 
@p3p - yeah I misread the 103 bit - it was late !

closed 2017-05-31 by BlackLightUK

Bob-the-Kuhn, 2017-05-31:
PR #6914 implements this fix.