MarlinFirmware/Marlin #6904: [Observation] Small "Gotcha" with the new G-code parser and homing

FHeilmann, 2017-05-31:
Just noticed this today, so I thought I'd share:

The old parser accepted ```G28 XY``` to home X and Y axis. Notice the lack of a space between X and Y. The new parser will only home X with this command and requires a space to be inserted between X and Y like so: ```G28 X Y``` to home both axis.

Not necessarily a bug, thats why I included the "observation" tag.

Roxy-3D, 2017-05-31:
I think this is a 'bug' just because it will do something different depending on whether you put the X or the Y first.

fiveangle, 2017-06-02:
what it turn out that having to fix the tokenizer to recognize this throws out all the optimization gains ;)

Bob-the-Kuhn, 2017-06-03:
Take a look at PR #6940 & see what you think.

FHeilmann, 2017-06-12:
Above mentioned PR was merged. Closing

closed 2017-06-12 by FHeilmann