MarlinFirmware/Marlin #6905: 1.1.2 compile issue with  MENU_ITEM_CASE_LIGHT

DaveAhrendt, 2017-05-31:
Using the default config files with the distribution...
If I enable MENU_ITEM_CASE_LIGHT and CASE_LIGHT_PIN all will compile correctly until I enable AUTO_BED_LEVELING_UBL and all the required defines for UBL including BLTOUCH, EEPROM and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER I get a compiler error.

Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc"  -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560  -o "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611/Marlin.ino.elf" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\G26_Mesh_Validation_Tool.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\M100_Free_Mem_Chk.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\Marlin.ino.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\MarlinSerial.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\Marlin_main.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\Sd2Card.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\SdBaseFile.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\SdFatUtil.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\SdFile.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\SdVolume.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\blinkm.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\cardreader.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\configuration_store.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\dac_mcp4728.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\digipot_mcp4018.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\digipot_mcp4451.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\endstops.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\gcode.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\hex_print_routines.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\least_squares_fit.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\mesh_bed_leveling.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\nozzle.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\planner.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\planner_bezier.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\printcounter.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\qr_solve.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\serial.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\servo.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\stepper.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\stepper_dac.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\stepper_indirection.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\stopwatch.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\temperature.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\twibus.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\ubl.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\ubl_G29.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\ubl_motion.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\ultralcd.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\utility.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\vector_3.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch\watchdog.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\libraries\LiquidCrystal\LiquidCrystal.cpp.o" "C:\Users\Dave\AppData\Local\Temp\arduino_build_648611/core\core.a" "-LC:\Users\Dave\AppData\Local\Temp\arduino_build_648611" -lm
C:\Users\Dave\AppData\Local\Temp\ccW6QoXO.ltrans1.ltrans.o: In function `menu_action_setting_edit_bool':

C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch/ultralcd.cpp:3789: undefined reference to `case_light_on'

C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch/ultralcd.cpp:3789: undefined reference to `case_light_on'

C:\Users\Dave\AppData\Local\Temp\ccW6QoXO.ltrans1.ltrans.o: In function `menu_action_setting_edit_callback_bool':

C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch/ultralcd.cpp:3792: undefined reference to `update_case_light()'

C:\Users\Dave\AppData\Local\Temp\ccW6QoXO.ltrans1.ltrans.o: In function `lcd_main_menu()':

C:\Users\Dave\AppData\Local\Temp\arduino_build_648611\sketch/ultralcd.cpp:874: undefined reference to `case_light_on'

col

DaveAhrendt, 2017-05-31:
This issue is basically the same one I reported in #6888.  The define CASE_LIGHT_ENABLE is eventually used to define bool case_light_on in marlin_main.cpp.  Unless you crawl through the code you don't know that you need to manually define  CASE_LIGHT_ENABLE.

Changing configuration_adv.cpp to the following fixes the compile and work correctly:

// Define a pin to turn case light on/off
#define CASE_LIGHT_PIN 4           
#if PIN_EXISTS(CASE_LIGHT)
  #define INVERT_CASE_LIGHT false   // Set to true if HIGH is the OFF state (active low)
  //#define CASE_LIGHT_DEFAULT_ON   // Uncomment to set default state to on
  #define CASE_LIGHT_ENABLE         // ADDED DaveA 5/31/17
  #define MENU_ITEM_CASE_LIGHT    // Uncomment to have a Case Light On / Off entry in main menu
#endif

Bob-the-Kuhn, 2017-05-31:
This was just fixed/updated on PR #6797.

closed 2017-06-01 by thinkyhead