MarlinFirmware/Marlin #6916: Bl touch not homing and crushing to bed

wingsaber, 2017-05-31:
As I described on title. I used Tarantula 2. version and tried everything but Z not homing and crushing on bed  I have to stop it everytime and also sometimes it shows stopped and not Z homing again. I'm trying to figure it out for hours but nothing improved

Roxy-3D, 2017-05-31:
A couple of comments.  First: https://github.com/MarlinFirmware/Marlin/issues/6807

The second comment is this...   When I don't full trust things...  I raise the Z 150mm and do a G28.  I reach under the probe and verify it is doing the right thing several times.   With the Z raised 150mm I have enough time to react if it looks like it is going to crash the nozzle into the bed.

I don't know what a Tarantula-2 is.   Is that a BL-Touch type of probe?

wingsaber, 2017-05-31:
I'm sorry you can see from here tarantula https://github.com/JimBrown/MarlinTarantula. 

İt is working. I tried what you said. i put something else and it did. But when it comes to bed it is not going back. when I try to screw more it gives error. So I make it shorter idk what is problem with touch or firmware.

closed 2018-02-07 by thinkyhead