MarlinFirmware/Marlin #6917: UBL Z is odd at the corner

mcmatrix, 2017-05-31:
I'm using Fabrikator Mini V1.5 with UBL bed leveling (Aus3D IR Z Probe).
The firmware version is the latest release 1.1.2
Bed size is (X:0..80 Y:0..80)
G29 W
```
Unified Bed Leveling System v1.01 inactive.
No Mesh Loaded.
UBL object count: 1
planner.z_fade_height : 10.0000
zprobe_zoffset: -1.2500000
UBL_MESH_MIN_X  (0)=0
UBL_MESH_MIN_Y  (0 + 5)=5
UBL_MESH_MAX_X  (80 - 3)=77
UBL_MESH_MAX_Y  (80 - 10)=70
GRID_MAX_POINTS_X  6
GRID_MAX_POINTS_Y  6
MESH_X_DIST  15.40
MESH_Y_DIST  13.00
X-Axis Mesh Points at: 0.000  15.400  30.800  46.200  61.600  77.000  
Y-Axis Mesh Points at: 5.000  18.000  31.000  44.000  57.000  70.000  
Kill pin on :41  state:1

ubl_state_at_invocation :0

ubl_state_recursion_chk :0

Meshes go from 0x0248 to 0x0F7F
sizeof(ubl) :  1

z_value[][] size: 144

EEPROM free for UBL: 0x0D37
EEPROM can hold 23 meshes.

Unified Bed Leveling sanity checks passed.
ok

```
G29 T
```
Bed Topography Report:

(0,5)                                   (5,5)
(0,70)                                (77,70)
[ 0.000]   0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
(0,5)                                    (77,5)
(0,0)                                     (5,0)
ok

```

When bed leveling is inactive (G29 D) and executing:
G1 Z2
G1 X80 Y80
then it will execute normally.

When bed leveling is activated (G29 A) and executing:
G1 Z2
G1 X80 Y80
then near at the corner Z axis starts continually rise so that I have to kill the printer.

It makes no difference whether I have probed values on grid or empty with zeros.

Roxy-3D, 2017-05-31:
> Unified Bed Leveling System v1.01 inactive.
> No Mesh Loaded.

Your Topology Map says all the mesh points are 0.000.    

But your question has to do with the behavior when you go off the mesh.   I'm working on handling this nicely.   But the code isn't done yet.   In the mean time....   Don't go off the mesh.   Define the 
```cpp
UBL_MESH_MIN_X  (0)=0
UBL_MESH_MIN_Y  (0 + 5)=5
UBL_MESH_MAX_X  (80 - 3)=77
UBL_MESH_MAX_Y  (80 - 10)=70
```
such that it covers your entire bed.

Roxy-3D, 2017-05-31:
Please check back in a week or so...  I'll have fully tested improvements for this issue.

Roxy-3D, 2017-06-01:
@mcmatrix    I know the issue you are describing is real.   But I'm having trouble duplicating the failure.   Can you attach your Configuration.h (and Configuration_adv.h) files?  That would help me duplicate the failure!

mcmatrix, 2017-06-02:
@Roxy-3D In the meanwhile I was experimenting with other leveling options (3pos etc) and changed the configuration. Now when trying to recall the same config i had. I'm also not able reproduce this issue.
In the evening I should have more time for this, so then I'll try again to find the exact recipe

Roxy-3D, 2017-06-02:
Pretty much, I always have things setup this way:
```cpp
// If enabled, axes won't move below MIN_POS in response to movement commands.
//#define MIN_SOFTWARE_ENDSTOPS
// If enabled, axes won't move above MAX_POS in response to movement commands.
#define MAX_SOFTWARE_ENDSTOPS
```
I'm wondering if you also had the MAX_SOFTWARE_ENDSTOPS turned off when you were seeing the failure?

mcmatrix, 2017-06-02:
This is the UBL corner Z issue configuration. I'm able to reproduce it. Lets hope it's useful.
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/1048084/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1048083/Configuration.h.txt)

>EEPROM init from lcd menu.
>G28
>G29 S0
>G29 T
```
Bed Topography Report:

(0,5)                                   (5,5)
(0,70)                                (77,70)
[ 0.000]   0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
 
  0.000    0.000    0.000    0.000    0.000    0.000  
(0,5)                                    (77,5)
(0,0)                                     (5,0)
ok

```
>G29 W
```
Unified Bed Leveling System v1.01 inactive.
Mesh 0 Loaded.
UBL object count: 1
planner.z_fade_height : 10.0000
zprobe_zoffset: -1.2500000
UBL_MESH_MIN_X  (0)=0
UBL_MESH_MIN_Y  (0 + 5)=5
UBL_MESH_MAX_X  (80 - 3)=77
UBL_MESH_MAX_Y  (80 - 10)=70
GRID_MAX_POINTS_X  6
GRID_MAX_POINTS_Y  6
MESH_X_DIST  15.40
MESH_Y_DIST  13.00
X-Axis Mesh Points at: 0.000  15.400  30.800  46.200  61.600  77.000  
Y-Axis Mesh Points at: 5.000  18.000  31.000  44.000  57.000  70.000  
Kill pin on :41  state:1

ubl_state_at_invocation :0

ubl_state_recursion_chk :0

Meshes go from 0x0248 to 0x0F7F
sizeof(ubl) :  1

z_value[][] size: 144

EEPROM free for UBL: 0x0D37
EEPROM can hold 23 meshes.

Unified Bed Leveling sanity checks passed.
ok
```

>G1 Z3
>G29 A
>G1 X80 Y80

Near to the X80 Y80 Z starts rising

If you need something more then let me know.

Roxy-3D, 2017-06-02:
@mcmatrix 
> The firmware version is the latest release 1.1.2

Are you using: https://github.com/MarlinFirmware/Marlin/tree/1.1.x  ???

Does this mean you are not using:  https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x   ???

The reason is I'm trying all kinds of different things...  And I'm not seeing the problem.   But I'm doing the work in the bugfix-1.1.x branch.

mcmatrix, 2017-06-02:
Yes, i'm using release branch 1.1.x not bugfix-1.1.x
Sorry i forgot to mention that.

Roxy-3D, 2017-06-02:
May I ask you to try the bugfix-1.1.x version with the Configuration.h files that caused the problem?
When @oldmcg did the work to get UBL running on Delta's, I think he did a few things to make sure only valid mesh points were used in the Z correction calculations.

If that is true...   We probably should get those corrections moved over to the Release Version but in the mean time it might be best if you moved to the bugfix-v1.1.x version.

oldmcg, 2017-06-03:
The UBL_DELTA code should effectively extrapolate outside the outer mesh points, but I do need to test that.  For deltas the grid is usually bigger than the bed, but I can make a smaller grid and see what happens.  

Briefly looking at the UBL generic get_z_correction, it appears to return zero if coordinates outside a mesh cell or any invalid points.  We might be able to improve this by extrapolating from nearest mesh cell to edge similar to UBL_DELTA.

The simplest short term approach is probably to define a grid size with UBL_MESH_INSET zero so there are no printable points outside the mesh.  @mcmatrix, I would recommend trying with UBL_MESH_INSET zero and make sure all G29 T points are filled in.

There is a way to try the UBL_DELTA version of segmented lines on a cartesian machine, but it is untested as far as I know -- you'd be on the bleeding edge, and I'd recommend waiting for my pending PR with recent UBL_DELTA changes first.  To enabled this on cartesian machine, in addition to AUTO_BED_LEVELING_UBL, you would #define UBL_GRANULAR_SEGMENTATION_FOR_CARTESIAN.  I'd recommend re-homing after enabling UBL (G29 A) and again after disabling UBL (G29 D) for now as I suspect there are some lingering issues around this.

Roxy-3D, 2017-06-03:
Thanks @oldmcg !    If the other changes go smoothly...   I might try UBL_GRANULAR_SEGMENTATION_FOR_CARTESIAN and see what happens!

closed 2017-11-08 by mcmatrix