MarlinFirmware/Marlin #6918: Strange behavior on X after a G28 command

Bergerac56, 2017-05-31:
Hello
I found a strange behavior starting with the version of bugfix 1.1.x of 05/28 (till today) and not present in the bugfix 1.1.x of 05/26 and before.

Issue :
If option « define ENDSTOPS_ALWAYS_ON_DEFAULT » is inactive (commented) in configuration_adv.h (line 239) :
G28 homes the extruder as expected. (On my printer: Extruder is located in the center of the bed, at location X155, Y105, Z5)
G1 X0  => X stepper motor is activated to go to the position. When X reaches « 0 », the ENDSTOP X is activated but the motor seems to try to go further and « rebound » on the 0 position for around 1 or 2 seconds. This happens only on X and with or without other axes activated in the G1 (or G0) command. It happens even if a first move of X was done before (and after a G28) provided that it did not already reach “0” (where the ENDSTOP is located on my printer)

This behavior occurs only one time and only on X. The next moves seem to be normal. Obviously, if a new G28 command is issued, the “cyclus” starts again…
 Example of sequence:
G28
G1 X0 => Problem
G1 X100
G1 X0 => No problem
...
G28
G1 X50
G1 X0 => Problem
G1 X100
G1 X0 => No problem

If option « define ENDSTOPS_ALWAYS_ON_DEFAULT » is active (uncommented): The behavior is normal. No problem on X, Y and Z.

If I reverse to a version of Bugfix 1.1.x issued before the 05/27, the problem does not occur.

Is there a reason for that ?

Thanks

Roxy-3D, 2017-05-31:
It might be worth while to turn off:
```cpp
/**
 * Spend 28 bytes of SRAM to optimize the GCode parser
 */
#define FASTER_GCODE_PARSER
```
in Configuration_adv.h and see if the behavior changes.

Bergerac56, 2017-05-31:
Hello Roxy,

Sorry, I did not mention it. I tried with and without the "parse" option. Same behavior

Bergerac56, 2017-05-31:
I just checked the #6904. It is not the same thing. I do not use XY together. But it could be in the same area.

Bergerac56, 2017-06-09:
After the fix initiated here: #6993 the problem seems to be solved

closed 2017-06-09 by Bergerac56