MarlinFirmware/Marlin #6919: 1.1.2 resets on USB disconnect as well as connect, no functionality without USB

BlackLightUK, 2017-05-31:
MKS Base 1.4 board - I'm probably missing something in the config somewhere, but I can't find it. 

Previously I had 1.1.0 flashed onto the board; when disconnecting USB, the board would maintain the same state (and lcd functionality) until a new USB connection was attempted, at which point the board would reset. 

Now, when a USB disconnection happens (either slic3r or pronterface) the board resets - but loses all disconnected functionality at the same time (with the LCD completely covered in white boxes) 

Any ideas ?

thinkyhead, 2017-06-01:
I'm stumped. How the board behaves with respect to the USB connection is outside the purview of the firmware. Marlin has no control over that.

Tannoo, 2017-06-01:
The 5vdc rail overloaded?
Possibly bad/weak connection to the LCd?

BlackLightUK, 2017-06-01:
I'm thinking it has to be hardware at this point in time - I've just fired up the printer, connected to it and disconnected, and it's sitting there quite happy heating the nozzle and bed without an active USB link. 

Intermittant hardware issues. Deep joy.

closed 2017-06-01 by BlackLightUK

Andrei-Pozolotin, 2017-06-02:
try 
```
stty -F /dev/ttyACM0 -hupcl
```
to disable CTS -> RESET action