MarlinFirmware/Marlin #6920: The case light/Viki 2 problems again

mperdue, 2017-06-01:
The case light changes have broken the Viki 2 on the Azteeg X3 Pro again. Since I don't have a case light I've just commented out the block that changes the pin assignment.

thinkyhead, 2017-06-01:
@Bob-the-Kuhn This one's for you….

Bob-the-Kuhn, 2017-06-01:
I'm scratching my head as to how the changes affected Viki 2.

Please do a few things for me:
- ZIP up & attach your config files.  Maybe then I can re-produce the problem.
- Please detail what the problem is.  Is it not compiling, does it compile but the display is blank, ...
- Please detail what you did to fix the problem.

mperdue, 2017-06-01:
The code compiles and runs but the Viki 2 LCD screen is blank.

How the changes affect the Viki 2 is pretty easy - you're stealing a pin that is used as a standard connection to it on the AZTEEG X3 Pro.

In pins_AZTEEG_X3_PRO.h at/near line 33 you check to see if it is OK to reassign this pin for use with the CASE_LIGHT

`#if !PIN_EXISTS(CASE_LIGHT)         // doesn't already exist so OK to change the definition coming
  #define OK_TO_CHANGE_CASE_LIGHT   // in from from the include file
#
`
I haven't looked at what PIN_EXISTS() actually does but the idea seems to be that if the pin has been assigned for use already then return true. That's great but as near as I can tell the pin doesn't get assigned until line 37

`#include "pins_RAMPS.h"`

So the previous code will always determine that it's OK to resign this pin.

The problem with reassigning the pin is that the standard connection specified by Panucatt makes use of this pin so anyone with a Viki 2 and an X3 Pro will have it connected this way.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1044402/config.zip)

Bob-the-Kuhn, 2017-06-01:
Try this [file](https://www.dropbox.com/s/dx5juznyske5icx/pins_AZTEEG_X3_PRO.h?dl=0).  It only takes 44 for the case light if DOGLCD_A0 doesn't already have it.

I need to check the other pins_XXX.h files to see if there are any other similar oversights.

Could you point me to a date where Marlin didn't have this issue?  It sounds like this comes & goes.  I want to track down the changes & see where they're coming from.

mperdue, 2017-06-01:
I've got some stuff printing right now but I'll give it a shot as soon as that's done. I'll also try to determine the last file that worked without changes for me.

Bob-the-Kuhn, 2017-06-01:
I really need to get more sleep.

Please try these [files](https://www.dropbox.com/s/jyoyj98krkq5rou/pins_AZTEEG_X3_PRO.zip?dl=0).  They'll keep the pesky case light away unless it's enabled.

mperdue, 2017-06-02:
These files resolved the issue. At least they work for the case where you don't have a case light.  I haven't had a chance to determine where the problem originally occurred.

Bob-the-Kuhn, 2017-06-02:
That's more of a curiosity question.

The changes have been merged into Marlin.

closed 2017-07-07 by mperdue