MarlinFirmware/Marlin #6923: Z movement issue with bed leveling.

mperdue, 2017-06-01:
I mentioned this before in a different issue but I think it needs its own. On my printer after loading the firmware I can start up a print with no issues. The printer homes, then does my startup script which moves the extruder near the X0Y0 location, lowers to Z0.2 or so and prints two vectors at right angles to each other. From there the print is normal. 

After I do G29 and M500, there is a change in the way my print session starts. The home is normal then it starts the extruder toward X0Y0 but it also drops to about Z0.2 within a few mm of movement. Everything is normal from there. The problem with this is if there is any excess filament dripping from the extruder, it gets wiped off ring in the middle of the print bed. As you might suspect, this is less than desirable.

thinkyhead, 2017-06-01:
What form of bed leveling are you using?

mperdue, 2017-06-01:
Bilinear bed leveling

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1043479/config.zip)

mperdue, 2017-06-01:
After a bit of testing, this only happens when bed leveling is enabled with "M420 S1." As a work-around I've relocated this to just before I want the Z axis lowered. Previously it was right after the "G28."

tcm0116, 2017-06-06:
Where is your call to `G29` in relation to `G28` and `M420 S1`?

mperdue, 2017-06-06:
I probe an 11x15 grid so i don't do G29 for every print. I do it once and save the result then enable it with the M42 when printing.

The problem occurs when I execute the following at startup.
```
G28			                     ; Home
M420 S1			              ; Enable bed leveling
G92 E0.0			       ; Zero extruder
G1 X60.0 Y8.0 F2000		; Move to wipe start
G1 Z0.4			             ; lower Extruder
G1 X60.0 Y8.0 Z0.4 F1000.0	; Wipe Start
G1 X50.0 Y8.0 E4.0 F1000.0
G1 X8.0 E8.5 F1000.0		; Wipe #1
G1 Y60.0 E8.5  F1000.0		; Wipe #2
G92 E0.0			       ; Zero the extruder
```

Changing the sequence a bit works around it.

```
G28			                     ; Home
G92 E0.0			       ; Zero extruder
G1 X60.0 Y8.0 F2000		; Move to wipe start
M420 S1			              ; Enable bed leveling
G1 Z0.4			             ; lower Extruder
G1 X60.0 Y8.0 Z0.4 F1000.0	; Wipe Start
G1 X50.0 Y8.0 E4.0 F1000.0
G1 X8.0 E8.5 F1000.0		; Wipe #1
G1 Y60.0 E8.5  F1000.0		; Wipe #2
G92 E0.0			       ; Zero the extruder
```

thinkyhead, 2018-02-06:
I have seem some drift in the Z value in my testing, most likely due to the conversion to steps and then back to mm, which is used in any instance where the physical position is used to figure the logical position.
```
echo:Enable/disable Z drift?

  current_position=(15.00, 15.00, 5.00) : sync_plan_position
echo:Bed Leveling Off
echo:Fade Height 10.00

Raw Z after unapply_leveling 4.962500
  current_position=(15.00, 15.00, 4.96) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height 10.00

  current_position=(15.00, 15.00, 5.00) : sync_plan_position
echo:Bed Leveling Off
echo:Fade Height 10.00

Raw Z after unapply_leveling 4.962500
  current_position=(15.00, 15.00, 4.96) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height 10.00

  current_position=(15.00, 15.00, 5.00) : sync_plan_position
echo:Bed Leveling Off
echo:Fade Height 10.00

Raw Z after unapply_leveling 4.962500
  current_position=(15.00, 15.00, 4.96) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height 10.00

  current_position=(15.00, 15.00, 5.00) : sync_plan_position
echo:Bed Leveling Off
echo:Fade Height 10.00

Raw Z after unapply_leveling 4.962500
  current_position=(15.00, 15.00, 4.96) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height 10.00

echo:DEBUG:off

Raw Z after unapply_leveling 4.925000
echo:Bed Leveling On
echo:Fade Height Off

echo:DEBUG:INFO,ERRORS,COMMUNICATION,LEVELING

Fade Z drift?

Raw Z after unapply_leveling 4.962500
  current_position=(0.00, 0.00, 4.96) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height 10.00

Raw Z after unapply_leveling 4.925000
  current_position=(0.00, 0.00, 4.93) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height Off

Raw Z after unapply_leveling 4.943750
  current_position=(0.00, 0.00, 4.94) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height 20.00

Raw Z after unapply_leveling 4.925000
  current_position=(0.00, 0.00, 4.93) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height Off

Raw Z after unapply_leveling 4.937500
  current_position=(0.00, 0.00, 4.94) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height 30.00

Raw Z after unapply_leveling 4.925000
  current_position=(0.00, 0.00, 4.93) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height Off

Raw Z after unapply_leveling 4.987500
  current_position=(0.00, 0.00, 4.99) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height 6.00

Raw Z after unapply_leveling 4.925000
  current_position=(0.00, 0.00, 4.93) : sync_plan_position
echo:Bed Leveling On
echo:Fade Height Off
```

boelle, 2019-03-06:
@mperdue is the problem gone in latest bugfix 2.0?

mperdue, 2019-03-06:
I don't know. I needed to add several motors to my setup and moved to a Duet WiFi with a Duex5 so I'm running the firmware that they have developed. I still watch the development here because you never know what the future holds.

boelle, 2019-03-06:
nope, but i dont think its fair to keep this open when you are not able to test and you where pretty much the only one having the problem

but that just my thoughts

mperdue, 2019-03-06:
I agree. I wasn’t aware it was still open.

boelle, 2019-03-06:
its open until you click the close button below :-D

mperdue, 2019-03-06:
Done.

closed 2019-03-06 by mperdue