MarlinFirmware/Marlin #6924: 1.1.2 G33 Bed Radius implasuable

novio8, 2017-06-01:
Hi,
I'm struggeling with G33,
when I send G33 I get message M665 Bed radius implausible

I'v increased or lowered 
#define DELTA_PRINTABLE_RADIUS 130

but I'm always getting M665 Bed radius implausible

If I run Delta Calibration from LCD
I can only run set Delta Hight and X,Y Z separatelly
and not Auto Calibration

I'm stuck
Any advise will be most welcome

Regards
Mario

oldmcg, 2017-06-01:
If you have probe offset from nozzle, use this in your configuration.h if not already (all one line)

`  #define DELTA_CALIBRATION_RADIUS ((DELTA_PRINTABLE_RADIUS) * 0.869 - HYPOT(X_PROBE_OFFSET_FROM_EXTRUDER,Y_PROBE_OFFSET_FROM_EXTRUDER))
`

Also, if you have probe offset from nozzle, wait for pending change from @LVD-AC.

Then, if you use EEPROM after recompile/upload make sure to do M502, M500 so the new settings you compiled are used instead of what was previously saved in EEPROM.  This bites me regularly.

novio8, 2017-06-01:
thanx oldmcg,
I already find the same solution, a little more brutal: I'v just reduced calibr. ratius to 60
#define DELTA_CALIBRATION_RADIUS 60
and now it seems to work, or I just don't gett error message anymore
And I'm beaten every time I recompiling/uploading and forgettin that EEPROM holds
old memories
Regards
Mario

LVD-AC, 2017-06-01:
That is why you can change this as well with e.g. M665 B60

LVD-AC, 2017-06-01:
This issue can be closed!

closed 2017-06-01 by thinkyhead