MarlinFirmware/Marlin #6925: Compilation error with LCD_BED_LEVELING enabled (v1.1.2)

SCiunczyk, 2017-06-01:
Hi,

With version 1.12 I got compilation error: `'lcd_wait_for_move' was not declared in this scope` but only when:

```
LCD_BED_LEVELING enebled
MESH_BED_LEVELING enabled
PROBE_MANUALLY enabled

```
there is no problems with compilation if:
```
LCD_BED_LEVELING disabled
MESH_BED_LEVELING enabled
PROBE_MANUALLY enabled
```
System Windows 10
Aruino version 1.8.2 (Windows Store 1.8.5.0)
U8glib version 1.19.1

error message output:

```
ultralcd.cpp:1585: error: 'lcd_wait_for_move' was not declared in this scope
             lcd_wait_for_move = true;
             ^
ultralcd.cpp:1587: error: '_lcd_level_bed_done' was not declared in this scope
             lcd_goto_screen(_lcd_level_bed_done);
                             ^
C:\Users\SlawekC\AppData\Local\Temp\arduino_build_168886\sketch\ultralcd.cpp: In function 'void _lcd_level_bed_moving()':
ultralcd.cpp:1642: error: 'lcd_wait_for_move' was not declared in this scope
         if (!lcd_wait_for_move) lcd_goto_screen(_lcd_level_bed_get_z);
              ^

```
I'm still waiting for the Z-Probe to be installed so MBL iwith LCD is very good option for me.

Any idea how to fix this?

Tannoo, 2017-06-01:
I believe this is being worked on here: #6922

SCiunczyk, 2017-06-01:
Yes you right, I don't know how can I miss this.

closed 2017-06-01 by SCiunczyk